mezz64/pyHik #1: compatible firmware

boverby, 2016-12-27:
Thank you for contributing phHik. When using homeassistant, I ran into the log filling up with the error 

`16-12-27 20:53:30   pyhik.hikvision: Problem finding attribute: list index out of range`

My experience with hikvision firmware leads me to believe this a firmware related problem.
I am documenting my experiences to save other people some head scratching:

hikvision firmware V5.2.0 build 140721   (works)
hikvision firmware V5.3.0 build 150327  ( list index out of range)
hikvision firmware V5.3.0 build 150327  ( list index out of range)
   >>>  updated this same  camera to  5.4.0 build 160530  (works)

hikvision firmware V5.3.0 build 150513  ( list index out of range)
hikvision firmware v5.3.3 build 151019   (works)

Looks like 5.3.0 has a problem, firmwares below and above that seem to work fine.  The first solution for problems might be to upgrade the firmware on the camera.

@boverby

mezz64, 2016-12-29:
Did you do these tests with the latest 0.0.7 version, or the previous 0.0.6 version?  I don't believe the PR to bump HASS up the 0.0.7 has been merged into stable yet.

The 0.0.6 version uses a hard coded XML namespace which it seems may not match as many cameras as I though it did so I made the XML namespace discovery dynamic in the 0.0.7 version.

If you were on the 0.0.7 version could you post some XML samples from the cameras when they are running a firmware version that produces the error?  It's more than likely just an XML formatting change that I should be able to easily accommodate.

boverby, 2016-12-29:
I found 0.0.6 was under the deps directory and 0.0.7 was in dist-packages.

I tried a simple test of pyHik from command line and it failed on loading
pydispatch, so I will assume that the deps version (0.0.6) is running in
hass, because pydispatch is also in "deps".

Unfortunately (for troubleshooting) ,  I have since upgraded my 5 cameras
to 5.4.0 , and all are working fine.
I have never successfully downgraded a camera, so I won't be much help for
XML samples since all mine are working.
If you have any additional test or confirmations you need, I'll be glad to
help any way I can.

boverby

mezz64, 2016-12-29:
Also, as another data point, this library was developed with two cameras, both on V5.3.0 build 150513 firmware, which you seemed to have issues with.  So maybe there is something else at play here.

boverby, 2016-12-30:
Since you mentioned the build number, I recall  that on other programs
following the IPMD docs from v1.4.4 and v2.0, I frequently ran into onvif
commands that would stop working or act differently between 5.x.0 and 5.x.N
versions of the firmware. I have also made the mistake of skipping a
revision level ( 5.1.x to 5.3.0) and having to revert back with a hard tftp
upgrade (no fun).  That might be the extra sauce in the mix.  If yours were
virgin 5.3.0, and mine were migrated versions, there may have been some
additional differences we will never know.

ChipSkylark37, 2017-01-13:
Im experiencing the same issue with V5.4.0 build 160826 where its flooding the logs with list index out of range. Previous,y was on 5.3.4, and that was also having the same issue, decided to upgrade thinking it would resolve and no go. There was no interim release from LTS (rebranded firmware - not directly Hik) so I had to go straight to the latest.

On the other hand, I have another camera (EU version) running on V5.4.4 build 161125 and that works fine without issue and able to see all the binary sensors associated (line crossing, motion, etc).

If there's further logs or testing I could perform, let me know as I'd like to get this working inside of Home Assistant. Thanks!

mezz64, 2017-01-13:
See my reply to your issue on the HASS repository.  You're running one version behind on the pyHik library.  Your LTS camera should work with pyHik 0.0.7.

joachimleite, 2017-01-19:
I have two DS-2CD2532F-IWS cameras, (chinese version) with fw V5.2.5 build 141201 onboard. I get all motion, line detection and Intrution detection sensors triggered, but after a while, these sensors stop updating (in e.x HA).
Is fw V5.2.5 build 141201 tested ?

mezz64, 2017-01-19:
The cameras used for development were both on V5.3.0 build 150513.  I have not personally tested with anything else.  If the updates stopping is repeatable, please turn on debug logging for the component in HASS and look for any instances of the connection being dropped.  There is logic to reconnect automatically, but something may be getting hung up somewhere.

joachimleite, 2017-01-24:
I did a new test with 2CD2142 V5.4.0 build160511 which I borrow at work.
I get the same results: motion, line and intrution detection triggers for around 12 hours, then it get hung up. All activity is logged in the HASS log (activated debug logging). I scouted for some connections being droppet, but couldnt find any reason for the events to stop fireing...
Is there any other debug possibilities?
I really want to get this working, since it saves me from installing separate PIR sensors in my setup.

mezz64, 2017-01-24:
Can you post the debug log?

testermax, 2017-01-24:
At the moment, the line crossing and field detection is hung up, since I restarted HASS:
Log from HASS:
17-01-24 18:30:17 homeassistant.loader: Unable to find component switchWD
17-01-24 18:30:19 homeassistant.components.binary_sensor: Setting up binary_sensor.hikvision
17-01-24 18:30:19 homeassistant.components.binary_sensor: Setting up binary_sensor.hikvision
17-01-24 18:30:19 homeassistant.components.binary_sensor: Setting up binary_sensor.hikvision
17-01-24 18:30:19 homeassistant.components.binary_sensor.hikvision: Entity: Test - Illegal_Access, Options - Ignore: None, Delay: None
17-01-24 18:30:19 homeassistant.components.binary_sensor.hikvision: Entity: Test - Line_Crossing, Options - Ignore: None, Delay: None
17-01-24 18:30:19 homeassistant.components.binary_sensor.hikvision: Entity: Test - Disk_Error, Options - Ignore: None, Delay: None
17-01-24 18:30:19 homeassistant.components.binary_sensor.hikvision: Entity: Test - Field_Detection, Options - Ignore: None, Delay: None
17-01-24 18:30:19 homeassistant.components.binary_sensor.hikvision: Entity: Test - Motion, Options - Ignore: None, Delay: None
17-01-24 18:30:19 homeassistant.components.binary_sensor.hikvision: Entity: Inngang - Line_Crossing, Options - Ignore: True, Delay: 0
17-01-24 18:30:19 homeassistant.components.binary_sensor.hikvision: Entity: Inngang - Disk_Error, Options - Ignore: None, Delay: None
17-01-24 18:30:19 homeassistant.components.binary_sensor.hikvision: Entity: Inngang - Motion, Options - Ignore: True, Delay: 0
17-01-24 18:30:19 homeassistant.components.binary_sensor.hikvision: Entity: Inngang - Illegal_Access, Options - Ignore: None, Delay: None
17-01-24 18:30:19 homeassistant.components.binary_sensor.hikvision: Entity: Inngang - Disk_Full, Options - Ignore: None, Delay: None
17-01-24 18:30:19 homeassistant.components.binary_sensor.hikvision: Entity: Inngang - Field_Detection, Options - Ignore: None, Delay: None
17-01-24 18:30:21 homeassistant.components.binary_sensor.hikvision: Entity: Hage - Line_Crossing, Options - Ignore: None, Delay: None
17-01-24 18:30:21 homeassistant.components.binary_sensor.hikvision: Entity: Hage - Disk_Error, Options - Ignore: None, Delay: None
17-01-24 18:30:21 homeassistant.components.binary_sensor.hikvision: Entity: Hage - Motion, Options - Ignore: True, Delay: 0
17-01-24 18:30:21 homeassistant.components.binary_sensor.hikvision: Entity: Hage - Illegal_Access, Options - Ignore: None, Delay: None
17-01-24 18:30:21 homeassistant.components.binary_sensor.hikvision: Entity: Hage - Disk_Full, Options - Ignore: None, Delay: None
17-01-24 18:30:21 homeassistant.components.binary_sensor.hikvision: Entity: Hage - Field_Detection, Options - Ignore: True, Delay: 0
17-01-24 18:30:48 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Field Detection
17-01-24 18:30:48 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Line Crossing
17-01-24 18:31:00 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Line Crossing
17-01-24 18:31:04 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Line Crossing
17-01-24 18:31:14 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Field Detection
17-01-24 18:31:14 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Line Crossing
17-01-24 18:31:46 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Field Detection
17-01-24 18:31:51 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Line Crossing
17-01-24 18:31:56 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Field Detection
17-01-24 18:32:01 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Line Crossing
17-01-24 18:34:08 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Field Detection
17-01-24 18:34:08 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Line Crossing
17-01-24 18:34:18 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Field Detection
17-01-24 18:34:18 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Line Crossing
17-01-24 18:34:28 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Field Detection
17-01-24 18:34:38 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Field Detection
[... cut ...]
17-01-24 18:36:54 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Field Detection
17-01-24 18:37:05 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Field Detection
17-01-24 18:37:05 homeassistant.components.binary_sensor.hikvision: Dispatcher callback, signal: ValueChanged.5ced97f8-bb27-11b3-820e-006cfd17ff88, sender: Line Crossing

mezz64, 2017-01-24:
The key to debugging is enabling it for pyHik, not so much the HASS hikvision component.  Add the following to your logger settings:
```yaml
pyhik.hikvision: debug
```

testermax, 2017-01-24:
Good tip! I have activated logging now, and I can see all the event changes. I'll check the log after a while.
Thanks again.

testermax, 2017-01-25:
I have now got the logs filling up, and lost the events from the cameras at around 17:04 tonight.
The log says "Timeout getting camera image", but I get updated pictures in HASS, but no events from pyHik...

17-01-25 00:52:03 homeassistant.loader: Unable to find component switchWD
17-01-25 00:52:05 pyhik.hikvision: Initializing new hikvision device at: http://192.168.1.254
17-01-25 00:52:05 pyhik.hikvision: Initializing new hikvision device at: http://192.168.1.251
17-01-25 00:52:05 pyhik.hikvision: Initializing new hikvision device at: http://192.168.1.252
17-01-25 00:52:05 pyhik.hikvision: Using Namespace: http://www.hikvision.com/ver20/XMLSchema
17-01-25 00:52:05 pyhik.hikvision: Using Namespace: http://www.hikvision.com/ver20/XMLSchema
17-01-25 00:52:05 pyhik.hikvision: Found events: ['VMD', 'diskerror', 'illaccess', 'linedetection', 'fielddetection']
17-01-25 00:52:05 pyhik.hikvision: Initialized Dictionary: {'Disk Error': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 5, 912137)], 'Line Crossing': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 5, 912169)], 'Motion': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 5, 912064)], 'Illegal Access': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 5, 912153)], 'Field Detection': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 5, 912183)]}
17-01-25 00:52:05 pyhik.hikvision: Stream Thread Started: Test, 5ced97f8-bb27-11b3-820e-006cfd17ff88
17-01-25 00:52:05 pyhik.hikvision: Found events: ['VMD', 'diskfull', 'diskerror', 'illaccess', 'linedetection', 'fielddetection']
17-01-25 00:52:05 pyhik.hikvision: Initialized Dictionary: {'Disk Error': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 5, 954631)], 'Field Detection': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 5, 954674)], 'Illegal Access': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 5, 954647)], 'Line Crossing': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 5, 954661)], 'Motion': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 5, 954563)], 'Disk Full': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 5, 954614)]}
17-01-25 00:52:05 pyhik.hikvision: Stream Thread Started: Inngang, c08e578c-043b-11b4-826b-aca213d702f2
17-01-25 00:52:06 pyhik.hikvision: Using Namespace: http://www.hikvision.com/ver20/XMLSchema
17-01-25 00:52:07 pyhik.hikvision: Found events: ['VMD', 'diskfull', 'diskerror', 'illaccess', 'linedetection', 'fielddetection']
17-01-25 00:52:07 pyhik.hikvision: Initialized Dictionary: {'Disk Error': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 7, 48869)], 'Field Detection': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 7, 48911)], 'Illegal Access': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 7, 48883)], 'Line Crossing': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 7, 48897)], 'Motion': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 7, 48806)], 'Disk Full': [False, 1, 0, datetime.datetime(2017, 1, 25, 0, 52, 7, 48853)]}
17-01-25 00:52:07 pyhik.hikvision: Stream Thread Started: Hage, 5d779be9-3b69-11b5-83f7-aca213d6af38
17-01-25 00:52:46 homeassistant.components.recorder: Ended unfinished session (id=4 from 2017-01-24 19:59:35.661897)
17-01-25 00:52:46 homeassistant.components.recorder: Found unfinished sessions
17-01-25 01:06:12 homeassistant.components.recorder: Error during query: (sqlite3.OperationalError) database is locked
17-01-25 01:06:16 aiohttp.server: Error handling request
Traceback (most recent call last):
  File "/home/homeassistant/.homeassistant/deps/aiohttp/web_server.py", line 61, in handle_request
    resp = yield from self._handler(request)
  File "/home/homeassistant/.homeassistant/deps/aiohttp/web.py", line 249, in _handle
    resp = yield from handler(request)
  File "/usr/lib/python3.4/asyncio/coroutines.py", line 143, in coro
    res = yield from res
  File "/usr/lib/python3.4/asyncio/coroutines.py", line 143, in coro
    res = yield from res
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/homeassistant/components/http/__init__.py", line 427, in handle
    result = yield from result
  File "/srv/homeassistant/homeassistant_venv/lib/python3.4/site-packages/homeassistant/components/history.py", line 236, in get
    self.filters)
  File "/usr/lib/python3.4/asyncio/futures.py", line 388, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.4/asyncio/tasks.py", line 286, in _wakeup
    value = future.result()
  File "/usr/lib/python3.4/asyncio/futures.py", line 269, in result
    raise CancelledError
concurrent.futures._base.CancelledError
17-01-25 01:06:17 homeassistant.components.recorder: Error during query: (sqlite3.OperationalError) database is locked
17-01-25 01:06:22 homeassistant.components.recorder: Error during query: (sqlite3.OperationalError) database is locked
17-01-25 01:21:27 pyhik.hikvision: Inngang Update: Motion, [True, 1, 1, datetime.datetime(2017, 1, 25, 1, 21, 27, 781207)]
17-01-25 01:21:37 pyhik.hikvision: Inngang Update: Motion, [False, 1, 17, datetime.datetime(2017, 1, 25, 1, 21, 37, 635166)]
17-01-25 02:31:45 homeassistant.components.verisure: Could not login to Verisure MyPages, Not logged in
17-01-25 02:41:22 pyhik.hikvision: Inngang Update: Motion, [True, 1, 18, datetime.datetime(2017, 1, 25, 2, 41, 22, 660359)]
17-01-25 02:41:32 pyhik.hikvision: Inngang Update: Motion, [False, 1, 34, datetime.datetime(2017, 1, 25, 2, 41, 32, 586686)]
[... cut ...]
17-01-25 18:05:53 homeassistant.components.camera.generic: Error getting new camera image: 
17-01-25 18:06:25 homeassistant.components.camera.generic: Error getting new camera image: 
17-01-25 18:06:55 homeassistant.components.camera.generic: Error getting new camera image: 
17-01-25 18:06:55 homeassistant.components.camera.generic: Error getting new camera image: 
17-01-25 18:06:55 homeassistant.components.camera.generic: Error getting new camera image: 
17-01-25 18:07:59 homeassistant.components.camera.generic: Error getting new camera image:

mezz64, 2017-01-25:
You have lots of network errors in that log, not just from pyHik, but multiple other components.  What platform are you running HASS on?  Can you ping the cameras from the HASS host after the updates stop?

testermax, 2017-01-25:
Hi,
Im running HASS on a RPi 3. Im able to ping the cameras from HASS after updates stops.
The Pi is connected to my main router, through ethernet over powerline.
The cameras are either conencted to my main router directly or through extenders...

mezz64, 2017-01-25:
The only thing I can think of, given all your connection errors, is that the stream is getting corrupted in such a way that it's not disconnecting, but it's also no longer producing usable output.

If I remember correctly the "videoloss" alarm is continuous within the event stream so I suppose I could possibly use that as a keep-alive message.  Would you be able to do a simple curl request direct to your camera event stream and post the output here after letting it run for 60s or so?

From your logs, I still believe the root of your problems lies in networking, but i'm willing to work with you to make the connection lost/reconnect methods more robust and hopefully work around your issue.

testermax, 2017-01-26:
The log is really not representative of how my connections are. I usually get +-5 connection error in 24 hours. Not sure why I got all those in the previous log.
Now, I have connected the Rpi directly to my router, and the camera is connected through wifi.
Anyway, the get the log you mention, I`m not familiar with "curl request", but is it possible to do this through ssh when connected to the Rpi?
Thanks for helping me out.

mezz64, 2017-01-27:
I think most of the Pi distributions have curl installed by default.  Try the following command in an ssh window replacing the relevant details with the correct ones for your setup.

```bash
curl -u USER:PASS http://YOURCAMIP/ISAPI/Event/notification/alertStream
```

It should start dumping a bunch of XML into your session from the stream.  Let it run for 15-30seconds then break the connection with CTRL+C.  Post the output here.

testermax, 2017-01-28:
Here is my log:

[putty.txt](https://github.com/mezz64/pyHik/files/736791/putty.txt)

mezz64, 2017-01-29:
Ok, you also get a repeated 'videoloss' event when connected to the stream so I can use that as a connection check.  Give me some time to work in the changes and i'll have you do some testing.

mezz64, 2017-02-02:
I put together a simple watchdog class and implemented it to monitor the event stream.  I think this should fix your issue, but it's hard for me to test.

In order to minimize the HASS interactions i've pushed the changes to a new testing branch you can find here: https://github.com/mezz64/pyHik/tree/watchdog-test

Please download that branch to your PI and edit the "test_simple.py" file to match up with the settings for a camera you are having problems with.  Once the file is correct run it from your ssh session with:
```python3 test_simple.py```
Note that this will tie up that ssh session and will stop if you close it so you'll need to keep it open for 24hrs or so to do this test.

If all is well you should get the camera name, id, and supported events printed in the shell and a log file named "api_out.log" should be created in the same directory.  The log file will catch all pyhik actions and should let me know if things are working as expected.  Feel free to check it periodically and make sure events are being logged.  Upload it here after a 24hr run.

It may also be beneficial to start this test in conjunction with a fresh start of HASS.  That way I can compare HASS logs with the logs from the watchdog version.

testermax, 2017-02-05:
Hi,
I tried to run the testapp today, but it seems I`m missing some dependencies. I copied the repo to "Documents" on the pi and started test_simple.py from there:

pi@HomeAssistant:~/Documents/pyHik-watchdog-test $ python3 test_simple.py
Traceback (most recent call last):
  File "test_simple.py", line 6, in <module>
    import pyhik.hikvision as hikvision
  File "/home/pi/Documents/pyHik-watchdog-test/pyhik/hikvision.py", line 24, in <module>
    from pydispatch import dispatcher
ImportError: No module named 'pydispatch'

mezz64, 2017-02-05:
Quick solution is to install pydispatcher through pip.  The command below should get it in the right place for your python3 install.
```bash
sudo python3 -m pip install PyDispatcher
```

testermax, 2017-02-07:
Here is my 24 hour log:

02/06/2017 07:54:20 PM Initializing new hikvision device at: http://192.168.1.254
02/06/2017 07:54:20 PM Starting new HTTP connection (1): 192.168.1.254
02/06/2017 07:54:20 PM "GET /ISAPI/System/deviceInfo HTTP/1.1" 200 974
02/06/2017 07:54:20 PM Using Namespace: http://www.hikvision.com/ver20/XMLSchema
02/06/2017 07:54:20 PM Resetting dropped connection: 192.168.1.254
02/06/2017 07:54:20 PM "GET /ISAPI/Event/triggers HTTP/1.1" 200 6130
02/06/2017 07:54:20 PM Found events: ['VMD', 'diskerror', 'illaccess', 'linedetection', 'fielddetection']
02/06/2017 07:54:20 PM Initialized Dictionary: {'Illegal Access': [False, 1, 0, datetime.datetime(2017, 2, 6, 19, 54, 20, 942354)], 'Line Crossing': [False, 1, 0, datetime.datetime(2017, 2, 6, 19, 54, 20, 942369)], 'Field Detection': [False, 1, 0, datetime.datetime(2017, 2, 6, 19, 54, 20, 942384)], 'Disk Error': [False, 1, 0, datetime.datetime(2017, 2, 6, 19, 54, 20, 942337)], 'Motion': [False, 1, 0, datetime.datetime(2017, 2, 6, 19, 54, 20, 942297)]}
02/06/2017 07:54:20 PM Stream Thread Started: Test, 5ced97f8-bb27-11b3-820e-006cfd17ff88
02/06/2017 07:54:20 PM Starting new HTTP connection (1): 192.168.1.254
02/06/2017 07:54:20 PM "GET /ISAPI/Event/notification/alertStream HTTP/1.1" 200 None
02/06/2017 07:54:50 PM Test Update: Field Detection, [True, 1, 1, datetime.datetime(2017, 2, 6, 19, 54, 50, 786474)]
02/06/2017 07:55:00 PM Test Update: Field Detection, [False, 1, 17, datetime.datetime(2017, 2, 6, 19, 55, 0, 714517)]
02/06/2017 07:55:16 PM Test Update: Line Crossing, [True, 1, 1, datetime.datetime(2017, 2, 6, 19, 55, 16, 383997)]
02/06/2017 07:55:25 PM Test Update: Field Detection, [True, 1, 18, datetime.datetime(2017, 2, 6, 19, 55, 25, 696259)]
02/06/2017 07:55:25 PM Test Update: Line Crossing, [False, 1, 16, datetime.datetime(2017, 2, 6, 19, 55, 25, 998585)]
02/06/2017 07:55:35 PM Test Update: Field Detection, [False, 1, 34, datetime.datetime(2017, 2, 6, 19, 55, 35, 676153)]
02/06/2017 08:17:14 PM Test Update: Field Detection, [True, 1, 35, datetime.datetime(2017, 2, 6, 20, 17, 14, 316350)]
02/06/2017 08:17:17 PM Test Update: Line Crossing, [True, 1, 17, datetime.datetime(2017, 2, 6, 20, 17, 17, 301685)]
02/06/2017 08:17:23 PM Test Update: Field Detection, [False, 1, 50, datetime.datetime(2017, 2, 6, 20, 17, 23, 633752)]
02/06/2017 08:17:26 PM Test Update: Line Crossing, [False, 1, 32, datetime.datetime(2017, 2, 6, 20, 17, 26, 947485)]
02/06/2017 08:17:48 PM Test Update: Line Crossing, [True, 1, 33, datetime.datetime(2017, 2, 6, 20, 17, 48, 622568)]
02/06/2017 08:17:58 PM Test Update: Line Crossing, [False, 1, 49, datetime.datetime(2017, 2, 6, 20, 17, 58, 560435)]
02/06/2017 08:55:12 PM Test Update: Line Crossing, [True, 1, 50, datetime.datetime(2017, 2, 6, 20, 55, 12, 779644)]
02/06/2017 08:55:22 PM Test Update: Line Crossing, [False, 1, 66, datetime.datetime(2017, 2, 6, 20, 55, 22, 712211)]
02/07/2017 07:22:16 AM Test Update: Field Detection, [True, 1, 51, datetime.datetime(2017, 2, 7, 7, 22, 16, 137846)]
02/07/2017 07:22:26 AM Test Update: Field Detection, [False, 1, 67, datetime.datetime(2017, 2, 7, 7, 22, 26, 83777)]
02/07/2017 07:22:45 AM Test Update: Field Detection, [True, 1, 68, datetime.datetime(2017, 2, 7, 7, 22, 45, 345568)]
02/07/2017 07:22:45 AM Test Update: Line Crossing, [True, 1, 67, datetime.datetime(2017, 2, 7, 7, 22, 45, 649430)]
02/07/2017 07:22:55 AM Test Update: Field Detection, [False, 1, 85, datetime.datetime(2017, 2, 7, 7, 22, 55, 301459)]
02/07/2017 07:22:55 AM Test Update: Line Crossing, [False, 1, 83, datetime.datetime(2017, 2, 7, 7, 22, 55, 603818)]
02/07/2017 07:23:06 AM Test Update: Field Detection, [True, 1, 86, datetime.datetime(2017, 2, 7, 7, 23, 6, 442501)]
02/07/2017 07:23:06 AM Test Update: Line Crossing, [True, 1, 84, datetime.datetime(2017, 2, 7, 7, 23, 6, 742929)]
02/07/2017 07:23:15 AM Test Update: Field Detection, [False, 1, 99, datetime.datetime(2017, 2, 7, 7, 23, 15, 183787)]
02/07/2017 07:23:16 AM Test Update: Line Crossing, [False, 1, 100, datetime.datetime(2017, 2, 7, 7, 23, 16, 685196)]
02/07/2017 05:00:22 PM Test Update: Field Detection, [True, 1, 100, datetime.datetime(2017, 2, 7, 17, 0, 22, 998076)]
02/07/2017 05:00:32 PM Test Update: Field Detection, [False, 1, 115, datetime.datetime(2017, 2, 7, 17, 0, 32, 622264)]
02/07/2017 05:00:43 PM Test Update: Line Crossing, [True, 1, 101, datetime.datetime(2017, 2, 7, 17, 0, 43, 153410)]
02/07/2017 05:00:48 PM Test Update: Field Detection, [True, 1, 116, datetime.datetime(2017, 2, 7, 17, 0, 48, 871613)]
02/07/2017 05:00:53 PM Test Update: Line Crossing, [False, 1, 117, datetime.datetime(2017, 2, 7, 17, 0, 53, 92189)]
02/07/2017 05:00:58 PM Test Update: Field Detection, [False, 1, 131, datetime.datetime(2017, 2, 7, 17, 0, 58, 510987)]
02/07/2017 05:01:18 PM Test Update: Field Detection, [True, 1, 132, datetime.datetime(2017, 2, 7, 17, 1, 18, 698096)]
02/07/2017 05:01:33 PM Test Update: Field Detection, [False, 1, 164, datetime.datetime(2017, 2, 7, 17, 1, 33, 454260)]
02/07/2017 05:01:50 PM Test Update: Field Detection, [True, 1, 165, datetime.datetime(2017, 2, 7, 17, 1, 50, 21557)]
02/07/2017 05:01:59 PM Test Update: Field Detection, [False, 1, 181, datetime.datetime(2017, 2, 7, 17, 1, 59, 973880)]

mezz64, 2017-02-07:
Ok, so it looks like everything stayed connected just fine for the complete 24hrs, correct?  Did the HASS version stop itself during this same time period?

I don't see any instances of the watchdog or connection reset mechanisms kicking in during your test so there may be something weird going on with the pyhik/hass interaction on your system.

testermax, 2017-02-07:
Yes the HASS version stopped registering the line detections pretty quick after I restarted HASS, yesterday at around eight oclock afternoon. 
I agree, there seems to be something with the pyhik/hass interaction, but Its quite simple to set this up, so I cant really understand what it should be.

mezz64, 2017-02-07:
What other components are you running in HASS along with hikvision?  Have you tried running a HASS instance with just the hikvision component setup?

testermax, 2017-02-07:
I have not tried that, but I have noticed that when I run generic cameras (hikvision) together with binary sensors (hikvision), it seems to be more unstable. Other components are Hue, sonos, some mqtt, snmp...
But I will try to isolate this down, by removing all other components, and add the to see if there is any interference.

mezz64, 2017-02-26:
Watchdog and improved notification functionality are now available in 0.0.9 and should show up in the next version of HASS.  I'm closing this issue for now.  Please open a new one if issues persist after using 0.0.9.

closed 2017-02-26 by mezz64

testermax, 2017-02-26:
Thank you for your support. Im looking forward to test v 0.0.9 in Hass.