mezz64/pyHik #2: high cpu usage when feeding from DS-2CD2542FWD-IS

vladm, 2017-03-03:
ticket resurrected from https://github.com/home-assistant/home-assistant/issues/6315

camera firmware 5.4.5

vladm, 2017-03-03:
/ISAPI/Event/triggers

```
<EventNotification xmlns="http://www.hikvision.com/ver20/XMLSchema" version="2.0">
<EventTriggerList xmlns="http://www.hikvision.com/ver20/XMLSchema" version="2.0">
<EventTrigger>
<id>IO-1</id>
<eventType>IO</eventType>
<eventDescription>IO Event trigger Information</eventDescription>
<inputIOPortID>1</inputIOPortID>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>VMD-1</id>
<eventType>VMD</eventType>
<eventDescription>VMD Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
<EventTriggerNotification>
<id>record-1</id>
<notificationMethod>record</notificationMethod>
<videoInputID>1</videoInputID>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
<EventTriggerNotification>
<id>beep</id>
<notificationMethod>beep</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
<EventTriggerNotification>
<id>FTP</id>
<notificationMethod>FTP</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>tamper-1</id>
<eventType>tamperdetection</eventType>
<eventDescription>shelteralarm Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>diskfull</id>
<eventType>diskfull</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
<EventTriggerNotification>
<id>email</id>
<notificationMethod>email</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>diskerror</id>
<eventType>diskerror</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
<EventTriggerNotification>
<id>beep</id>
<notificationMethod>beep</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>nicbroken</id>
<eventType>nicbroken</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>ipconflict</id>
<eventType>ipconflict</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>illaccess</id>
<eventType>illaccess</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>linedetection-1</id>
<eventType>linedetection</eventType>
<eventDescription>Linedetection Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>fielddetection-1</id>
<eventType>fielddetection</eventType>
<eventDescription>fielddetection Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>videomismatch</id>
<eventType>videomismatch</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
<EventTriggerNotification>
<id>beep</id>
<notificationMethod>beep</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>badvideo</id>
<eventType>badvideo</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>storageDetection-1</id>
<eventType>storageDetection</eventType>
<eventDescription>storageDetection Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
</EventTriggerList>
</EventNotification>
```

vladm, 2017-03-03:
/ISAPI/System/Video/inputs/channels/1/motionDetection

```
<MotionDetection xmlns="http://www.hikvision.com/ver20/XMLSchema" version="2.0">
<enabled>true</enabled>
<enableHighlight>false</enableHighlight>
<samplingInterval>2</samplingInterval>
<startTriggerTime>500</startTriggerTime>
<endTriggerTime>500</endTriggerTime>
<regionType>grid</regionType>
<Grid>
<rowGranularity>18</rowGranularity>
<columnGranularity>22</columnGranularity>
</Grid>
<MotionDetectionLayout xmlns="http://www.hikvision.com/ver20/XMLSchema" version="2.0">
<sensitivityLevel>60</sensitivityLevel>
<layout>
<gridMap>
000000000000000000000000000000000000000000007ffce07ffce07ffce07ffce07ffce07ffce07ffce07ffcfffffcfffffcfffffc
</gridMap>
</layout>
</MotionDetectionLayout>
</MotionDetection>

```

mezz64, 2017-03-03:
Is your working camera also on firmware 5.4.5?  Can you try the high cpu camera with only one sensor type enabled to see if it persists?

vladm, 2017-03-03:
yes, both are on the same FW version. tested with single camera, only one motion alarm - still bad

```
17-03-03 08:30:56 INFO (MainThread) [homeassistant.loader] Loaded binary_sensor.hikvision from homeassistant.components.binary_sensor.hikvision
17-03-03 08:30:56 INFO (MainThread) [homeassistant.bootstrap] Setting up binary_sensor
17-03-03 08:30:56 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.hikvision
17-03-03 08:30:56 DEBUG (Thread-3) [pyhik.hikvision] Initializing new hikvision device at: http://10.7.7.179
17-03-03 08:30:56 DEBUG (Thread-3) [pyhik.hikvision] Using Namespace: http://www.hikvision.com/ver20/XMLSchema
17-03-03 08:30:56 DEBUG (Thread-3) [pyhik.hikvision] Found events: ['VMD']
17-03-03 08:30:56 DEBUG (Thread-3) [pyhik.hikvision] Initialized Dictionary: {'Motion': [False, 1, 0, datetime.datetime(2017, 3, 3, 8, 30, 56, 322019)]}
17-03-03 08:30:56 DEBUG (Thread-12) [pyhik.hikvision] Stream Thread Started: CAMERA, 5269afcc-ea79-11b4-8390-a4143724644a
17-03-03 08:30:56 DEBUG (Thread-3) [homeassistant.components.binary_sensor.hikvision] Entity: CAMERA - Motion, Options - Ignore: None, Delay: None
17-03-03 08:30:56 INFO (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: new_state=<state binary_sensor.camera_motion=off; last_tripped_time=2017-03-03T02:30:56.322019-06:00, friendly_name=CAMERA Motion, device_class=motion @ 2017-03-03T08:30:56.344510-06:00>, old_state=None, entity_id=binary_sensor.camera_motion>
17-03-03 08:30:56 INFO (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=binary_sensor>

```

mezz64, 2017-03-03:
First, in the interest of working with the latest, please test against the latest HASS dev build which now utilizes the latest version of pyHik.

Second, with the high cpu usage do you still get events registered or does it just suck up the cpu and do nothing?

vladm, 2017-03-03:
Seems like with the latest dev image there is no high-cpu usage, however it doesn't register the motion on HA side.

Perhaps, I can test it with an isolated script? Do you remember the camera URL with the event feed? I can capture it from camera via some wget/curl and provide for the reference.

mezz64, 2017-03-03:
This would be the curl command to get the event stream:
```
curl -u USER:PASS http://YOURCAMIP/ISAPI/Event/notification/alertStream
```

If you can, capture the output during a motion event and post it here.  If nothing looks odd there i'll provide a test file to work with the library directly.

vladm, 2017-03-03:
update.
I tried to access /ISAPI/Event/notification/alertStream via browser (after logged in as the same user as used on haas into camera). I've got XML with insufficient permission error message instead of a stream.
Turns out that camera username didn't have "Remote: Notify Surveillance Center" permission enabled. 

So, it looks like missing permission is what was causing the whole snafu, but for some reason it wasn't handled gracefully by pyHik module...

Hope, this helps.

mezz64, 2017-03-03:
Glad you got it figured out.  The latest version does a check to ensure the connection response is OK, that's why you got no response on motion events.  You should have seen a "Connection Unsuccessful" message in the logs though.  Do you recall if that was the case?

vladm, 2017-03-03:
don't have the log anymore, but now as I know what was causing the problem, it's easy to check by removing that permission and checking what happens in the log,  if you want me to.

Thank you for the help and tweaks; I'm closing the ticket.

closed 2017-03-03 by vladm