mezz64/pyHik #3: Using LTS NVR with pyHik

wizwuz7, 2017-03-08:
I'm running HA 0.39.3.
LTS (Hikvision) CMIP3022W IP camera - working great.
LTS (Hikvision) LTN8716T-HT NVR errors with 'Problem finding attribute: list index out of range' error.

mezz64, 2017-03-08:
I'm attaching the device trigger file you posted over on the HASS forum here so I can easily find it later.
[dvr_triggers.txt](https://github.com/mezz64/pyHik/files/826668/dvr_triggers.txt)

wizwuz7, 2017-03-16:
No events found if using existing code.  Changed to: 
ettype = eventtrigger.find('{http://www.std-cgi.com/ver20/XMLSchema}eventType')
This worked, but many duplicates due to multiple ports (like 16 iterations of VMD).
Perhaps <id> value is unique (like VMD-1, VMD-2, etc.)

mezz64, 2017-03-17:
I've finished the changes on my other library so I should have some time to tackle this now.  Also recently got access to a hikvision NVR I can use for testing.  I'll make a Dev branch for the changes and post back here when I have something ready for you to test.

mezz64, 2017-03-21:
Could you post a sample from your NVR alert stream?  The following command will retrieve it with curl and send the output to a file.

```
curl -u USER:PASS http://YOUR-NVR-IP/ISAPI/Event/notification/alertStream > stream.txt
```

I'm seeing some variations in naming between the test NVR I have access to and the triggers XML sample you provided earlier so I want to make sure I cover all the variations.

wizwuz7, 2017-03-21:
I'm sure you wish the naming conventions were a little more standardized.
This NVR is a hybrid with 16 analog inputs plus 16 IP camera inputs. It also has IO inputs.
Included several incidents - IO alarm event, videoloss event, motion detection event, and line crossing event.
[stream.txt](https://github.com/mezz64/pyHik/files/859042/stream.txt)

mezz64, 2017-03-22:
Are the IO inputs a priority for you?  They are reported a little differently in the event triggers list so they may not be included in the first NVR revision.  

Your stream also shows some slightly different behavior in terms of the the 'videoloss' event that's continually posted to channel 0.  Some adjustments might be needed to the watchdog to keep things working smoothly, but we can get that all figured out once we get to the testing phase.

wizwuz7, 2017-03-22:
IO not a priority. I will be using this to trigger camera notifications.

mezz64, 2017-03-25:
I've gotten things to a point where they seem to be working reliably with my test NVR so it's time to see if you can break it. :)

I think the easiest way to test it out is download the "nvr" branch of pyhik from github and copy the pyhik folder to your hass "deps" directory.
https://github.com/mezz64/pyHik/tree/nvr

Then grab the updated hass hikvision binary sensor component from the commit linked below and place it in the following folder tree in your hass config directory: ```custom_components/binary_senors/```.  Once it's there comment out the ```REQUIREMENTS``` line at the top of the file so HASS just looks in the deps directory rather than trying to use the currently non-existent 0.1.1 version.
https://github.com/mezz64/home-assistant/blob/8b31311ae83283d9341e5cdb0574d1d168104aa0/homeassistant/components/binary_sensor/hikvision.py

Let me know if you encounter any issues.

mezz64, 2017-03-25:
Use this hass commit.  I made some changes to keep sensor names the same for existing users.
https://github.com/mezz64/home-assistant/blob/6e86886120e5e9cd80dbf0aeda9418df2885567f/homeassistant/components/binary_sensor/hikvision.py

wizwuz7, 2017-03-25:
See attached. Adding debug logger did not increase error logging. Using curl to the eventStream still works. Changing login password to the NVR gives a different error message. 
[error_1.txt](https://github.com/mezz64/pyHik/files/870245/error_1.txt)

mezz64, 2017-03-25:
Make sure you have the following logging definitions in your hass config to get debug logs:
```
pyhik: debug
custom_components.binary_sensor.hikvision: debug
```

Once those are defined start things up again and post the full log.

wizwuz7, 2017-03-25:
Noted the constants file was using the hikvision style - changed to std-cgi.
Attached error.
[error_2.txt](https://github.com/mezz64/pyHik/files/870308/error_2.txt)

mezz64, 2017-03-25:
No need to change the constants.py for the namespace.  It gets updated dynamically during setup.  Based on that log it looks like your are still using the last release version of pyHik, not the one from the nvr branch.

Can you double check that you commented out the requirements line in the hass hikvision binary sensor file and that the you have a ```pyhik``` folder in your ```custom_components``` directory that contains the pyhik files ```__init__.py, constants.py, hikvision.py, watchdog.py```?

wizwuz7, 2017-03-25:
Requirements line in binary_sensor/hikvision.py has been removed.
The pyhik folder was in loaded into deps.
Copied pyhik folder into custom_components folder.
Same errors.

mezz64, 2017-03-25:
Directories should be as follows:
```custom_components/binary_sensor/hikvision.py```  *This should be the file from the commit linked above with REQUIREMENTS commented out.
```deps/pyhik/**pyhik library files**``` *These should be from the nvr branch of this repo.

If that's how you have it setup then something isn't working right in hass in terms of order of imports.  What version of hass are you testing with?

wizwuz7, 2017-03-25:
HA v0.40.2.
Redropped the pyhik folder into deps folder.
More info now in the error file.
[error_3.txt](https://github.com/mezz64/pyHik/files/870349/error_3.txt)

mezz64, 2017-03-25:
Ok, we are making progress. 👍 

Your logs show you are utilizing the new pyhik library, but not the new hikvision.py hass component.  If for some reason you can't get it to load from the ```config/custom_components``` directory you can always overwrite the existing file in the ```components/binary_sensor``` directory also.

wizwuz7, 2017-03-25:
Overwrote existing hikvision.py in components/binary_sensor folder.
Looks better now.
See attached.
[error_4.txt](https://github.com/mezz64/pyHik/files/870364/error_4.txt)

mezz64, 2017-03-25:
I think I fixed your error with the latest commit to the nvr branch.  Please download and replace the files in your ```deps``` directory and give it another shot.

wizwuz7, 2017-03-25:
Yep - that looks like it.
Both Line Detection and motion are working on various inputs, both analog and digital.
No errors - excellent work.
[error_5.txt](https://github.com/mezz64/pyHik/files/870442/error_5.txt)

mezz64, 2017-03-26:
Awesome. Glad it's working for you. I'll wait a day or so to merge into master and submit a PR to hass just in case something pops up after you use it for a while.

mezz64, 2017-03-27:
Closing this now... Version 0.1.1 with NVR support has been pushed to pypi.  PR to integrate with hass is submitted.

closed 2017-03-27 by mezz64