mezz64/pyHik #5: Re-Branded Hik - Different URL

Tim-GitHub, 2017-03-30:
I have several re-branded Hikvision cameras as well which aren't working. Looks like the triggers.xml has a different path on those models. To access the triggers.xml the URL is:

`http://XXX.XXX.XXX.XXX/Event/triggers`

Based on the code here at GitHub you have the URL hardcoded. Looks like we need to add a second URL possibility to check so that these re-branded models will be detected.

Additionally these cameras have a different `<notificationMethod>`. Instead of center they use:

```
- <EventTriggerNotification>
    <id>3</id> 
    <notificationMethod>HTTP</notificationMethod> 
    <notificationRecurrence>beginning</notificationRecurrence> 
  </EventTriggerNotification>
```

(Originally posted to: https://community.home-assistant.io/t/hikvision-event-stream-sensors/6789/71)

mezz64, 2017-03-30:
What type of http response do you receive when you query the ```/ISAPI/Event/triggers``` url on your cameras?  Is it just a 404?

To the second part, you are saying that for your cams when you check ```Notify Surveillance Center``` in the camera configuration the method is reported as ```HTTP``` in the event stream, correct?

I just want to make sure that adding in the alternate options won't cause unintended behavior on the branded Hikvision cameras.

Tim-GitHub, 2017-03-30:
When attempting to access `/ISAPI/Event/triggers` on the re-branded models I get the following pages:

Chrome:

```
Access Error: 404 -- Not Found

Can't locate document: /ISAPI/Event/triggers
```
[image]

IE11:

```
The webpage cannot be found

Most likely causes:
     • There might be a typing error in the address.
     • If you clicked on a link, it may be out of date.
 
What you can try:
 
    • Retype the address.  
    • Go back to the previous page.  
    • Go to  and look for the information you want.  

More information

This error (HTTP 404 Not Found) means that Internet Explorer was able to connect to the website, but the page you wanted was not found. It’s possible that the webpage is temporarily unavailable. Alternatively, the website might have changed or removed the webpage.
```

[image]

For the second part this is the complete contents of `/Event/triggers` with only `Notify Surveillance Center` enabled for motion on the camera. Forgive the formatting if it's not quite right:

```
 <?xml version="1.0" encoding="UTF-8" ?> 
- <EventTriggerList version="1.0" xmlns="http://www.std-cgi.com/ver10/XMLSchema">
	- <EventTrigger version="1.0" xmlns="http://www.std-cgi.com/ver10/XMLSchema">
		  <id>1</id> 
		  <eventType>VMD</eventType> 
		  <eventDescription>VMD Event trigger Information</eventDescription> 
		- <EventTriggerNotificationList version="1.0" xmlns="http://www.std-cgi.com/ver10/XMLSchema">
			- <EventTriggerNotification>
				  <id>3</id> 
				  <notificationMethod>HTTP</notificationMethod> 
				  <notificationRecurrence>beginning</notificationRecurrence> 
			</EventTriggerNotification>
			</EventTriggerNotificationList>
		</EventTrigger>
	  <EventTrigger version="1.0" xmlns="http://www.std-cgi.com/ver10/XMLSchema" /> 
	  <EventTrigger version="1.0" xmlns="http://www.std-cgi.com/ver10/XMLSchema" /> 
	</EventTriggerList>
```

Tim-GitHub, 2017-03-30:
Ah something interesting. On my genuine USA Hikvision `/Event/triggers` works as well. The list is much more condensed though like the sample above. Also the `notificationMethod` is `HTTP` in `/Event/triggers` but `center` in `/ISAPI/Event/triggers`

Same camera (USA Hikvision):

`/ISAPI/Event/triggers`:
[iaspi-events-triggers.txt](https://github.com/mezz64/pyHik/files/880912/iaspi-events-triggers.txt)
`/Event/triggers`:
[events-triggers.txt](https://github.com/mezz64/pyHik/files/880913/events-triggers.txt)

mezz64, 2017-03-30:
Your triggers XML is very similar to what gets returned for NVR's, but with the center/HTTP variation and unfortunately yet another naming scheme for the channel id.

Since you get a 404 on fetching the existing url it should be easy enough to catch it and switch to the other one.  Adding in the HTTP shouldn't be an issue.  Only thing I need to put some thought into is detecting the difference between this camera and an NVR so the channels get processed correctly.

I'll post back once i've got something for you to test.

Tim-GitHub, 2017-03-30:
Fantastic thanks!

mezz64, 2017-03-31:
I made the changes that should get things working for you.

Easiest way to test it out is download the "altcam" branch of pyhik from github and copy the pyhik folder to your hass ```configuration/deps``` directory.
https://github.com/mezz64/pyHik/tree/altcam

Then comment out the ```REQUIREMENTS``` line at the top of the ```hikvision.py``` file in your ```component/binary_sensor``` folder so HASS just looks in the deps directory rather than trying to use the current pip version.

Let me know how things work out.

Tim-GitHub, 2017-03-31:
I copied the altcam branch and commented out the `REQUIREMENTS` line. Still no luck. Relevant log snippet below:

```
17-03-30 20:50:00 INFO (MainThread) [homeassistant.loader] Loaded binary_sensor.hikvision from homeassistant.components.binary_sensor.hikvision
17-03-30 20:50:01 INFO (MainThread) [homeassistant.bootstrap] Setting up binary_sensor
17-03-30 20:50:01 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.hikvision
17-03-30 20:50:01 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5EB90>_4) [pyhik.hikvision] Initializing new hikvision device at: http://192.168.1.41
17-03-30 20:50:01 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5EB90>_4) [pyhik.hikvision] Using Namespace: http://www.hikvision.com/ver20/XMLSchema
17-03-30 20:50:01 ERROR (MainThread) [homeassistant.components.binary_sensor] Error while setting up platform hikvision
Traceback (most recent call last):
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\helpers\entity_component.py", line 151, in _async_setup_platform
    entity_platform.add_entities, discovery_info
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\concurrent\futures\thread.py", line 55, in run
    result = self.fn(*self.args, **self.kwargs)
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\components\binary_sensor\hikvision.py", line 85, in setup_platform
    data = HikvisionData(hass, url, port, name, username, password)
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\components\binary_sensor\hikvision.py", line 123, in __init__
    self._username, self._password)
  File "C:\Users\ha\AppData\Roaming\.homeassistant\deps\pyhik\hikvision.py", line 104, in __init__
    self.initialize()
  File "C:\Users\ha\AppData\Roaming\.homeassistant\deps\pyhik\hikvision.py", line 145, in initialize
    device_info = self.get_device_info()
  File "C:\Users\ha\AppData\Roaming\.homeassistant\deps\pyhik\hikvision.py", line 284, in get_device_info
    tag = item.tag.split('}')[1]
IndexError: list index out of range
```

mezz64, 2017-03-31:
Problem is it seems like you have no ISAPI endpoint at all so all the default API urls won't work.  I pushed another update that tries your variant when it can't get the first for deviceinfo, triggers, and the event stream.  Give it a shot.

Tim-GitHub, 2017-04-01:
Getting closer (I hope!)

```
17-03-31 18:36:51 INFO (MainThread) [homeassistant.loader] Loaded binary_sensor.hikvision from homeassistant.components.binary_sensor.hikvision
17-03-31 18:36:51 INFO (MainThread) [homeassistant.bootstrap] Setting up binary_sensor
17-03-31 18:36:51 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.hikvision
17-03-31 18:36:51 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x0265CB90>_0) [pyhik.hikvision] Initializing new hikvision device at: http://192.168.1.41
17-03-31 18:36:52 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x0265CB90>_0) [pyhik.hikvision] Using Namespace: http://www.hikvision.com/ver10/XMLSchema
17-03-31 18:36:52 ERROR (MainThread) [homeassistant.components.binary_sensor] Error while setting up platform hikvision
Traceback (most recent call last):
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\helpers\entity_component.py", line 151, in _async_setup_platform
    entity_platform.add_entities, discovery_info
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\concurrent\futures\thread.py", line 55, in run
    result = self.fn(*self.args, **self.kwargs)
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\components\binary_sensor\hikvision.py", line 85, in setup_platform
    data = HikvisionData(hass, url, port, name, username, password)
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\components\binary_sensor\hikvision.py", line 123, in __init__
    self._username, self._password)
  File "C:\Users\ha\AppData\Roaming\.homeassistant\deps\pyhik\hikvision.py", line 104, in __init__
    self.initialize()
  File "C:\Users\ha\AppData\Roaming\.homeassistant\deps\pyhik\hikvision.py", line 159, in initialize
    events_available = self.get_event_triggers()
  File "C:\Users\ha\AppData\Roaming\.homeassistant\deps\pyhik\hikvision.py", line 234, in get_event_triggers
    for notifytrigger in etnotify:
TypeError: 'NoneType' object is not iterable
```

mezz64, 2017-04-01:
Your logs are indicating that unexpected data is getting returned during the initial device info stage.  I've pushed a new commit with some more debugging statements that should lend some more insight.  Can you also verify that this URL is accessible on your cameras and post the output here?
```
http://YOUR_CAM_IP/System/deviceInfo
```

Tim-GitHub, 2017-04-01:
Tested.

Relevant logs:

```
17-04-01 10:05:13 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_0) [pyhik.hikvision] Initializing new hikvision device at: http://192.168.1.41
17-04-01 10:05:14 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_0) [pyhik.hikvision] Using alternate deviceInfo URL.
17-04-01 10:05:15 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_0) [pyhik.hikvision] Auto Namespace Result: http://www.hikvision.com/ver10/XMLSchema
17-04-01 10:05:15 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_0) [pyhik.hikvision] Using Namespace: http://www.hikvision.com/ver10/XMLSchema
17-04-01 10:05:17 ERROR (MainThread) [homeassistant.components.binary_sensor] Error while setting up platform hikvision
Traceback (most recent call last):
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\helpers\entity_component.py", line 151, in _async_setup_platform
    entity_platform.add_entities, discovery_info
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\concurrent\futures\thread.py", line 55, in run
    result = self.fn(*self.args, **self.kwargs)
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\components\binary_sensor\hikvision.py", line 85, in setup_platform
    data = HikvisionData(hass, url, port, name, username, password)
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\components\binary_sensor\hikvision.py", line 123, in __init__
    self._username, self._password)
  File "C:\Users\ha\AppData\Roaming\.homeassistant\deps\pyhik\hikvision.py", line 104, in __init__
    self.initialize()
  File "C:\Users\ha\AppData\Roaming\.homeassistant\deps\pyhik\hikvision.py", line 159, in initialize
    events_available = self.get_event_triggers()
  File "C:\Users\ha\AppData\Roaming\.homeassistant\deps\pyhik\hikvision.py", line 234, in get_event_triggers
    for notifytrigger in etnotify:
TypeError: 'NoneType' object is not iterable
```

deviceInfo:

```
<DeviceInfo xmlns="http://www.hikvision.com/ver10/XMLSchema" version="1.0">
<deviceName>Front Door</deviceName>
<deviceID>88</deviceID>
<deviceDescription>IPCamera</deviceDescription>
<deviceLocation>hangzhou</deviceLocation>
<systemContact>Hikvision.China</systemContact>
<model>NC-KD032</model>
<serialNumber>NC-KD0320120130609CCRR424264952</serialNumber>
<macAddress>8c:e7:48:f7:0e:97</macAddress>
<firmwareVersion>V5.0.2</firmwareVersion>
<firmwareReleasedDate>130805</firmwareReleasedDate>
<bootVersion>V1.3.4</bootVersion>
<bootReleasedDate>100316</bootReleasedDate>
<hardwareVersion>V4.0</hardwareVersion>
</DeviceInfo>
```

mezz64, 2017-04-01:
Ok, something doesn't make sense here.  The first trigger response you posted listed a std-cgi.com namespace yet the device info shows a hikvision.com namespace.  Are you testing with the same camera?

In any case I adding handling for the error you posted, give it another shot.

Tim-GitHub, 2017-04-02:
Oops yes your right I mixed up which camera I was testing on from the previous night. Regardless it looks like your new commits were successful! There still appears that there may be an issue based on an error in the log and I don't see the sensor appear on my HA states web ui.

```
17-04-01 22:17:21 INFO (MainThread) [homeassistant.bootstrap] Setting up binary_sensor
17-04-01 22:17:21 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.hikvision
17-04-01 22:17:21 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_2) [pyhik.hikvision] Initializing new hikvision device at: http://192.168.1.41
17-04-01 22:17:21 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_2) [pyhik.hikvision] Using alternate deviceInfo URL.
17-04-01 22:17:21 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_2) [pyhik.hikvision] Auto Namespace Result: http://www.hikvision.com/ver10/XMLSchema
17-04-01 22:17:21 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_2) [pyhik.hikvision] Using Namespace: http://www.hikvision.com/ver10/XMLSchema
17-04-01 22:17:21 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_2) [pyhik.hikvision] Using alternate triggers URL.
17-04-01 22:17:21 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_2) [pyhik.hikvision] Processed CAM Device.
17-04-01 22:17:21 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_2) [pyhik.hikvision] Found events: {'VMD': [1]}
17-04-01 22:17:21 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_2) [pyhik.hikvision] Initialized Dictionary: {'Motion': [[False, 1, 0, datetime.datetime(2017, 4, 1, 22, 17, 21, 795821)]]}
17-04-01 22:17:21 DEBUG (Thread-3) [pyhik.hikvision] Stream Thread Started: Front Door, 88
17-04-01 22:17:22 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_2) [homeassistant.components.binary_sensor.hikvision] Entity: Front Door - Motion, Options - Ignore: None, Delay: None
17-04-01 22:17:22 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\helpers\entity_component.py", line 341, in _async_process_entity
    new_entity, self, update_before_add=update_before_add
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\helpers\entity_component.py", line 209, in async_add_entity
    yield from entity.async_update_ha_state()
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\helpers\entity.py", line 228, in async_update_ha_state
    device_attr = self.device_state_attributes
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\components\binary_sensor\hikvision.py", line 221, in device_state_attributes
    attr[ATTR_LAST_TRIP_TIME] = self._sensor_last_update()
  File "C:\Users\ha\AppData\Local\Programs\Python\Python36-32\lib\site-packages\homeassistant\components\binary_sensor\hikvision.py", line 186, in _sensor_last_update
    return self._cam.sensors[self._sensor][3]
IndexError: list index out of range
17-04-01 22:17:22 DEBUG (Thread-3) [pyhik.hikvision] Front Door Connection Successful.
```

mezz64, 2017-04-02:
Looks like you aren't using the latest hikvision.py currently in the HASS dev branch.  Please test with that version.  Make sure to comment out the REQUIREMENTS line.

Tim-GitHub, 2017-04-02:
That did it! Works perfectly now with all my various re-branded Hikvision cameras! Thank you for all your help getting this working. Ever since installing my cameras a couple years ago I always wanted to integrate motion events my HA system. Now to start playing with them!

mezz64, 2017-04-03:
Once you multiple cam issue gets fixed and I verify everything still works as expected with normal hik cams and nvrs I'll make a release for the changes and get the updates merged into HASS.

mezz64, 2017-04-06:
Updated version has been merged into HASS dev.  Closing this issue.

closed 2017-04-06 by mezz64