mezz64/pyHik #6: Multiple Cameras

Tim-GitHub, 2017-04-02:
pyHik appears unable to handle multiple Hikvision cameras simultaneously. Individually tested each with pyHik without issue. When adding them both to configuration.yaml, however, only one is displayed in states.

**configuration.yaml:**

```
#Hikvision Setup
binary_sensor:
  - platform: hikvision
    name: Porch
    host: 192.168.1.41
    port: 8889
    username: removed
    password: removed

  - platform: hikvision
    name: Driveway
    host: 192.168.1.40
    port: 8888
    username: removed
    password: removed

```
**Log:**
```
17-04-02 11:10:28 INFO (MainThread) [homeassistant.loader] Loaded binary_sensor.hikvision from homeassistant.components.binary_sensor.hikvision
17-04-02 11:10:28 INFO (MainThread) [homeassistant.bootstrap] Setting up binary_sensor
17-04-02 11:10:28 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.hikvision
17-04-02 11:10:28 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.hikvision
17-04-02 11:10:28 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_8) [pyhik.hikvision] Initializing new hikvision device at: http://192.168.1.41
17-04-02 11:10:28 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_4) [pyhik.hikvision] Initializing new hikvision device at: http://192.168.1.40
17-04-02 11:10:29 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_8) [pyhik.hikvision] Using alternate deviceInfo URL.
17-04-02 11:10:29 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_4) [pyhik.hikvision] Using alternate deviceInfo URL.
17-04-02 11:10:29 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_8) [pyhik.hikvision] Auto Namespace Result: http://www.hikvision.com/ver10/XMLSchema
17-04-02 11:10:29 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_8) [pyhik.hikvision] Using Namespace: http://www.hikvision.com/ver10/XMLSchema
17-04-02 11:10:29 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_4) [pyhik.hikvision] Auto Namespace Result: http://www.std-cgi.com/ver10/XMLSchema
17-04-02 11:10:30 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_4) [pyhik.hikvision] Using Namespace: http://www.std-cgi.com/ver10/XMLSchema
17-04-02 11:10:29 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_8) [pyhik.hikvision] Using alternate triggers URL.
17-04-02 11:10:30 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_4) [pyhik.hikvision] Using alternate triggers URL.
17-04-02 11:10:30 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_8) [pyhik.hikvision] Processed CAM Device.
17-04-02 11:10:30 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_8) [pyhik.hikvision] Found events: {'VMD': [1]}
17-04-02 11:10:30 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_4) [pyhik.hikvision] Processed CAM Device.
17-04-02 11:10:30 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_4) [pyhik.hikvision] Found events: {'VMD': [1]}
17-04-02 11:10:30 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_4) [pyhik.hikvision] Initialized Dictionary: {'Motion': [[False, 1, 0, datetime.datetime(2017, 4, 2, 11, 10, 30, 377829)]]}
17-04-02 11:10:30 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_8) [pyhik.hikvision] Initialized Dictionary: {'Motion': [[False, 1, 0, datetime.datetime(2017, 4, 2, 11, 10, 30, 362229)]]}
17-04-02 11:10:30 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_4) [homeassistant.components.binary_sensor.hikvision] Entity: Driveway - Motion, Options - Ignore: None, Delay: None
17-04-02 11:10:30 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_4) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Driveway Motion: off>> on 88.Motion.1
17-04-02 11:10:30 INFO (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=binary_sensor.driveway_motion, old_state=None, new_state=<state binary_sensor.driveway_motion=off; last_tripped_time=2017-04-02T04:10:30.377829-07:00, friendly_name=Driveway Motion, device_class=motion @ 2017-04-02T11:10:30.455829-07:00>>
17-04-02 11:10:30 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_8) [homeassistant.components.binary_sensor.hikvision] Entity: Porch - Motion, Options - Ignore: None, Delay: None
17-04-02 11:10:30 DEBUG (<concurrent.futures.thread.ThreadPoolExecutor object at 0x02A5E3B0>_8) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Porch Motion: off>> on 88.Motion.1
17-04-02 11:10:30 INFO (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=binary_sensor>
```

mezz64, 2017-04-03:
Issue here is with the ID returned by your cameras using the alternate URL's... they are both the same.

I made a commit to the altcam branch that should fix it, give it a shot.

Tim-GitHub, 2017-04-03:
Works correctly now, no more errors. Thanks again!

mezz64, 2017-04-06:
Updated version has been merged into HASS dev.  Closing this issue.

closed 2017-04-06 by mezz64