mezz64/pyHik #8: NVR DS-7108N-SN/P - No Events / NVR Connection Failed

macf0x, 2017-05-04:
`17-05-05 00:47:11 DEBUG (Thread-7) [pyhik.hikvision] Initializing new hikvision device at: http://192.168.10.21
17-05-05 00:47:11 DEBUG (Thread-7) [pyhik.hikvision] Using Namespace: http://www.isapi.org/ver20/XMLSchema
17-05-05 00:47:11 DEBUG (Thread-7) [pyhik.hikvision] Processed 48353836-3332-3131-3830-2857beef23b4 as NVR Device.
17-05-05 00:47:11 DEBUG (Thread-7) [pyhik.hikvision] Found events: {}
17-05-05 00:47:11 DEBUG (Thread-7) [pyhik.hikvision] No Events available in dictionary.
17-05-05 00:47:26 DEBUG (Thread-16) [pyhik.hikvision] Stream Thread Started: NVR, 48353836-3332-3131-3830-2857beef23b4
17-05-05 00:47:26 WARNING (Thread-16) [pyhik.hikvision] NVR Connection Failed. Waiting 10s. Err: Connection unsucessful.`

`http://x.x.x.x/ISAPI/System/deviceInfo

  <?xml version="1.0" encoding="UTF-8" ?> 
- <DeviceInfo version="1.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
  <deviceName>NVR</deviceName> 
  <deviceID>48353836-3332-3131-3830-2857beef23b4</deviceID> 
  <model>DS-7108N-SN/P</model> 
  <serialNumber>DS-7108N-SN/P0820160328AARR586321180WCVU</serialNumber> 
  <macAddress>28:57:be:ef:23:b4</macAddress> 
  <firmwareVersion>V3.0.13</firmwareVersion> 
  <firmwareReleasedDate>build 150503</firmwareReleasedDate> 
  <encoderVersion>V5.0</encoderVersion> 
  <encoderReleasedDate>build 150413</encoderReleasedDate> 
  <deviceType>IPC</deviceType> 
  <telecontrolID>255</telecontrolID> 
  </DeviceInfo>

`

mezz64, 2017-05-04:
Please post the XML you receive from either:
```http://XXX.XXX.XXX.XXX/ISAPI/Event/triggers`` or ```http://XXX.XXX.XXX.XXX/Event/triggers```

Your namespace url is not one i've seen before, so I assume this is a rebranded Hikvision NVR you are using?

macf0x, 2017-05-04:
Hi John,

It's a Chinese unit with English firmware.

I've posted the out put for the URL above.

Rob

mezz64, 2017-05-04:
Based on the triggers XML you provided it doesn't look like you have any of your events set to “Notify the surveillance center” as a trigger in your NVR/ setup.  To avoid having all events show up this library only searches for those that have that flag set.

macf0x, 2017-05-05:
```
<?xml version="1.0" encoding="UTF-8"?>
<EventTriggerList xmlns="http://www.isapi.org/ver20/XMLSchema" version="1.0">
<EventTrigger xmlns="http://www.isapi.org/ver20/XMLSchema" version="1.0">
<id>VMD-1</id>
<eventType>VMD</eventType>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList xmlns="http://www.isapi.org/ver20/XMLSchema" version="1.0">
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger xmlns="http://www.isapi.org/ver20/XMLSchema" version="1.0">
<id>VMD-2</id>
<eventType>VMD</eventType>
<dynVideoInputChannelID>2</dynVideoInputChannelID>
<EventTriggerNotificationList xmlns="http://www.isapi.org/ver20/XMLSchema" version="1.0">
<EventTriggerNotification>
<id>record-1</id>
<notificationMethod>record</notificationMethod>
<dynVideoInputID>1</dynVideoInputID>
</EventTriggerNotification>
<EventTriggerNotification>
<id>record-2</id>
<notificationMethod>record</notificationMethod>
<dynVideoInputID>2</dynVideoInputID>
</EventTriggerNotification>
<EventTriggerNotification>
<id>record-3</id>
<notificationMethod>record</notificationMethod>
<dynVideoInputID>3</dynVideoInputID>
</EventTriggerNotification>
<EventTriggerNotification>
<id>record-4</id>
<notificationMethod>record</notificationMethod>
<dynVideoInputID>4</dynVideoInputID>
</EventTriggerNotification>
<EventTriggerNotification>
<id>record-5</id>
<notificationMethod>record</notificationMethod>
<dynVideoInputID>5</dynVideoInputID>
</EventTriggerNotification>
<EventTriggerNotification>
<id>record-6</id>
<notificationMethod>record</notificationMethod>
<dynVideoInputID>6</dynVideoInputID>
</EventTriggerNotification>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger xmlns="http://www.isapi.org/ver20/XMLSchema" version="1.0">
<id>VMD-3</id>
<eventType>VMD</eventType>
<dynVideoInputChannelID>3</dynVideoInputChannelID>
<EventTriggerNotificationList xmlns="http://www.isapi.org/ver20/XMLSchema" version="1.0">
<EventTriggerNotification>
<id>record-1</id>
<notificationMethod>record</notificationMethod>
<dynVideoInputID>1</dynVideoInputID>
</EventTriggerNotification>
<EventTriggerNotification>
<id>record-2</id>
<notificationMethod>record</notificationMethod>
<dynVideoInputID>2</dynVideoInputID>
</EventTriggerNotification>
[... 209 lines cut ...]
<id>illaccess</id>
<eventType>illaccess</eventType>
<EventTriggerNotificationList xmlns="http://www.isapi.org/ver20/XMLSchema" version="1.0">
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger xmlns="http://www.isapi.org/ver20/XMLSchema" version="1.0">
<id>recordingfailure</id>
<eventType>recordingfailure</eventType>
<EventTriggerNotificationList xmlns="http://www.isapi.org/ver20/XMLSchema" version="1.0">
<EventTriggerNotification>
<id>email</id>
<notificationMethod>email</notificationMethod>
</EventTriggerNotification>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
</EventTriggerList>
```

macf0x, 2017-05-05:
Setting the events to notify center, got the binary_sensors discovered, but something isn't right with the NVR. The /ISAPI/Event/notification/alertStream always results in 404. I guess it's due to the hacked firmware. Nothing to do with the library, so I'll close  this.

Only contribution is an error about not liking the "recordingfailure" event type.

Anyway thanks for the tips. I've ordered another NVR which will hopefully address this issue.

closed 2017-05-05 by macf0x

mezz64, 2017-05-05:
You should give the ```/Event/notification/alertStream``` endpoint a try as well if you haven't already.  The library should already try that path automatically if the ISAPI path fails, but you never know.

macf0x, 2017-05-06:
I get a slightly different error

/ISAPI/Event/notification/alertStream - No XML just
```Can't locate the url```
/Event/notification/alertStream
```
<!DOCTYPE html>
<html><head><title>Document Error: Not Found</title></head>
<body><h2>Access Error: 404 -- Not Found</h2>
<p>Can't open document: /Event/notification/alertStream</p>
</body>
</html>
```