mezz64/pyHik #9: No attribute error

Tim-GitHub, 2017-06-24:
My HASS server died a few months ago and I finally replaced it with a Raspberry Pi 3. I got everything up and running except for pyHik. I used my exact same YAML configuration that was working fine up until my server died. Error log below seems to indicate an issue with finding the event. Not sure if pyHik has been updated and broke something for my cameras, or if I'm just forgetting something. Everything looks correct to me in my config.

```
#Hikvision Setup
binary_sensor:
  - platform: hikvision
    name: Porch
    host: 192.168.1.41
    port: 8889
    username: removed
    password: removed

  - platform: hikvision
    name: Driveway
    host: 192.168.1.40
    port: 8888
    username: removed
    password: removed
```
**Log:**

```
2017-06-24 11:07:23 DEBUG (Thread-4) [pyhik.hikvision] Initializing new hikvision device at: http://192.168.1.40
2017-06-24 11:07:23 DEBUG (Thread-9) [pyhik.hikvision] Initializing new hikvision device at: http://192.168.1.41
2017-06-24 11:07:23 DEBUG (Thread-9) [pyhik.hikvision] Using alternate deviceInfo URL.
2017-06-24 11:07:23 DEBUG (Thread-4) [pyhik.hikvision] Using alternate deviceInfo URL.
2017-06-24 11:07:23 DEBUG (Thread-9) [pyhik.hikvision] Using Namespace: http://www.hikvision.com/ver10/XMLSchema
2017-06-24 11:07:23 DEBUG (Thread-4) [pyhik.hikvision] Using Namespace: http://www.std-cgi.com/ver10/XMLSchema
2017-06-24 11:07:23 DEBUG (Thread-9) [pyhik.hikvision] Using alternate triggers URL.
2017-06-24 11:07:23 DEBUG (Thread-4) [pyhik.hikvision] Using alternate triggers URL.
2017-06-24 11:07:23 ERROR (Thread-9) [pyhik.hikvision] There was a problem finding an element: 'NoneType' object has no attribute 'text'
2017-06-24 11:07:23 DEBUG (Thread-9) [pyhik.hikvision] No Events available in dictionary.
2017-06-24 11:07:23 ERROR (Thread-4) [pyhik.hikvision] There was a problem finding an element: 'NoneType' object has no attribute 'text'
2017-06-24 11:07:23 DEBUG (Thread-4) [pyhik.hikvision] No Events available in dictionary.
2017-06-24 11:07:23 DEBUG (Thread-13) [pyhik.hikvision] Stream Thread Started: Front Door, fb5d3746-9a2b-4dc7-881f-b0d4e8e93094
2017-06-24 11:07:23 DEBUG (Thread-12) [pyhik.hikvision] Stream Thread Started: Driveway 8888, a5d8c037-c3da-4081-8320-cb3fed37af8c
2017-06-24 11:07:23 DEBUG (Thread-12) [pyhik.hikvision] Driveway 8888 Connection Successful.
2017-06-24 11:07:23 DEBUG (Thread-13) [pyhik.hikvision] Front Door Connection Successful.
```
***192.168.1.40 /events/triggers***

```
<?xml version="1.0" encoding="UTF-8"?>
-<EventTriggerList xmlns="http://www.std-cgi.com/ver10/XMLSchema" version="1.0">
-<EventTrigger xmlns="http://www.std-cgi.com/ver10/XMLSchema" version="1.0">
<id>1</id>
<eventType>VMD</eventType>
<eventDescription>VMD Event trigger Information</eventDescription>
-<EventTriggerNotificationList xmlns="http://www.std-cgi.com/ver10/XMLSchema" version="1.0">
-<EventTriggerNotification>
<id>3</id>
<notificationMethod>HTTP</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger xmlns="http://www.std-cgi.com/ver10/XMLSchema" version="1.0"> </EventTrigger>
<EventTrigger xmlns="http://www.std-cgi.com/ver10/XMLSchema" version="1.0"> </EventTrigger>
</EventTriggerList>
```

***192.168.1.41 /event/triggers***

```
  <?xml version="1.0" encoding="UTF-8" ?> 
- <EventTriggerList version="1.0" xmlns="http://www.hikvision.com/ver10/XMLSchema">
- <EventTrigger version="1.0" xmlns="http://www.hikvision.com/ver10/XMLSchema">
  <id>1</id> 
  <eventType>VMD</eventType> 
  <eventDescription>VMD Event trigger Information</eventDescription> 
- <EventTriggerNotificationList version="1.0" xmlns="http://www.hikvision.com/ver10/XMLSchema">
- <EventTriggerNotification>
  <id>3</id> 
  <notificationMethod>HTTP</notificationMethod> 
  <notificationRecurrence>beginning</notificationRecurrence> 
  </EventTriggerNotification>
  </EventTriggerNotificationList>
  </EventTrigger>
  <EventTrigger version="1.0" xmlns="http://www.hikvision.com/ver10/XMLSchema" /> 
  <EventTrigger version="1.0" xmlns="http://www.hikvision.com/ver10/XMLSchema" /> 
  </EventTriggerList>
```

mezz64, 2017-06-25:
Do you know what version you were on before? There haven't been any changes recently. I'll be unavailable for the next week or so, but I'll take a closer look when I get back.

Tim-GitHub, 2017-06-30:
I spun up a VM of a backup of my old server and verified that pyHik is working without issue on it. My old server was/is running Home Assistant 0.39.3. My Raspberry Pi 3 is running Home Assistant 0.47.1.

Tim-GitHub, 2017-06-30:
I updated my VM of my old server from 0.39.3 to current 0.47.1. Looks like pyHik stopped working, and gives the exact same errors:

```
[pyhik.hikvision] There was a problem finding an element: 'NoneType' object has no attribute 'text'
[pyhik.hikvision] No Events available in dictionary.
```

Based on that, it appears the issue is due to incompatibility of pyHik and updated code for Home Assistant.

mezz64, 2017-07-02:
Hass 39.3 was a while ago, back on pyHik 0.0.7, before NVR support was added.  There were a lot of changes to the event parsing code to accommodate that addition.

I'll try to run your xml through that section locally with some more debugging and see if I can locate the issue.

mezz64, 2017-07-07:
This should now be fixed in version 0.1.3 of pyHik.  You can either edit your local files manually to point to the new version or wait until it gets merged into HASS.  The PR to update the dev branch is currently pending.

closed 2017-07-27 by mezz64