mezz64/pyHik #10: DS-7608NI-E2 / A - Not working

mnj, 2017-07-26:
Using the latest version 0.1.3 i just get:

2017-07-26 23:14:25 WARNING (Thread-12) [pyhik.hikvision] None Connection Failed. Waiting 10s. Err: Connection unsucessful.
Exception in thread Thread-12:
Traceback (most recent call last):
  File "/home/homeassistant/.homeassistant/deps/pyhik/hikvision.py", line 358, in alert_stream
    raise ValueError('Connection unsucessful.')
ValueError: Connection unsucessful.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/home/homeassistant/.homeassistant/deps/pyhik/hikvision.py", line 410, in alert_stream
    self.watchdog.stop()
  File "/home/homeassistant/.homeassistant/deps/pyhik/watchdog.py", line 36, in stop
    self._timer.cancel()
AttributeError: 'Watchdog' object has no attribute '_timer'

Using a plain original DS-7608NI-E2 (V3.4.92 build 170228) with DS-2CD2142FWD-IS / DS-2CD2142FWD-I cameras, getting the above error no matter if i connect to the NVR or the cameras directly.

http://x.x.x.x/ISAPI/System/deviceInfo

<?xml version="1.0" encoding="UTF-8" ?>
<DeviceInfo version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
    <deviceName>Network Video Recorder</deviceName>
    <deviceID>48353633-3339-3230-3434-2857be9b7a78</deviceID>
    <model>DS-7608NI-E2/A</model>
    <serialNumber>DS-7608NI-E2/A0820151226AARR534392044WCVU</serialNumber>
    <macAddress>28:57:be:9b:7a:78</macAddress>
    <firmwareVersion>V3.4.92</firmwareVersion>
    <firmwareReleasedDate>build 170228</firmwareReleasedDate>
    <encoderVersion>V5.0</encoderVersion>
    <encoderReleasedDate>build 170228</encoderReleasedDate>
    <deviceType>IPC</deviceType>
    <telecontrolID>255</telecontrolID>
</DeviceInfo>

http://x.x.x.x/ISAPI/Event/triggers

<?xml version="1.0" encoding="UTF-8" ?>
<EventTriggerList version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
    <EventTrigger version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
        <id>IO-1</id>
        <eventType>IO</eventType>
        <inputIOPortID>1</inputIOPortID>
        <EventTriggerNotificationList version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema"></EventTriggerNotificationList>
    </EventTrigger>
    <EventTrigger version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
        <id>IO-2</id>
        <eventType>IO</eventType>
        <inputIOPortID>2</inputIOPortID>
        <EventTriggerNotificationList version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema"></EventTriggerNotificationList>
    </EventTrigger>
    <EventTrigger version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
        <id>IO-3</id>
        <eventType>IO</eventType>
        <inputIOPortID>3</inputIOPortID>
        <EventTriggerNotificationList version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema"></EventTriggerNotificationList>
    </EventTrigger>
    <EventTrigger version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
        <id>IO-4</id>
        <eventType>IO</eventType>
        <inputIOPortID>4</inputIOPortID>
        <EventTriggerNotificationList version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema"></EventTriggerNotificationList>
    </EventTrigger>
    <EventTrigger version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
[... 135 lines cut ...]
        <id>illaccess</id>
        <eventType>illaccess</eventType>
        <EventTriggerNotificationList version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema"></EventTriggerNotificationList>
    </EventTrigger>
    <EventTrigger version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
        <id>recordingfailure</id>
        <eventType>recordingfailure</eventType>
        <EventTriggerNotificationList version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
            <EventTriggerNotification>
                <id>beep</id>
                <notificationMethod>beep</notificationMethod>
            </EventTriggerNotification>
        </EventTriggerNotificationList>
    </EventTrigger>
</EventTriggerList>

Sending a GET request to http://xxxx/ISAPI/Event/notification/alertStream just never returns anything (it keeps waiting for a response) when testing with postman, sending an invalid request like PUT to it returns
<?xml version="1.0" encoding="UTF-8" ?>
<ResponseStatus version="1.0" xmlns="urn:psialliance-org">
    <requestURL>/ISAPI/Event/notification/alertStream</requestURL>
    <statusCode>4</statusCode>
    <statusString>Invalid Operation</statusString>
    <subStatusCode>methodNotAllowed</subStatusCode>
</ResponseStatus>
so the url does exist.

mezz64, 2017-07-26:
Ensure the user you are authenticating with has access to the notification center. Navigating to the alert stream url in a browser should produce a continually downloading file. Since the endpoint exists it looks like a permission issue to me.

mnj, 2017-07-26:
If i try with Curl that does work:

 curl -i -X GET -H "Authorization: Basic XXXXXXXXXXXXXXX=" http://xxxxxxx/ISAPI/Event/notification/alertStream
HTTP/1.0 200 OK
MIME-Version: 1.0
Content-Type: multipart/mixed;boundary=myboundary

--boundary
Content-Type: application/xml; charset="UTF-8"
Content-Length: 467

<EventNotificationAlert version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<ipAddress>xxxxxx</ipAddress>
<portNo>80</portNo>
<protocol>HTTP</protocol>
<macAddress>28:57:be:9b:7a:78</macAddress>
<channelID>0</channelID>
<dateTime>2017-07-26T23:52:28+01:00</dateTime>
<activePostCount>0</activePostCount>
<eventType>videoloss</eventType>
<eventState>inactive</eventState>
<eventDescription>videoloss alarm</eventDescription>
</EventNotificationAlert>
--boundary
Content-Type: application/xml; charset="UTF-8"
Content-Length: 467

<EventNotificationAlert version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<ipAddress>xxxxx</ipAddress>
<portNo>80</portNo>
<protocol>HTTP</protocol>
<macAddress>28:57:be:9b:7a:78</macAddress>
<channelID>0</channelID>
<dateTime>2017-07-26T23:52:32+01:00</dateTime>
<activePostCount>0</activePostCount>
<eventType>videoloss</eventType>
<eventState>inactive</eventState>
<eventDescription>videoloss alarm</eventDescription>
</EventNotificationAlert>

Using the same user/pass/ip with pyhik, it still generates the above error.

mezz64, 2017-07-27:
Was the curl command run on the same machine/container that hass is running on?  Can you ensure you have debug logging enabled for the pyhik component and then post a full startup log here?  I'm trying to see if there is any connectivity at all (events get discovered) or if it's just failing on the alert stream.

mnj, 2017-07-27:
Yep curl was ran from the same user/machine/ssh session.

The full log is on: https://pastebin.com/rsNT81up, but the relevant parts are probably:
```
2017-07-27 18:15:02 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.hikvision
2017-07-27 18:15:02 DEBUG (Thread-10) [pyhik.hikvision] Initializing new hikvision device at: http://xx.xx.xx.xx
2017-07-27 18:15:02 DEBUG (Thread-10) [pyhik.hikvision] Unable to fetch device info.
2017-07-27 18:15:02 ERROR (Thread-10) [homeassistant.components.binary_sensor.hikvision] Hikvision event stream has no data, unable to setup
2017-07-27 18:15:02 INFO (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=binary_sensor>
2017-07-27 18:15:02 DEBUG (Thread-12) [pyhik.hikvision] Stream Thread Started: None, None
2017-07-27 18:15:02 WARNING (Thread-12) [pyhik.hikvision] None Connection Failed. Waiting 10s. Err: Connection unsucessful.
Exception in thread Thread-12:
Traceback (most recent call last):
  File "/home/homeassistant/.homeassistant/deps/pyhik/hikvision.py", line 358, in alert_stream
    raise ValueError('Connection unsucessful.')
ValueError: Connection unsucessful.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/home/homeassistant/.homeassistant/deps/pyhik/hikvision.py", line 410, in alert_stream
    self.watchdog.stop()
  File "/home/homeassistant/.homeassistant/deps/pyhik/watchdog.py", line 36, in stop
    self._timer.cancel()
AttributeError: 'Watchdog' object has no attribute '_timer'

```

Curl from the same session:
```
(homeassistant) homeassistant@hass:~/.homeassistant/deps/pyhik$ curl -i -X GET -H "Authorization: Basic xxxxxxxxxxxx=" http://xxxxxx/ISAPI/Event/notification/alertStream
HTTP/1.0 200 OK
MIME-Version: 1.0
Content-Type: multipart/mixed;boundary=myboundary

--boundary
Content-Type: application/xml; charset="UTF-8"
Content-Length: 467

<EventNotificationAlert version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<ipAddress>xxxxxx</ipAddress>
<portNo>80</portNo>
<protocol>HTTP</protocol>
<macAddress>28:57:be:9b:7a:78</macAddress>
<channelID>0</channelID>
<dateTime>2017-07-27T18:26:14+01:00</dateTime>
<activePostCount>0</activePostCount>
<eventType>videoloss</eventType>
<eventState>inactive</eventState>
<eventDescription>videoloss alarm</eventDescription>
</EventNotificationAlert>
^C
(homeassistant) homeassistant@hass:~/.homeassistant/deps/pyhik$

```

The configuration is just straight from the docs / just starting to set up hass so it doesn't have much config yet:
```
binary_sensor:
  platform: hikvision
  host: xxxxx
  port: 80
  ssl: False
  username: homeassistant
  password: xxxxxx

```

So seems to trip really early on even just getting the deviceInfo (also works using curl).

mezz64, 2017-07-27:
Can you check to see if another hass component that utilizes the network functions correctly?  Maybe even try setting up a hass mjpeg camera to the hikvision nvr.  
It's failing right from the gate where nothing special is going on except requesting a URL so it seems network related despite the successful curl calls.

mnj, 2017-07-27:
Added some more logging statements, its tripping over:
```
 if response.status_code != 200:
            # If we didn't recieve 200, abort
            _LOGGING.debug('Unable to fetch device info.')
            return None

```
Its returning 401, doing a simple:
```
>>> import requests
>>> r = requests.get('http://xxxxxxx:80/ISAPI/System/deviceInfo', auth=('homeassistant', 'xxxxxxxxx'))
>>> r.status_code
401
```

Does the same, super strange when curl is fine in the exact same env/session, but ill dig around some more.

mnj, 2017-07-27:
Hmn this looks to be a bug with hikvision<>python requests somehow, ill close the issue seems unrelated to pyHik.

User: admin (level admin in the nvr), then it works from python, User: Homeassistant (cant select admin lvl, but it has all access rights ticked): doesn't work with python

Both work from curl and both work form my pc with Postman, i'm puzzled.

closed 2017-07-27 by mnj