mezz64/pyHik #11: Binary Sensors not being triggered

resoai, 2017-07-27:
Good day,

I have an DS-7608NI-K2/8P NVR with 5 cameras attached to it. I am able to see all sensors appear in homeassistant (strangely enough even the one I've removed previously) however none of them appear to be triggering any events. When I've enabled debug I can see that all callbacks are being setup properly but homeassistant loses connection to the NVR? 

``
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Initializing new hikvision device at: http://192.168.1.200
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Using Namespace: http://www.hikvision.com/ver20/XMLSchema
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Processed 48373635-3035-3033-3638-1868cb846fbf as NVR Device.
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Found events: {'videoloss': [1, 2, 3, 4, 5], 'fielddetection': [1, 2, 3, 4, 5], 'VMD': [3], 'tamperdetection': [1, 2, 3, 4, 5]}
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Initialized Dictionary: {'Tamper Detection': [[False, 1, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346371)], [False, 2, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346373)], [False, 3, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346376)], [False, 4, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346378)], [False, 5, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346379)]], 'Field Detection': [[False, 1, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346358)], [False, 2, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346360)], [False, 3, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346362)], [False, 4, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346364)], [False, 5, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346366)]], 'Video Loss': [[False, 1, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346342)], [False, 2, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346348)], [False, 3, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346350)], [False, 4, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346353)], [False, 5, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346355)]], 'Motion': [[False, 3, 0, datetime.datetime(2017, 7, 27, 11, 16, 20, 346369)]]}
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Tamper_Detection_1, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Tamper Detection 1: off>> on 48373635-3035-3033-3638-1868cb846fbf.Tamper Detection.1
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Tamper_Detection_2, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Tamper Detection 2: off>> on 48373635-3035-3033-3638-1868cb846fbf.Tamper Detection.2
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Tamper_Detection_3, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Tamper Detection 3: off>> on 48373635-3035-3033-3638-1868cb846fbf.Tamper Detection.3
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Tamper_Detection_4, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Tamper Detection 4: off>> on 48373635-3035-3033-3638-1868cb846fbf.Tamper Detection.4
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Tamper_Detection_5, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Tamper Detection 5: off>> on 48373635-3035-3033-3638-1868cb846fbf.Tamper Detection.5
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Field_Detection_1, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Field Detection 1: off>> on 48373635-3035-3033-3638-1868cb846fbf.Field Detection.1
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Field_Detection_2, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Field Detection 2: off>> on 48373635-3035-3033-3638-1868cb846fbf.Field Detection.2
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Field_Detection_3, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Field Detection 3: off>> on 48373635-3035-3033-3638-1868cb846fbf.Field Detection.3
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Field_Detection_4, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Field Detection 4: off>> on 48373635-3035-3033-3638-1868cb846fbf.Field Detection.4
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Field_Detection_5, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Field Detection 5: off>> on 48373635-3035-3033-3638-1868cb846fbf.Field Detection.5
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Video_Loss_1, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Video Loss 1: off>> on 48373635-3035-3033-3638-1868cb846fbf.Video Loss.1
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Video_Loss_2, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Video Loss 2: off>> on 48373635-3035-3033-3638-1868cb846fbf.Video Loss.2
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Video_Loss_3, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Video Loss 3: off>> on 48373635-3035-3033-3638-1868cb846fbf.Video Loss.3
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Video_Loss_4, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Video Loss 4: off>> on 48373635-3035-3033-3638-1868cb846fbf.Video Loss.4
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Video_Loss_5, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Video Loss 5: off>> on 48373635-3035-3033-3638-1868cb846fbf.Video Loss.5
2017-07-27 11:16:20 DEBUG (Thread-21) [homeassistant.components.binary_sensor.hikvision] Entity: Saturn NVR - Motion_3, Options - Ignore: None, Delay: None
2017-07-27 11:16:20 DEBUG (Thread-21) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Motion 3: off>> on 48373635-3035-3033-3638-1868cb846fbf.Motion.3
2017-07-27 11:16:25 DEBUG (Thread-25) [pyhik.hikvision] Stream Thread Started: Saturn NVR, 48373635-3035-3033-3638-1868cb846fbf
2017-07-27 11:16:25 WARNING (Thread-25) [pyhik.hikvision] Saturn NVR Connection Failed. Waiting 10s. Err: Connection unsucessful.
```

mezz64, 2017-07-27:
Check to make sure "Remote: Notify Surveillance Center" permission is enabled for the user you are connecting with and make sure you can access the alertstream path from your hass instance via curl:

```
curl -u USER:PASS http://YOURNVRIP/ISAPI/Event/notification/alertStream
```

resoai, 2017-07-27:
It turns out that camera username didn't have "Remote: Notify Surveillance Center" permission enabled. I have enabled it now and the error is gone! I will test if actual events are registered tonight once I'm back home.

resoai, 2017-07-27:
All events are being registered correctly. Sorry for the confusion!

mezz64, 2017-07-27:
Great, glad it's working for you.

closed 2017-07-27 by mezz64