mezz64/pyHik #12: binary sensor stop reporting requiring hass reboot

kucau0901, 2017-07-29:
hi , my DS-2CD2432F-IW binary sensor stop working after a day and some time after few triggers and needs HASS restart. the camera still trigger the cross line detection as i set the audible warning too. the only way to fix this is by rebooting HASS. im using latest HASS.IO. I  didnt see any error from HASS log.  if u search hass forum at least there is another user having same prob.

mezz64, 2017-07-30:
Enable debug logging for the pyhik library and post a log from startup until when you notice it's not working any longer.  There is a watchdog implemented to reset the connection if it should drop for some reason, but if the connection stays active, but the stream stops being updated from the camera I have no way to detect that.

Is the camera connected via a wired or wireless connection?

testermax, 2019-01-20:
I have identical problems with my 3 DS-2CD2532F-IWS over wifi:

In this log, 2 out of 3 cameras have stopped updating the binary sensor:

019-01-18 21:05:11 DEBUG (SyncWorker_11) [pyhik.hikvision] pyHik 0.1.9 initializing new hikvision device at: http://192.168.1.252
2019-01-18 21:05:11 DEBUG (SyncWorker_1) [pyhik.hikvision] pyHik 0.1.9 initializing new hikvision device at: http://192.168.1.254
2019-01-18 21:05:11 DEBUG (SyncWorker_6) [pyhik.hikvision] pyHik 0.1.9 initializing new hikvision device at: http://192.168.1.251
2019-01-18 21:05:11 DEBUG (SyncWorker_1) [pyhik.hikvision] Using Namespace: http://www.hikvision.com/ver20/XMLSchema
2019-01-18 21:05:11 DEBUG (SyncWorker_11) [pyhik.hikvision] Using Namespace: http://www.hikvision.com/ver20/XMLSchema
2019-01-18 21:05:11 DEBUG (SyncWorker_1) [pyhik.hikvision] Processed 5ced97f8-bb27-11b3-820e-006cfd17ff88 as CAM Device.
2019-01-18 21:05:11 DEBUG (SyncWorker_1) [pyhik.hikvision] Found events: {'VMD': [1], 'diskfull': [1], 'diskerror': [1], 'illaccess': [1], 'linedetection': [1], 'fielddetection': [1]}
2019-01-18 21:05:11 DEBUG (SyncWorker_1) [pyhik.hikvision] Initialized Dictionary: {'Motion': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 11, 663150)]], 'Disk Full': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 11, 663166)]], 'Disk Error': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 11, 663174)]], 'Illegal Access': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 11, 663183)]], 'Line Crossing': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 11, 663192)]], 'Field Detection': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 11, 663201)]]}
2019-01-18 21:05:11 DEBUG (SyncWorker_6) [pyhik.hikvision] Using Namespace: http://www.hikvision.com/ver20/XMLSchema
2019-01-18 21:05:11 DEBUG (SyncWorker_1) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Garage Disk Full: off>> on 5ced97f8-bb27-11b3-820e-006cfd17ff88.Disk Full.1
2019-01-18 21:05:11 DEBUG (SyncWorker_1) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Garage Disk Error: off>> on 5ced97f8-bb27-11b3-820e-006cfd17ff88.Disk Error.1
2019-01-18 21:05:11 DEBUG (SyncWorker_1) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Garage Illegal Access: off>> on 5ced97f8-bb27-11b3-820e-006cfd17ff88.Illegal Access.1
2019-01-18 21:05:11 DEBUG (SyncWorker_1) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Garage Line Crossing: off>> on 5ced97f8-bb27-11b3-820e-006cfd17ff88.Line Crossing.1
2019-01-18 21:05:11 DEBUG (SyncWorker_1) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Garage Field Detection: off>> on 5ced97f8-bb27-11b3-820e-006cfd17ff88.Field Detection.1
2019-01-18 21:05:11 DEBUG (SyncWorker_11) [pyhik.hikvision] Processed c08e578c-043b-11b4-826b-aca213d702f2 as CAM Device.
2019-01-18 21:05:11 DEBUG (SyncWorker_11) [pyhik.hikvision] Found events: {'VMD': [1], 'diskfull': [1], 'diskerror': [1], 'illaccess': [1], 'linedetection': [1], 'fielddetection': [1]}
2019-01-18 21:05:11 DEBUG (SyncWorker_11) [pyhik.hikvision] Initialized Dictionary: {'Motion': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 11, 864482)]], 'Disk Full': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 11, 864521)]], 'Disk Error': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 11, 864531)]], 'Illegal Access': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 11, 864540)]], 'Line Crossing': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 11, 864549)]], 'Field Detection': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 11, 864559)]]}
2019-01-18 21:05:12 DEBUG (SyncWorker_11) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Inngang Disk Full: off>> on c08e578c-043b-11b4-826b-aca213d702f2.Disk Full.1
2019-01-18 21:05:12 DEBUG (SyncWorker_11) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Inngang Disk Error: off>> on c08e578c-043b-11b4-826b-aca213d702f2.Disk Error.1
2019-01-18 21:05:12 DEBUG (SyncWorker_11) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Inngang Illegal Access: off>> on c08e578c-043b-11b4-826b-aca213d702f2.Illegal Access.1
2019-01-18 21:05:12 DEBUG (SyncWorker_11) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Inngang Line Crossing: off>> on c08e578c-043b-11b4-826b-aca213d702f2.Line Crossing.1
2019-01-18 21:05:12 DEBUG (SyncWorker_11) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Inngang Field Detection: off>> on c08e578c-043b-11b4-826b-aca213d702f2.Field Detection.1
2019-01-18 21:05:12 DEBUG (SyncWorker_6) [pyhik.hikvision] Processed 5d779be9-3b69-11b5-83f7-aca213d6af38 as CAM Device.
2019-01-18 21:05:12 DEBUG (SyncWorker_6) [pyhik.hikvision] Found events: {'VMD': [1], 'diskfull': [1], 'diskerror': [1], 'illaccess': [1], 'linedetection': [1], 'fielddetection': [1]}
2019-01-18 21:05:12 DEBUG (SyncWorker_6) [pyhik.hikvision] Initialized Dictionary: {'Motion': [[False, 1, 0, datetime.datetime(2019, 1, 18, 21, 5, 12, 755964)]], 'Disk Full': [[False, 1, 0, datetime.datetime(2019, 1, 18, 2 [... cut ...]
2019-01-20 16:45:18 DEBUG (Thread-3) [pyhik.hikvision] Inngang Update: Motion, [False, 1, 3942, datetime.datetime(2019, 1, 20, 16, 45, 18, 967163)]
2019-01-20 16:52:55 DEBUG (Thread-3) [pyhik.hikvision] Inngang Update: Motion, [True, 1, 3943, datetime.datetime(2019, 1, 20, 16, 52, 55, 650766)]
2019-01-20 16:53:12 DEBUG (Thread-3) [pyhik.hikvision] Updating stale event Motion on CH(1)
2019-01-20 16:53:12 DEBUG (Thread-3) [pyhik.hikvision] Inngang Update: Motion, [False, 1, 3978, datetime.datetime(2019, 1, 20, 16, 53, 12, 583328)]

testermax, 2019-03-15:
Unfortunately the version 0.2.2 together with home assistant v 0.89 is the most unstable yet, at least for my cameras (DS-2CD2532F-IWS). I was hoping that the reconnection logic solved my problem, but it seems that now the events arent recognized, eventhough I restart home assistant...

mezz64, 2019-03-27:
The underlying problem seems to be the library doesn't know the connection is lost and a result isn't even trying to reconnect.

Since I personally can't reproduce the issue it's very difficult for me to troubleshoot since I don't know if the connection is actually dropping or it stays active, but the data is getting garbled for some reason.

The only insight I can offer is to try in an ssh session on the HASS machine/container to start a curl output of the camera alertStream URL.  Leave it open for as long as possible and see if after some period of time the connection drops or if the data starts behaving differently.

Martin781, 2019-06-10:
Finally managed to catch what happens when Home Assistant stop detecting the sensor state.
This is what curl gets last time before the sensor stops working in HA:

```
Content-Type: application/xml; charset="UTF-8"
Content-Length: 1116

<EventNotificationAlert version="2.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<ipAddress>192.168.0.200</ipAddress>
<portNo>8250</portNo>                                                          i
<?xml version="1.0" encoding="UTF-8"?>
<ResponseStatus version="2.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<requestURL>/ISAPI/Event/notificati</requestURL>
<statusCode>4</statusCode>
<statusString>Invalid Operation</statusString>
<subStatusCode>invalidOperation</subStatusCode>
</ResponseStatus>
```

After this curl stops and drops back to the command line. Starting curl again right after it and it continues to read the alarm stream just fine.
I also run ping side by side with curl to see if the camera lost the wifi connection but it didn't.
I hope this helps solve this issue.

mezz64, 2019-06-10:
Are you sure you are feeding the correct URL into curl during that check?

The camera is returning a requestURL of `/ISAPI/Event/notificati` which is invalid.  The proper request path should be `/ISAPI/Event/notification/alertStream`

Martin781, 2019-06-11:
I must have messed up something after it stopped the alert stream. Probably pressed the the up key and run some previous command. So disregard that.

I tried again twice now. It seems it simply closes the connection:

`<RegionCoordinates>
<positionX>757</positionX>
<positionY>757</positionY>
</RegionCoordinates>
</RegionCoordinatesList>
</DetectionRegionEntry>
</DetectionRegionList>
<channelName>Kapu</channelName>
</EventNotificationAlert>
--boundary
Content-Type: application/xml; charset="UTF-8"
Content-Length: 1116

<EventNotificationAlert version="2.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<ipAddress>192.168.0.200</ipAddress>
<portNo>8250</portNo>
<protocol>HTTP</protocol>
<macAddress>98:8b:0a:33:65:a4</macAddress>
<channelID>1</channelID>
<dateTime>201
core-ssh:~#`

This is how it stopped first in mid read.

And this is the second time (ran it with the --verbose parameter this time):

`--boundary
Content-Type: application/xml; charset="UTF-8"
Content-Length: 1116

<EventNotificationAlert version="2.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<ipAddress>192.168.0.200</ipAddress>
<portNo>8250</portNo>
<protocol>HTTP</protocol>
<macAddress>98:8b:0a:33:65:a4</macAddress>
<channelID>1</channelID>
<dateTime>2019-06-11T11:13:27+01:00</dateTime>
<activePostCount>1654</activePostCount>
<eventType>VMD</eventType>
<eventState>active</eventState>
<eventDescription>Motion alarm</eventDescription>
<DetectionRegionList>
<DetectionRegionEntry>
<regionID>2</regionID>
<sensitivityLevel>34</sensitivityLevel>
<RegionCoordinatesList>
<RegionCoordinates>
<positionX>757</positionX>
<positionY>757</positionY>
</RegionCoordinates>
<RegionCoordinates>
<positionX>864</positionX>
<positionY>864</positionY>
</RegionCoordinates>
<RegionCoordinates>
<positionX>864</positionX>
<positionY>864</positionY>
</RegionCoordinates>
<RegionCoordinates>
<positionX>757</positionX>
<positionY>757</positionY>
</RegionCoordinates>
</RegionCoordinatesList>
</DetectionRegionEntry>
</DetectionRegionList>
<channelName>Kapu</channelName>
</EventNotificationAlert>
* Closing connection 0
core-ssh:~#`

mezz64, 2019-06-18:
If things work the way they are supposed to this shouldn't cause an issue.  The stream connection logic is in a while loop that should just keep trying to reconnect, spitting out a log message each time it does it.  

I'm afraid I don't have an easy answer or solution to this at this time.  It's something that is very difficult to debug when I can't produce it locally and is likely a result of some odd behavior in the requests library that is only showing itself in this specific scenario.

Martin781, 2019-06-25:
So I finally had some time I downloaded the project and tried to debug it in local Python environment.

I ran the basic_usage.py to see what's happening when it stops reporting in HA.

I got this error after a while:

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "c:\program files\python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "c:\program files\python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "c:\program files\python37\lib\site-packages\pyhik\hikvision.py", line 492, in alert_stream
    tree = ET.fromstring(parse_string)
  File "c:\program files\python37\lib\xml\etree\ElementTree.py", line 1315, in XML
    parser.feed(text)
  File "<string>", line None
xml.etree.ElementTree.ParseError: syntax error: line 1, column 0
```

This lead me to believe that the XML that is received somehow malformed (as you suspected) so I changed the hikvision.py where it handles the errors like so:

```
            except (ValueError,
                    requests.exceptions.ConnectionError,
                    requests.exceptions.ChunkedEncodingError,
                    ET.ParseError) as err:
                fail_count += 1
                reset_event.clear()
                _LOGGING.warning('%s Connection Failed (count=%d). Waiting %ss. Err: %s Parse XML: %s',
                                 self.name, fail_count, (fail_count * 5) + 5, err, parse_string)
```

So now it can handle the XML parse error and logs the XML contents so I can see how is it malformed.

After this it logged two different errors:
```

06/22/2019 07:57:02 PM - WARNING - Kapu Connection Failed (count=1). Waiting 10s. Err: syntax error: line 1, column 0 Parse XML:

--boundary
Content-Type: application/xml; charset="UTF-8"Content-Length: 516
<EventNotificationAlert version="2.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<ipAddress>192.168.0.200</ipAddress>
<portNo>8250</portNo>
<protocol>HTTP</protocol>
<macAddress>00:00:00:00:00:00</macAddress>
<channelID>1</channelID>
<dateTime>2019-06-22T19:56:55+01:00</dateTime>
<activePostCount>0</activePostCount>
<eventType>videoloss</eventType>
<eventState>inactive</eventState>
<eventDescription>videoloss alarm</eventDescription>
<channelName>Kapu</channelName>
</EventNotificationAlert>
```

```
06/22/2019 07:51:44 PM - WARNING - Kapu Connection Failed (count=1). Waiting 10s. Err: not well-formed (invalid token): line 1, column 638 Parse XML:

<EventNotificationAlert version="2.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<ipAddress>192.168.0.200</ipAddress>
<portNo>8250</portNo>
<protocol>HTTP</protocol>
<macAddress>98:8b:0a:33:65:a4</macAddress>
<channelID>1</channelID>
<dateTime>2019-06-22T19:51:38+01:00</dateTime>
<activePostCount>501</activePostCount>
<eventType>VMD</eventType>
<eventState>active</eventState>
<eventDescription>Motion alarm</eventDescription>
<DetectionRegionList>
<DetectionRegionEntry>
<regionID>3</regionID>
<sensitivityLevel>63</sensitivityLevel>
<RegionCoordinatesList>
<RegionCoordinates><positionX>146</positionX><positionY>146</p
--boundary
Content-Type: application/xml; charset="UTF-8"Content-Length: 1113
<EventNotificationAlert version="2.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<ipAddress>192.168.0.200</ipAddress>
<portNo>8250</portNo>
<protocol>HTTP</protocol>
<macAddress>98:8b:0a:33:65:a4</macAddress>
<channelID>1</channelID>
<dateTime>2019-06-22T19:51:42+01:00</dateTime>
<activePostCount>1</activePostCount>
<eventType>VMD</eventType>
<eventState>active</eventState>
<eventDescription>Motion alarm</eventDescription>
<DetectionRegionList>
<DetectionRegionEntry>
<regionID>3</regionID>
<sensitivityLevel>63</sensitivityLevel>
<RegionCoordinatesList>
<RegionCoordinates>
<positionX>146</positionX>
<positionY>146</positionY>
</RegionCoordinates>
<RegionCoordinates>
<positionX>231</positionX>
<positionY>231</positionY>
</RegionCoordinates>
<RegionCoordinates>
<positionX>231</positionX>
<positionY>231</positionY>
</RegionCoordinates>
<RegionCoordinates>
<positionX>146</positionX>
<positionY>146</positionY>
</RegionCoordinates>
</RegionCoordinatesList>
</DetectionRegionEntry>
</DetectionRegionList>
<channelName>Kapu</channelName>
</EventNotificationAlert>
```

I the first case as far as I can tell it doesn't supposed to contain the --boundary line that's why it errors out. In the second case it doesn't receive a complete XML it restarts in the middle hence the error.

Anyway I used this modified version in Home Assistant the last three days and it works like a charm. Previously it didn't worked for more than 24 hours without needing to restart it (usually much less than that).

Probably not the best way to handle the error since this way it reconnects even though it's just an XML error, but my programming skills ends here. :)

mezz64, 2019-06-26:
Thanks for doing some local tests, this is exactly the type of information I need to get a fix incorporated.  All this time I've been assuming it was a connection drop issue when it's actually bad XML that isn't getting processed correctly.

I'll think through the best way to handle this (re-establishing the connection isn't actually a bad idea) and get a new version up on pypi.  Once that's done you can point your local hass to the new library version to test.  If all goes well i'll submit a PR to hass and hopefully get one of the main complaints of this library solved.

testermax, 2019-06-29:
This is great news! I look forward to get this into Hass!

mezz64, 2019-07-04:
@Martin781 since you've already got a local test setup can you pull the changes in new xml_fix branch (https://github.com/mezz64/pyHik/tree/xml_fix) and run it for a while?  It should catch the issue you found and just log when it happens and keep on moving through the stream without having to reconnect.

Martin781, 2019-07-04:
Ran it for three hours. It worked fine. Catches the error and continues as it should. 

Full log here: [https://pastebin.com/8xBFY2NF](url)

mezz64, 2019-07-05:
Awesome.  Pypi has been updated and a PR is in to HASS to bump the version.

testermax, 2019-07-12:
I havent done the longterm test yet, byt version 0.2.3 is catching all the events Im expecting, from all cameras, in HASS. This looks really promising. Good job!

closed 2019-11-15 by mezz64