mezz64/pyHik #13: Hikvison IO sensor.

xydix, 2017-08-28:
Using this in Home-Assistant wont give me a sensor for alarm in from my Hikvision DS-2CD2135FWD-IS.
Firmware in camera: V5.4.5 170124

mezz64, 2017-08-29:
Please post the XML output from your ```http://x.x.x.x/ISAPI/Event/triggers``` endpoint.

xydix, 2017-09-01:
Hi. Sorry for the delay.
Here is the XML.
I have ordered an DS-2CD2185FWD-IS but haven't recevied it yet.
I will check this in that camera as well, but i guess it is the same because they have the same firmware.
Is there any other sensors beside I/O that isn't working with the hass component?

This XML file does not appear to have any style information associated with it. The document tree is shown below.
<EventNotification xmlns="http://www.hikvision.com/ver20/XMLSchema" version="2.0">
<EventTriggerList xmlns="http://www.hikvision.com/ver20/XMLSchema" version="2.0">
<EventTrigger>
<id>IO-1</id>
<eventType>IO</eventType>
<eventDescription>IO Event trigger Information</eventDescription>
<inputIOPortID>1</inputIOPortID>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
<EventTriggerNotification>
<id>record-1</id>
<notificationMethod>record</notificationMethod>
<videoInputID>1</videoInputID>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>VMD-1</id>
<eventType>VMD</eventType>
<eventDescription>VMD Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
<EventTriggerNotification>
<id>record-1</id>
<notificationMethod>record</notificationMethod>
<videoInputID>1</videoInputID>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
<EventTriggerNotification>
<id>beep</id>
<notificationMethod>beep</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
<EventTriggerNotification>
<id>monitorAlarm</id>
<notificationMethod>monitorAlarm</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
<EventTriggerNotification>
<id>FTP</id>
<notificationMethod>FTP</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>tamper-1</id>
<eventType>tamperdetection</eventType>
<eventDescription>shelteralarm Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
[... 119 lines cut ...]
<id>unattendedBaggage-1</id>
<eventType>unattendedBaggage</eventType>
<eventDescription>UnattendedBaggage Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>attendedBaggage-1</id>
<eventType>attendedBaggage</eventType>
<eventDescription>AttendedBaggage Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>storageDetection-1</id>
<eventType>storageDetection</eventType>
<eventDescription>storageDetection Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
</EventTriggerList>
</EventNotification>

mezz64, 2017-09-06:
I assume you are able to configure the IO trigger to alert the notification center?  It's not currently setup that way in the XML you posted.

xydix, 2017-09-06:
Yes i am. Do you want a new XML with I/O alert enabled?

mezz64, 2017-09-06:
No need for a new XML.  For your use case this should be an easy change. I'll shoot for implementing it and publishing a new release this week.

xydix, 2017-09-06:
Sounds great. You know if things like linecrossing will work?
I checked my DS-2CD2185FWD-IS btw and i seems to have the same XML. The share the same FW.
You have a way to set the motion, alarm O/I and other alarms on and off from Hass?

mezz64, 2017-09-06:
Line crossing is an existing supported notification type, if you set it to notify the surveillance center it should be picked up.

There is not currently support to actually turn the events on/off through the library.

mezz64, 2017-09-09:
IO support should now be working in v0.1.4 of the library.  PR has been submitted to HASS for the updates so changes should be available in dev soon.

closed 2017-09-15 by mezz64

xydix, 2017-09-15:
Thank you so much.