mezz64/pyHik #17: Alarm IO Channel for NVR

randellhodges, 2018-02-14:
The channel suffix for alarm IO is always 0.

There is code that looks for a node dynVideoInputChannelID or videoInputChannelID. Either it needs to also include a check for dynInputIOPortID, or the cast 'id' to int needs to strip the "eventType-" prefix from the id before attempting to cast to int.

Here is an example:

```
    <EventTrigger version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
        <id>IO-501</id>
        <eventType>IO</eventType>
        <dynInputIOPortID>501</dynInputIOPortID>
        <EventTriggerNotificationList version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
            <EventTriggerNotification>
                <id>record-5</id>
                <notificationMethod>record</notificationMethod>
                <dynVideoInputID>5</dynVideoInputID>
            </EventTriggerNotification>
            <EventTriggerNotification>
                <id>center</id>
                <notificationMethod>center</notificationMethod>
            </EventTriggerNotification>
        </EventTriggerNotificationList>
    </EventTrigger>
```

gives

`binary_sensor.nvr_io_0`

expecting

`binary_sensor_nvr_io_501`

mezz64, 2018-02-14:
Can you have multiple IO sensors per channel?  Does the "dynInputIOPortID" field show up during any of your other event types, or just IO events?

randellhodges, 2018-02-14:
When querying the NVR, dynInputIOPortID shows up once for each of the cameras that have an alarm. I am assuming *if* a camera supported multiple alarm outputs, you'd get one per camera.

I have 6 cameras. 5 support alarm outputs, the cameras 1 thru 5. 

I have: 

- IO-1 thru IO-16 for the DVR alarms. They have a node named inputIOPortID and they have values 1 thru 16.
- IO-101 thru IO-501 for the camera alarms. They have a node named dynInputIOPortID and they have values 101 thru 501.

I had just forked and started working on change (thought I'd try to be useful) that would just take the <id> and do a rsplit('-', 1)[-1] for use as the etchannel. I then saw how you use that as a flag for NVR, but I was going to use the DeviceInfo, deviceType to set that flag. I see values of IPCamera and DVR for that.

Not sure if what I was going to propose works well with rebranded stuff or not.

mezz64, 2018-02-14:
Can you post the full xml for your event triggers?

I'm hesitant to modify the id behavior for the exact reason you mention.  The only reason it's in there is to allow rebranded stuff to work.  I'd rather just add in another option, just want to make sure it doesn't interfere with any existing fields.

randellhodges, 2018-02-14:
Attached. Also included is what I was playing with for getting the id. Works for me, but I have all hikvision stuff. I left the nvr detection alone. 

There is some debug stuff in the stream processing. I'm trying to see how the io alerts trigger. Either I don't have the camera configured, or the triggering of IO doesn't go thru the stream. Still learning how that works.

[nvr_with_io.zip](https://github.com/mezz64/pyHik/files/1724646/nvr_with_io.zip)

closed 2018-02-28 by randellhodges