mezz64/pyHik #21: invalid syntax

nor500, 2018-03-09:
I am really interested in your project. I replaced my credentials in the example but python3 gave an error message: SyntaxError: invalid syntax
:
import pyhik.hikvision

camera = pyhik.hikvision.HikCamera('http://X.X.X.X', port=80, user='admin', pass='1234')

What should I do?

Ruslan1990, 2018-03-16:
@nor500 I have the same problem. Probably because of the word  `pass`

mezz64, 2018-03-16:
Take a look at the Home Assistant implementation (https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/binary_sensor/hikvision.py) if you want a full featured example.  The example in the readme is a bit out of date.  The actual init parameters are now:
```python
def __init__(self, host=None, port=DEFAULT_PORT,usr=None, pwd=None)
```

Ruslan1990, 2018-03-16:
@mezz64 
Thank you, now it works.
```
import pyhik.hikvision
camera = pyhik.hikvision.HikCamera('http://192.168.X.X', port=80, usr='admin', pwd='admin')
print camera
```
`<pyhik.hikvision.HikCamera object at 0x7ff3a8ea0850>`

Home-assistant, this is a cool project, but for me it's difficult, although I managed to run it.
[image]

How can I get easy access to the motion detection status using pyhik, for use in my project?