mezz64/pyHik #22: How can I get messages about the movement?

Ruslan1990, 2018-03-19:
How can I get messages about the movement?
What do you need to add to this code?
`import pyhik.hikvision
camera = pyhik.hikvision.HikCamera('http://192.168.X.X', port=80, usr='admin', pwd='admin')`

mezz64, 2018-03-23:
Sorry guys i'm a bit busy these days and don't have much time to dedicate to this project beyond bug fixes.  Take a look at the Home Assistant implementation (https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/binary_sensor/hikvision.py) if you want a full featured example.  It covers how to setup a connection and a sensor class with callbacks to receive the events.

AhSem, 2018-04-16:
Hi @kfowlks, have you figured out how to get the motion events in real time?