mezz64/pyHik #24: triggering manual recording

kockockockoc, 2018-06-04:
http://down.dipol.com.pl/Cctv/-Hikvision-/isapi/HIKVISION%20ISAPI_2.0-RaCM%20Service.pdf

7.4.1 /ISAPI/ContentMgmt/record/control/manual/start/tracks/
This resource is used to manually Start the recording track, regardless of recording mode.
To Enable or Disable (i.e. permanent Stop) the track, the configuration interface should be used to update the track configuration object to set the enable/disable value accordingly.

7.4.2 /ISAPI/ContentMgmt/record/control/manual/stop/tracks/

kockockockoc, 2018-06-04:
Just tested on HiWatch(HikVision) DVR DS-H204Q:
curl -X PUT “http://user:passwd@192.168.1.28:80/ISAPI/ContentMgmt/record/control/manual/start/track/101”

Works fine for me

mezz64, 2018-06-05:
Seems to only work for NVR/DVR devices.