mezz64/pyHik #26: DVR integration

mastroantonio, 2018-11-19:
I believe the error is related to the server's response
http://xxx.xxx.x.xxx/ISAPI/System/deviceInfo.
I am attaching a screen of the answer.
The deviceType entry is DVR

[image]

mezz64, 2018-11-19:
Please add debug logging lines to your hass config for pyhik and post the output from a HASS start when you have the DVR configured.  

```
logs:
   pyhik.hikvision: debug
   homeassistant.components.binary_sensor.hikvision: debug
```

I don't perform any setup based on the deviceType field so that alone shouldn't be preventing things from working.  While you're at it post the output of your triggers endpoint and verify you get a continuously downloading file when you try to go to the alertStream url in a browser.  Unfortunately, many of these devices have only a partial API and the alertStream is left out.

mezz64, 2018-12-17:
I'm afraid I don't have any api documentation on what type of stream the "alarm center" reports.  If you can find some documentation and/or provide a sample from the stream showing some events being triggered I can look into adding it as an option.

madprunes, 2019-10-03:
I had a similar problem initially with my DVR/NVR there was no alertStream, updating to the latest USA firmware (as the others were still too old) gave me the alertStream.

As far as alarm host goes, that is a HTTP server which the NVR/DVR posts it's event messages to, I believe that is what it is explaining on page 344 of the HIKVISION ISAPI_2.6-IPMD Service.pdf document.

closed 2019-11-15 by mezz64