mezz64/pyHik #34: How to enable any kind of debug?

wojciiii, 2019-02-10:
Would you be so kind and update the wiki or documentation about how to enable enough debug to be able to determine why the lib is not working (in my case I try to register for "fielddetection" events from a NVR but get no events ) as expected? I tried setting debug level using:
_LOGGING.setLevel(logging.DEBUG)
but this had no effect at all.

mezz64, 2019-02-11:
The library will attach to whatever logger you define in your program.  As an example:
`logging.basicConfig(filename='out.log', filemode='w', level=logging.DEBUG, format='%(asctime)s - %(levelname)s - %(message)s', datefmt='%m/%d/%Y %I:%M:%S %p')`

I will also add a basic usage code example to github.

mezz64, 2019-02-11:
See here for a full example of usage:
https://github.com/mezz64/pyHik/blob/master/examples/basic_usage.py

closed 2019-02-11 by mezz64