mezz64/pyHik #38: Alert Stream: enable expert mode

erudita, 2019-04-24:
Hi mezz64,
First of all thanks for the work on this module.

I use it primarily as a binary sensor in home assistant, and see a use for creating (optionally) multiple motion sensors per camera if the configuration is set to "expert mode".

I believe the starting point to validate the camera capability using the ```/ISAPI/System/Video/inputs/channels/1/motionDetectionExt``` call to get marker XML ```<activeMode>expert</activeMode>```
adding to the API for "extended motion".

Do you see value in investigating this path?

Regards

mezz64, 2019-04-29:
It could be a worthwhile addition.  I'll need to play with the expert mode on my own cams to make sure I can debug locally.  Not sure when i'll have time to get to it, but i'll leave this issue open as a reminder.

erudita, 2019-05-04:
Ok, thanks. Time is always the enemy!
Let me know if you would like any assistance. I have written some routines consistent with PyHik which grab the data.

mezz64, 2019-05-13:
I'm always open to PR's.  Feel free to submit one with your changes and i'll make some time to review it.

laszlojakab, 2021-09-06:
@mezz64: could you please review PR for this issue?

closed 2026-09-01 by mezz64