mezz64/pyHik #41: NVR IO ports no channelID

madprunes, 2019-09-22:
Unsure which devices this affects, but I have it on the DS-9016HFI-ST (Running V3.4.5 build 170518)

channeIlD is not used on the IO ports, reading through the spec (HIKVISION ISAPI_2.6-IPMD Service.pdf) that tag is only used on camera events.

We should be easily able to deal with it by using the inputIOPortID tag or dynInputIOPortID if channelID is still none.

I would think putting this in under line 548 of hikvision.py would fix it.
--------------------------------------
            if echid is None:
                # channelID doesnt exist for io ports so we use inputIOPortID instead
                echid = tree.find(
                    self.element_query('inputIOPortID'))
            if echid is None:
                # channelID doesnt exist for io ports so we use inputIOPortID instead
                echid = tree.find(
                    self.element_query('dynInputIOPortID'))
----------------------------------------

madprunes, 2019-09-25:
I have now tested this with the basic_usage.py script.
Here is some an xml message for your viewing pleasure

--boundary
Content-Type: application/xml; charset="UTF-8"
Content-Length: 460

<EventNotificationAlert version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<ipAddress>192.168.2.6</ipAddress>
<portNo>80</portNo>
<protocol>HTTP</protocol>
<macAddress>c4:2f:90:95:30:17</macAddress>
<dateTime>2019-09-25T17:45:57+10:00</dateTime>
<activePostCount>2</activePostCount>
<eventType>IO</eventType>
<eventState>active</eventState>
<eventDescription>IO alarm</eventDescription>
<inputIOPortID>16</inputIOPortID>
</EventNotificationAlert>
--boundary

My python and git knowledge are fairly limited, I'm currently trying to figure it all out to give you a pull request.

madprunes, 2019-09-25:
I also noticed previous issues raised relating to this, and it appears it was implemented for the discovery but not for the alertStream parsing.

mezz64, 2019-09-25:
Thanks for the report.  I'm a little tied up with other things right now, but will try to take a look soon.

madprunes, 2019-09-26:
I have now put a custom component into home assistant which is a copy of the hikvision one which is importing my modified pyhik and it appears to have inputs working correctly.

mezz64, 2019-09-30:
Please grab the updated hikvision.py from the new id_fix branch (https://github.com/mezz64/pyHik/tree/id_fix) and test it please.  This should fix your issue and make it easier to add other fields in the future.

madprunes, 2019-09-30:
Will do in a couple of days as I'm away for work.

madprunes, 2019-10-03:
I have put the new hikvision.py in and it appears to be working as expected.

mezz64, 2019-10-08:
Thanks for the update.  I'll try to get this moving this week.

mezz64, 2019-10-12:
Released 0.2.4 with this fix to pypi and submitted a PR to bump the version in HASS.

closed 2019-10-12 by mezz64