mezz64/pyHik #44: TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'

alexxxxey, 2019-11-10:
I got this "parse_strnig" from cam:
> <EventNotificationAlert version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
    <ipAddress>192.168.1.99</ipAddress>
    <portNo>81</portNo>
    <protocolType>HTTP</protocolType>
    <macAddress>98:8b:aa:a8:7c:5d</macAddress>
    <channelID/>
    <dynChannelID/>
    <dateTime>2019-11-11T01:38:49+02:00</dateTime>
    <activePostCount>0</activePostCount>
    <eventType>videoloss</eventType>
    <eventState>inactive</eventState>
    <eventDescription>videoloss alarm</eventDescription>
    <channelName/>
</EventNotificationAlert>

and got error:

> hikvision.py", line 553, in process_stream
    echid = int(echid.text)
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'

start discovering this code from hikvision.py
> echid = tree.find(self.element_query(idtype))
    if echid is not None:
        try:
            # Need to make sure this is actually a number
            echid = int(echid.text)
            break
        except ValueError:
           # Field must not be an integer
           pass

and notice, that echid.text can be empty. In this case we get TypeError, not ValueError. I can add except TypeError also, or just change:

> if echid is not None and echid.text is not None:

but i don't know, if it will be right way and didn't break functionality later.
What is the right way? 
Thanks!

mezz64, 2019-11-15:
Is this a hikvision cam or a rebadge?  In any case accounting for blank fields is certainly a good idea and should probably be captured in some other spots as well.  I'll find some time to fix this soon.

mezz64, 2019-11-16:
I've merged in a fix for the blank channel ID field and released a new version.  A PR is pending to update the repository in HASS but you can manually specify 0.2.5 to check it now if you'd like.

alexxxxey, 2019-11-18:
> Is this a hikvision cam or a rebadge? In any case accounting for blank fields is certainly a good idea and should probably be captured in some other spots as well. I'll find some time to fix this soon.

it is original hikvision. Thank you for the fix, i will try it later.

closed 2020-02-13 by mezz64