mezz64/pyHik #45: binary_sensors not being updated

resoai, 2019-11-14:
Unfortunately I have recently stopped receiveiving updates for any binary_sensors in HomeASsistant. last_tripped_time is always of the last boot time. I've added logging as follows:

```
  logs:
   pyhik.hikvision: debug
   homeassistant.components.binary_sensor.hikvision: debug
```

I don't see anything particularly wrong except maybe for last two lines (not sure if this is an expenses behaviour):

```
2019-11-14 16:03:07 DEBUG (SyncWorker_5) [pyhik.hikvision] pyHik 0.2.4 initializing new hikvision device at: http://192.168.2.200
2019-11-14 16:03:07 DEBUG (SyncWorker_5) [pyhik.hikvision] Using Namespace: http://www.isapi.org/ver20/XMLSchema
2019-11-14 16:03:07 DEBUG (SyncWorker_13) [pyhik.hikvision] pyHik 0.2.4 initializing new hikvision device at: http://192.168.1.200
2019-11-14 16:03:07 DEBUG (SyncWorker_13) [pyhik.hikvision] Using Namespace: http://www.isapi.org/ver20/XMLSchema
2019-11-14 16:03:07 DEBUG (SyncWorker_5) [pyhik.hikvision] Processed 48373635-3035-3034-3032-1868cb846fe1 as NVR Device.
2019-11-14 16:03:07 DEBUG (SyncWorker_5) [pyhik.hikvision] Found events: {'tamperdetection': [1, 2, 3, 4, 5], 'videoloss': [1, 2, 4, 5, 6], 'linedetection': [1, 2, 3, 4, 5], 'diskfull': [0], 'diskerror': [0], 'illaccess': [0], 'recordingfailure': [0]}
2019-11-14 16:03:07 DEBUG (SyncWorker_5) [pyhik.hikvision] Initialized Dictionary: {'Tamper Detection': [[False, 1, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733207)], [False, 2, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733218)], [False, 3, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733223)], [False, 4, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733227)], [False, 5, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733232)]], 'Video Loss': [[False, 1, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733238)], [False, 2, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733243)], [False, 4, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733248)], [False, 5, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733252)], [False, 6, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733257)]], 'Line Crossing': [[False, 1, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733262)], [False, 2, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733267)], [False, 3, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733272)], [False, 4, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733276)], [False, 5, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733280)]], 'Disk Full': [[False, 0, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733286)]], 'Disk Error': [[False, 0, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733292)]], 'Illegal Access': [[False, 0, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733298)]], 'Recording Failure': [[False, 0, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 733304)]]}
2019-11-14 16:03:07 DEBUG (SyncWorker_13) [pyhik.hikvision] Processed 48373635-3035-3033-3638-1868cb846fbf as NVR Device.
2019-11-14 16:03:07 DEBUG (SyncWorker_13) [pyhik.hikvision] Found events: {'tamperdetection': [1, 2, 3, 4, 5], 'linedetection': [1, 2, 3, 4, 5], 'diskfull': [0], 'diskerror': [0], 'illaccess': [0]}
2019-11-14 16:03:07 DEBUG (SyncWorker_13) [pyhik.hikvision] Initialized Dictionary: {'Tamper Detection': [[False, 1, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767031)], [False, 2, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767042)], [False, 3, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767048)], [False, 4, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767052)], [False, 5, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767057)]], 'Line Crossing': [[False, 1, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767063)], [False, 2, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767068)], [False, 3, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767072)], [False, 4, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767077)], [False, 5, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767081)]], 'Disk Full': [[False, 0, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767087)]], 'Disk Error': [[False, 0, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767092)]], 'Illegal Access': [[False, 0, 0, datetime.datetime(2019, 11, 14, 16, 3, 7, 767098)]]}
2019-11-14 16:03:08 DEBUG (SyncWorker_5) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Neptune NVR Tamper Detection 1: off>> on 48373635-3035-3034-3032-1868cb846fe1.Tamper Detection.1
2019-11-14 16:03:08 DEBUG (SyncWorker_5) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Neptune NVR Tamper Detection 2: off>> on 48373635-3035-3034-3032-1868cb846fe1.Tamper Detection.2
2019-11-14 16:03:08 DEBUG (SyncWorker_5) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Neptune NVR Tamper Detection 3: off>> on 48373635-3035-3034-3032-1868cb846fe1.Tamper Detection.3
2019-11-14 16:03:08 DEBUG (SyncWorker_13) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Tamper Detection 1: off>> on 48373635-3035-3033-3638-1868cb846fbf.Tamper Detection.1
2019-11-14 16:03:08 DEBUG (SyncWorker_5) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Neptune NVR Tamper Detection 4: off>> on 48373635-3035-3034-3032-1868cb846fe1.Tamper Detection.4
2019-11-14 16:03:08 DEBUG (SyncWorker_13) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Tamper Detection 2: off>> on 48373635-3035-3033-3638-1868cb846fbf.Tamper Detection.2
2019-11-14 16:03:08 DEBUG (SyncWorker_5) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Neptune NVR Tamper Detection 5: off>> on 48373635-3035-3034-3032-1868cb846fe1.Tamper Detection.5
2019-11-14 16:03:08 DEBUG (SyncWorker_13) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Tamper Detection 3: off>> on 48373635-3035-3033-3638-1868cb846fbf.Tamper Detection.3
2019-11-14 16:03:08 DEBUG (SyncWorker_5) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Neptune NVR Video Loss 1: off>> on 48373635-3035-3034-3032-1868cb846fe1.Video Loss.1
2019-11-14 16:03:08 DEBUG (SyncWorker_13) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Tamper Detection 4: off>> on 48373635-3035-3033-3638-1868cb846fbf.Tamper Detection.4
2019-11-14 16:03:08 DEBUG (SyncWorker_5) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Neptune NVR Video Loss 2: off>> on 48373635-3035-3034-3032-1868cb846fe1.Video Loss.2
2019-11-14 16:03:08 DEBUG (SyncWorker_13) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Tamper Detection 5: off>> on 48373635-3035-3033-3638-1868cb846fbf.Tamper Detection.5
2019-11-14 16:03:08 DEBUG (SyncWorker_5) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Neptune NVR Video Loss 4: off>> on 48373635-3035-3034-3032-1868cb846fe1.Video Loss.4
2019-11-14 16:03:08 DEBUG (SyncWorker_13) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Line Crossing 1: off>> on 48373635-3035-3033-3638-1868cb846fbf.Line Crossing.1
2019-11-14 16:03:08 DEBUG (SyncWorker_5) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Neptune NVR Video Loss 5: off>> on 48373635-3035-3034-3032-1868cb846fe1.Video Loss.5
2019-11-14 16:03:08 DEBUG (SyncWorker_13) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Line Crossing 2: off>> on 48373635-3035-3033-3638-1868cb846fbf.Line Crossing.2
2019-11-14 16:03:08 DEBUG (SyncWorker_5) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Neptune NVR Video Loss 6: off>> on 48373635-3035-3034-3032-1868cb846fe1.Video Loss.6
2019-11-14 16:03:08 DEBUG (SyncWorker_13) [pyhik.hikvision] Added update callback to <bound method HikvisionBinarySensor._update_callback of <Entity Saturn NVR Line Crossing 3: off>> on 48373635-3035-3033-3638-1868cb846fbf.Line Crossing.3
[... cut ...]
2019-11-14 16:03:12 DEBUG (Thread-7) [pyhik.hikvision] Stream Thread Started: Saturn NVR, 48373635-3035-3033-3638-1868cb846fbf
2019-11-14 16:03:12 DEBUG (Thread-6) [pyhik.hikvision] Neptune NVR Connection Successful.
2019-11-14 16:03:12 DEBUG (Thread-7) [pyhik.hikvision] Saturn NVR Connection Successful.
....................................

2019-11-14 16:08:12 DEBUG (Thread-9) [pyhik.hikvision] Neptune NVR Watchdog expired. Resetting connection.
2019-11-14 16:08:12 DEBUG (Thread-10) [pyhik.hikvision] Saturn NVR Watchdog expired. Resetting connection.
```

mezz64, 2019-11-15:
Have you done any firmware updates on your NVR's recently?  If you force an event shortly after starting up the component does it show up?  Having the watchdog trigger on both streams at the same time may be a network error on the HASS side of things since it's extremely unlikely for both streams to drop at the same time.

resoai, 2019-11-15:
I'm afraid both NVRs have actually been updated recently to newer firmware.I will try to test if events are coming through after restart once I manage to get back home. I will also try to debug eventStream to see what exactly is coming through.

resoai, 2019-11-15:
I have also started seeing following in the logs after reboot:

```
2019-11-15 12:06:13 DEBUG (Thread-6) [pyhik.hikvision] Stream Thread Started: Saturn NVR, 48373635-3035-3033-3638-186
8cb846fbf                                                                                                            
2019-11-15 12:06:13 DEBUG (Thread-7) [pyhik.hikvision] Stream Thread Started: Neptune NVR, 48373635-3035-3034-3032-18
68cb846fe1                                                                                                           
2019-11-15 12:06:13 DEBUG (Thread-6) [pyhik.hikvision] Saturn NVR Connection Successful.                             
2019-11-15 12:06:13 DEBUG (Thread-7) [pyhik.hikvision] Neptune NVR Connection Successful.                            
Exception in thread Thread-6:                                                                                        
Traceback (most recent call last):                                                                                   
  File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner                                        
    self.run()                                                                                                       
  File "/usr/local/lib/python3.7/threading.py", line 870, in run                                                     
    self._target(*self._args, **self._kwargs)                                                                        
  File "/usr/local/lib/python3.7/site-packages/pyhik/hikvision.py", line 497, in alert_stream                        
    self.process_stream(tree)                                                                                        
  File "/usr/local/lib/python3.7/site-packages/pyhik/hikvision.py", line 552, in process_stream                      
    echid = int(echid.text)                                                                                          
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'                          
                                                                                                                     
Exception in thread Thread-7:                                                                                        
Traceback (most recent call last):                                                                                   
  File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner                                        
    self.run()                                                                                                       
  File "/usr/local/lib/python3.7/threading.py", line 870, in run                                                     
    self._target(*self._args, **self._kwargs)                                                                        
  File "/usr/local/lib/python3.7/site-packages/pyhik/hikvision.py", line 497, in alert_stream                        
    self.process_stream(tree)                                                                                        
  File "/usr/local/lib/python3.7/site-packages/pyhik/hikvision.py", line 552, in process_stream                      
    echid = int(echid.text)                                                                                          
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'  
```

resoai, 2019-11-15:
```
self.watchdog = Watchdog(300.0, self.watchdog_handler)
```
Watchdog appears to be expiring exactly 5 mins because `process_stream` never gets to `self.watchdog.pet()` if I understand correctly.

mezz64, 2019-11-15:
Alright, it looks like a recent firmware change may have altered the behavior on reporting the channel ID's in the event stream.  If you're getting the TypeError every time the watchdog expiration and lack of events makes sense.  

Just to be thorough, please try curling the event stream for one of the NVR's to make sure it's still sending out events.

resoai, 2019-11-15:
It does look plausable indeed. I did leave terminal open for about 30 mins and I'm getting a steady flow of events, mostly `videoloss` type:

```
--boundary
Content-Type: application/xml; charset="UTF-8"
Content-Length: 531

<?xml version="1.0" encoding="utf-8"?>
<EventNotificationAlert version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<ipAddress>192.168.1.200</ipAddress>
<portNo>80</portNo>
<protocolType>HTTP</protocolType>
<macAddress>18:68:cb:84:6f:bf</macAddress>
<channelID/>
<dynChannelID/>
<dateTime>2019-11-15T14:18:0600:00</dateTime>
<activePostCount>0</activePostCount>
<eventType>videoloss</eventType>
<eventState>inactive</eventState>
<eventDescription>videoloss alarm</eventDescription>
<channelName/>
</EventNotificationAlert>
```

mezz64, 2019-11-16:
I've merged in a fix for the blank channel ID field and released a new version.  A PR is pending to update the repository in HASS but you can manually specify 0.2.5 to check it now if you'd like.

resoai, 2019-11-17:
Thank you. Now I'm getting following error:

```
Exception in thread Thread-7:                                                                                        
Traceback (most recent call last):                                                                                   
  File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner                                        
    self.run()                                                                                                       
  File "/usr/local/lib/python3.7/threading.py", line 870, in run                                                     
    self._target(*self._args, **self._kwargs)                                                                        
  File "/usr/local/lib/python3.7/site-packages/pyhik/hikvision.py", line 497, in alert_stream                        
    self.process_stream(tree)                                                                                        
  File "/usr/local/lib/python3.7/site-packages/pyhik/hikvision.py", line 570, in process_stream                      
    state = self.fetch_attributes(etype, echid)                                                                      
  File "/usr/local/lib/python3.7/site-packages/pyhik/hikvision.py", line 618, in fetch_attributes                    
    if sensor[1] == int(channel):                                                                                    
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'                          
                                                                                                                     
Exception in thread Thread-6:                                                                                        
Traceback (most recent call last):                                                                                   
  File "/usr/local/lib/python3.7/threading.py", line 926, in _bootstrap_inner                                        
    self.run()                                                                                                       
  File "/usr/local/lib/python3.7/threading.py", line 870, in run                                                     
    self._target(*self._args, **self._kwargs)                                                                        
  File "/usr/local/lib/python3.7/site-packages/pyhik/hikvision.py", line 497, in alert_stream                        
    self.process_stream(tree)                                                                                        
  File "/usr/local/lib/python3.7/site-packages/pyhik/hikvision.py", line 570, in process_stream                      
    state = self.fetch_attributes(etype, echid)                                                                      
  File "/usr/local/lib/python3.7/site-packages/pyhik/hikvision.py", line 618, in fetch_attributes                    
    if sensor[1] == int(channel):                                                                                    
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType' 
```

mezz64, 2019-11-17:
Ok, I'm going to need samples from the alert stream from a couple different types of events. If the IDs are missing from everything it's not as straight forward to fix.

resoai, 2019-11-17:
Thank you for taking time to look into this. I can see that line crossing has `channelID` and `channelName`:

```
--boundary
Content-Type: application/xml; charset="UTF-8"
Content-Length: 792

<?xml version="1.0" encoding="utf-8"?>
<EventNotificationAlert version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<ipAddress>192.168.1.200</ipAddress>
<portNo>80</portNo>
<protocolType>HTTP</protocolType>
<macAddress>18:68:cb:84:6f:bf</macAddress>
<channelID>5</channelID>
<dynChannelID>5</dynChannelID>
<dateTime>2019-11-17T17:53:4200:00</dateTime>
<activePostCount>4</activePostCount>
<eventType>linedetection</eventType>
<eventState>active</eventState>
<eventDescription>linedetection alarm</eventDescription>
<channelName>Side Passage</channelName>
<DetectionRegionList>
<DetectionRegionEntry>
<regionID>0</regionID>
<sensitivityLevel>0</sensitivityLevel>
<RegionCoordinatesList>
</RegionCoordinatesList>
</DetectionRegionEntry>
</DetectionRegionList>
</EventNotificationAlert>
--boundary
Content-Type: application/xml; charset="UTF-8"
Content-Length: 531

<?xml version="1.0" encoding="utf-8"?>
<EventNotificationAlert version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<ipAddress>192.168.1.200</ipAddress>
<portNo>80</portNo>
<protocolType>HTTP</protocolType>
<macAddress>18:68:cb:84:6f:bf</macAddress>
<channelID/>
<dynChannelID/>
<dateTime>2019-11-17T17:53:5100:00</dateTime>
<activePostCount>0</activePostCount>
<eventType>videoloss</eventType>
<eventState>inactive</eventState>
<eventDescription>videoloss alarm</eventDescription>
<channelName/>
</EventNotificationAlert>
```

resoai, 2019-11-17:
Here is the list of my event triggers:

```
<?xml version="1.0" encoding="UTF-8" ?>
<EventTriggerList version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<EventTrigger>
<id>IO-1</id>
<eventType>IO</eventType>
<inputIOPortID>1</inputIOPortID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>IO-2</id>
<eventType>IO</eventType>
<inputIOPortID>2</inputIOPortID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>IO-3</id>
<eventType>IO</eventType>
<inputIOPortID>3</inputIOPortID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>IO-4</id>
<eventType>IO</eventType>
<inputIOPortID>4</inputIOPortID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>softIO-1</id>
<eventType>softIO</eventType>
<inputIOPortID>1</inputIOPortID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>softIO-2</id>
<eventType>softIO</eventType>
<inputIOPortID>2</inputIOPortID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>softIO-3</id>
<eventType>softIO</eventType>
<inputIOPortID>3</inputIOPortID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>softIO-4</id>
<eventType>softIO</eventType>
<inputIOPortID>4</inputIOPortID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>softIO-5</id>
<eventType>softIO</eventType>
<inputIOPortID>5</inputIOPortID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>softIO-6</id>
[... 689 lines cut ...]
<id>center</id>
<notificationMethod>center</notificationMethod>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>recordingfailure</id>
<eventType>recordingfailure</eventType>
<EventTriggerNotificationList>
<EventTriggerNotification>
<id>email</id>
<notificationMethod>email</notificationMethod>
</EventTriggerNotification>
<EventTriggerNotification>
<id>beep</id>
<notificationMethod>beep</notificationMethod>
</EventTriggerNotification>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
</EventTriggerList>
```

madprunes, 2019-11-17:
Perhaps disable the video devices which aren't connected or turn off video loss reporting to stop video loss flooding the alert stream, then post a few different events from the alert stream.

resoai, 2019-11-17:
I don't have any. devices which aren't connected, those are five Hikvision cameras. Sadly there is no way to disable video loss, as it just keeps coming. My understanding is that it is used to keep connection alive and without it, it would simply timeout.

madprunes, 2019-11-17:
Strange, I don't see that on my NVR and don't have timeout issues.

resoai, 2019-11-17:
Funny thing is that disabling "Notify Surveillance Center" for Video Loss actually solved the problem. I fail to understand why because curling still shows them:

```
curl --anyauth --user XXX:YYY http://192.168.1.200/ISAPI/Event/notification/alertStream
```

madprunes, 2019-11-17:
Ok so what is probably happening and what I saw myself with my NVR is that when things are in alarm they send through dozen of messages a second, and you can only really handle a couple of devices doing this before things start going funky, did you turn it off on the camera and the stuff under exceptions?

resoai, 2019-11-17:
Yes. I've disabled it for all cameras. It is still supposed to keep pushing it for keep alive:

https://github.com/mezz64/pyHik/blob/b0c60a01345e2f27d826343c5f63b3a8eaca6db5/pyhik/hikvision.py#L564

The reason it does not cause an error is because pyHik is ignoring these types of events since we are not "subscribed" to them. The reason is does fail is because execution of `process_stream` is not interrupted on line 566 but is allowed to continue and it obviously fails on line 570 while trying to call `fetch_attributes`.

mezz64, 2019-11-21:
For whatever reason videoloss events are always published in the stream no matter what settings you have turned on.  I don't know why they decided to do it that way, but it makes for a nice keep-alive indicator to know that the stream is active.

The problem in this case is a monitored event being published without an ID.  The way the individual events are tracked within pyHik is a combination of the name and the ID.  If there is no ID available you get the TypeError since the field ends up being blank.

I'm going to consider this as something that needs further exploration for now.  If videoloss events are the only culprit I may just consider removing them from the monitoring list.

resoai, 2019-11-21:
I think that whenever a real (active) videoloss event would occur, it would have an ID of the channel. We could have a branch in code specifically checking if this is inactive videoloss  and has no ID and consider it as heartbeat.

necromn, 2021-05-22:
Seems like this still might be a thing? I've recently upgrade my DS-7608NI-I2 to V4.22.000 build 201208 and now i'm getting

`2021-05-22 10:33:12 ERROR (Thread-3) [root] Uncaught thread exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.8/site-packages/pyhik/hikvision.py", line 528, in alert_stream
    self.process_stream(tree)
  File "/usr/local/lib/python3.8/site-packages/pyhik/hikvision.py", line 604, in process_stream
    state = self.fetch_attributes(etype, echid)
  File "/usr/local/lib/python3.8/site-packages/pyhik/hikvision.py", line 652, in fetch_attributes
    if sensor[1] == int(channel):
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'`

Homeassistant core-2021.5.5

mezz64, 2021-05-25:
@necromn Are you notifying the surveillance center for videoloss events?

necromn, 2021-06-01:
Yeah I read up the page and saw that so i thought i went and turned them all off but i'll double check

closed 2026-09-01 by mezz64