mezz64/pyHik #46: DS-KB6403-WIP Doorbell - binary_Sensor discovered in HA, but motion is never detected

lolouk44, 2019-11-29:
I've finally managed to connect to my Hikvision doorbell and have the sensor show up in HA.
It needs to use port 80 (whereas all my other HK cams need port 8000)
But motion is never detected.

I'm getting this in my HA logs:
```
2019-11-29 13:11:11 DEBUG (SyncWorker_0) [homeassistant.components.hikvision.binary_sensor] Entity: Doorbell - Motion, Options - Ignore: None, Delay: None
```
Nothing else...
What can I try please?

mezz64, 2019-11-30:
You need to check if the alertstream api endpoint actually exists.  Others have not had much luck with hikvision doorbell cams in the past (see issue #16) because it's usually not there.

closed 2020-02-13 by mezz64

pergolafabio, 2021-01-30:
Hey @lolouk44  , how did you add the doorbell to HA anyway? I have the ds-kd8003, want to try it also

lolouk44, 2021-01-30:
I use motion (https://github.com/Motion-Project/) to bring the video feed in HA 
I use motion to perform motion detection and send messages via mqtt
I use a spare android tablet with the Hik-Connect app and tasker to detect when the button is pressed and send this via mqtt

pergolafabio, 2021-01-30:
Ah ok, I have an ESP board, hooked up to indoor station, that captures an doorpress event, so I can start recording the stream on my synology station, and i also start displaying the stream to the google hubs