mezz64/pyHik #47: Hard drive errors

ayushpratap, 2019-12-24:
This API is unable to get errors/events related to HARD DRIVE/DISK 

DVR is sending that event on iVMS but that error/event is not obtained by this API.

mezz64, 2019-12-27:
```
'diskfull': 'Disk Full',
'diskerror': 'Disk Error',
```

These are both supported events.  If you are reporting an event that shows up in the event stream, but is not currently supported, please post a snippet of the XML from the event stream and the type can be added.  If the event does not show up the event stream it's not possible to add support.

ayushpratap, 2019-12-30:
I am testing the event stream extensively. 
I have added some more ISAPI specifications ( according to my usage )
I propose we can work together and implement more ISAPI specifications to this API, it will be very useful for others.
What do you  propose ?

mezz64, 2020-01-04:
I gladly accept PR's.  Please feel free to submit any new features and we can work together to get them merged.

closed 2020-02-13 by mezz64