mezz64/pyHik #51: Basic authentication failed. Using digest.

mvasicek, 2020-02-09:
Hi,
I do face error
[pyhik.hikvision] pyHik 0.2.5 initializing new hikvision device at: http://10.0.1.200
[pyhik.hikvision] Basic authentication failed. Using digest.
[pyhik.hikvision] Authentication failed

under HA 0.105.2 with configuration

`binary_sensor:
  - platform: hikvision
    host: 10.0.1.200
    port: 80
    ssl: false
    username: user
    password: pwd`

while camera HIKVISION DS-2CD2032F-I (V5.4.5 build 170123, latest) is fully accessible under those credentials and rstp video also in browser, ISAPI xml as well. Any ideas what to try?

mezz64, 2020-02-11:
Try going to this URL in a browser and logging in with your credentials.  Does it work?
` http://10.0.1.200/ISAPI/System/deviceInfo`

mvasicek, 2020-02-11:
Yes, that works with login. I do get xml...

<DeviceInfo xmlns="http://www.hikvision.com/ver20/XMLSchema" version="2.0">
<deviceName>Garaz</deviceName>
<deviceID>16c685bf-0117-11b4-8267-c42f90425f1a</deviceID>
<deviceDescription>IPCamera</deviceDescription>
<deviceLocation>hangzhou</deviceLocation>
<systemContact>Hikvision.China</systemContact>
<model>DS-2CD2032F-I</model>
<serialNumber>DS-2CD2032F-I20150615CCWR523830719</serialNumber>
<macAddress>c4:2f:90:42:5f:1a</macAddress>
<firmwareVersion>V5.4.5</firmwareVersion>
<firmwareReleasedDate>build 170123</firmwareReleasedDate>
<encoderVersion>V5.0</encoderVersion>
<encoderReleasedDate>build 170105</encoderReleasedDate>
<bootVersion>V1.3.4</bootVersion>
<bootReleasedDate>100316</bootReleasedDate>
<hardwareVersion>0x0</hardwareVersion>
<deviceType>IPCamera</deviceType>
<telecontrolID>88</telecontrolID>
<supportBeep>false</supportBeep>
<supportVideoLoss>false</supportVideoLoss>
</DeviceInfo>

mezz64, 2020-02-11:
Can you post the full formatted XML response?

mvasicek, 2020-02-11:
`
<DeviceInfo xmlns="http://www.hikvision.com/ver20/XMLSchema" version="2.0">
<deviceName>Garaz</deviceName>
<deviceID>16c685bf-0117-11b4-8267-c42f90425f1a</deviceID>
<deviceDescription>IPCamera</deviceDescription>
<deviceLocation>hangzhou</deviceLocation>
<systemContact>Hikvision.China</systemContact>
<model>DS-2CD2032F-I</model>
<serialNumber>DS-2CD2032F-I20150615CCWR523830719</serialNumber>
<macAddress>c4:2f:90:42:5f:1a</macAddress>
<firmwareVersion>V5.4.5</firmwareVersion>
<firmwareReleasedDate>build 170123</firmwareReleasedDate>
<encoderVersion>V5.0</encoderVersion>
<encoderReleasedDate>build 170105</encoderReleasedDate>
<bootVersion>V1.3.4</bootVersion>
<bootReleasedDate>100316</bootReleasedDate>
<hardwareVersion>0x0</hardwareVersion>
<deviceType>IPCamera</deviceType>
<telecontrolID>88</telecontrolID>
<supportBeep>false</supportBeep>
<supportVideoLoss>false</supportVideoLoss>
</DeviceInfo>
`

mezz64, 2020-02-11:
Thanks.  Can you check in the camera GUI on the Configuration tab, then the Advanced Configuration category and finally the Security menu option and let me know what type of Authentication is configured for the camera?

mvasicek, 2020-02-11:
Sure. This firmware allows to set for RTSP Disable or Basic. Behavior is the same for both values.
[image]

mvasicek, 2020-02-12:
Oh, crap. I found it. I haven't noticed HA warned me about two binary_sensor definitions. And f course, second one defined the same one under same IP, but different password and lower in the config file. It took this last credentials I haven't noticed as valid. Sorry for wasting your time.

mezz64, 2020-02-12:
No problem.  Seems like Hikvision is always changing things and not updating the API docs so anything can happen.

closed 2020-02-12 by mezz64