mezz64/pyHik #53: Acusense camera

kucau0901, 2020-02-25:
When used with newer hik acusense camera, apparently it uses the  VCA hence more false alarm despite the camera ability to detect human and vehicle

mezz64, 2020-02-25:
Sorry, i'm not familiar with the acusense cameras.  What is "VCA"?  You can always disable reporting for a certain even type by not allowing it to notify the surveillance center.

kucau0901, 2020-02-26:
Hikvision acusense use  more advance Vca ( video content analysis ). And  can clasify 2 subjects, Human and vehicle thus reducing false alarm  in cross line detection.  Maybe acusense use newer hikvision api?

mezz64, 2020-02-27:
Are you saying that that a hikvision acusense camera doesn't work with this library as-is?  I'd be happy to include changes for a new api if you can find the documentation.

closed 2020-04-08 by mezz64

robertvorthman, 2022-02-10:
@mezz64 I also have an issue with false motion events when using an Acusense camera's human detection.  I posted some debugging info in the [Home Assistant github issue tracker](https://github.com/home-assistant/core/issues/55081#issuecomment-1035415750) but I have repeated the relevant info below.

I am using motion detection with the human checkbox checked.  False positives do not appear in the camera's log but an event is still sent over the API.  Maybe the camera does not include false positives in it's own log if it does not find a human.  I wonder if the API includes the Acusense human/vehicle detection data to help filter out false positives?

### Here are my motion detection settings in my Hikvision 2387.

[image]

### The camera's log does not show the false event triggered at 12:59pm in the first JSON block below.

[image]

I was able to reproduce the false positives with [node-red-contrib-hikvision-ultimate](https://flows.nodered.org/node/node-red-contrib-hikvision-ultimate).

Here is a log of events I receive from node-red-contrib-hikvision-ultimate's raw alarm node.  The last entry in both JSON blocks shows "eventState":"inactive" while the rest show "eventState":"active".

### A motion false positive created these 6 raw events:
```json
[
{"channelID":"1","dateTime":"2022-02-10T12:59:50-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T12:59:50-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T12:59:51-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T12:59:51-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T12:59:52-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T12:59:52-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T12:59:53-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T12:59:53-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T12:59:54-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T12:59:54-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T12:59:55-05:00","activePostCount":"1","eventType":"VMD","eventState":"inactive","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T12:59:55-05:00,0,1.0"}}
]
```

### A genuine motion event caused by me walking in front of the camera produced these 9 raw events.

```json
[
{"channelID":"1","dateTime":"2022-02-10T13:32:02-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T13:32:02-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T13:32:03-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T13:32:03-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T13:32:04-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T13:32:04-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T13:32:05-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T13:32:05-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T13:32:06-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T13:32:06-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T13:32:07-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T13:32:07-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T13:32:08-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T13:32:08-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T13:32:09-05:00","activePostCount":"1","eventType":"VMD","eventState":"active","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T13:32:09-05:00,1.0"}},
{"channelID":"1","dateTime":"2022-02-10T13:32:11-05:00","activePostCount":"1","eventType":"VMD","eventState":"inactive","eventDescription":"Motion alarm","channelName":"Camera 01","Extensions":{"eventPush":"VMD&&DS-2CD2387G2-LU,2022-02-10T13:32:11-05:00,0,1.0"}}
]
```

### Home Assistant logs show both false and positive motion events.

[image]

mezz64, 2022-02-10:
Unfortunately, Hikvision doesn't seem to be adding any new identifying tag to the motion alert in the alert stream.  They all look exactly the same whether or not it's a human.  Since there isn't any information available to process there isn't much I can do to remove the non-human alerts.

Only potential workaround I can think of is if there is another type of trigger available that can be enabled to indicate a human.  It's possible that exists and just isn't in the current supported list.  If so, that can be added and used instead of the standard VMD trigger.

It's also likely there are some new endpoints for the newer cameras, but Hikvision hasn't published any new API docs in quite a while so somebody would have to figure out what is available.

stevedh42, 2022-05-21:
I know this is a very old discussion but I am facing the same issue, although for triggering my own scripts, and my comment might help anyone else facing this issue.

Note ftp uploads are also correctly triggered, so I may look at using that.

I find it odd they couldn't have added an extra tag to identify human detection in the alarm server messages. As they stand they are useless to me. Anyway will be emailing them to point out this issue, even though it seems to have been around for a couple of years.
Anyway looking at trapping ftp's

beatmag, 2022-07-31:
Hi all,
I just got a hikvision acusense DS-2CD2386G2-ISU/SL.
Getting exactly the same issue. 
Firmware [Firmware_V5.7.3_220112].

Really not sure why the ISAPI Alerts for Motion are coming through even though there were no Humans.
Like everyone has found out that, the camera log doesn't show this event either.

I've tried ISAPI event streams from alarmserver project. Clearly all motion events are coming through and its not applying the human filtering.

Likewise using Alarm Server, same thing is found. There doesn't seem to be any difference in the message hence cannot determine human was there or not.

I would call this a Hikvision bug to be honest. But why its still not fixed I have no idea.

It makes not sense for the event to now shot in Camera Log but fired off for ISAPI and Alarm Server.
Also makes even less sense that FTP upload is working correctly as expected.

I've written an email to hikvision. i assumed you all wrote already and they didn't seem to do anything about it.

But clearly firing an event thats not logged in the camera is a bug.

Has anyone found any solutions? Are there any other protocols or methods better than FTP..
Has any one tried email linkage to see if that works for humans only?

beatmag, 2022-07-31:
https://github.com/home-assistant/core/issues/55081#issuecomment-1200404098

Was trying Smart Events - Intrusion detection.
The event sent is fielddetection , which also includes the Class type human or vehicle.
Also the coord of the detection. The information is there, I think we just have to capture the new data.

So we should be able to this Event instead of Basic Motion Detection to make use of Acusense.