mezz64/pyHik #57: XML error on DS-7204HUHI

JoeArny, 2020-04-02:
Hi I have a DS-7204HUHI and when I run the script (basic_usage.py) I get the following:

> Motion Dectect State: None

I debug the problem and could find the following.
The device info XML is the following:

> <?xml version="1.0" encoding="UTF-8" ?>
> <DeviceInfo version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
> <deviceName>XXX</deviceName>
> <deviceID>XXXXXXXXXXXXXXXXX</deviceID>
> <model>DS-7204HUHI-K1</model>
> <serialNumber>DS-7204HUHI-K10420180912CCWRC49953259WCVU</serialNumber>
> <macAddress>XXXXXXXXXXXX</macAddress>
> <firmwareVersion>V4.21.000</firmwareVersion>
> <firmwareReleasedDate>build 190722</firmwareReleasedDate>
> <encoderVersion>V5.0</encoderVersion>
> <encoderReleasedDate>build 190716</encoderReleasedDate>
> <deviceType>IPC</deviceType>
> <telecontrolID>255</telecontrolID>
> <hardwareVersion>0x20c3500</hardwareVersion>
> </DeviceInfo>

Which sets the namespace to "**http://www.hikvision.com/ver20/XMLSchema**", but the event trigger is as follow:

> <?xml version="1.0" encoding="UTF-8" ?>
> <EventTriggerList version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
> <EventTrigger version="1.0" xmlns="http://www.hikvision.com/ver20/mmmm/XMLSchema">
> <id>IO-1</id>
> <eventType>IO</eventType>
> <inputIOPortID>1</inputIOPortID>
> <EventTriggerNotificationList version="1.0" xmlns="http://www.hikvision.com/ver20/mmmm/XMLSchema">
> </EventTriggerNotificationList>
> </EventTrigger>
> <EventTrigger version="1.0" xmlns="http://www.hikvision.com/ver20/mmmm/XMLSchema">
> <id>IO-2</id>
> <eventType>IO</eventType>
> <inputIOPortID>2</inputIOPortID>
> <EventTriggerNotificationList version="1.0" xmlns="http://www.hikvision.com/ver20/mmmm/XMLSchema">
> </EventTriggerNotificationList>
> </EventTrigger>
> ...

Note that here the first namespace matches the one in the device info, but not the rest, the ones in the EventTrigger are "**http://www.hikvision.com/ver20/_mmmm/_XMLSchema**", there is an extra mmmm/ in it.
If I comment the line (hikvision.py):

> self.namespace = nmsp if nmsp.startswith('http') else XML_NAMESPACE

and set XML_NAMESPACE = http://www.hikvision.com/ver20/mmmm/XMLSchema

Then I get a proper output:

> {'Motion': [[False, 1, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)], [False, 2, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)], [False, 3, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)], [False, 4, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)]], 'Field Detection': [[False, 1, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)], [False, 2, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)], [False, 3, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)], [False, 4, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)]], 'Line Crossing': [[False, 1, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)], [False, 2, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)], [False, 3, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)], [False, 4, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)]], 'Scene Change Detection': [[False, 1, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)], [False, 2, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)], [False, 3, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)], [False, 4, 0, datetime.datetime(2020, 4, 2, 23, 11, 7, 10110)]]}
> Motion Dectect State: True

Is there a way to remove this mmmm from the XML my device sends? Or perhaps a fix in this script that would allow my cameras to work? 
I would like to use my cameras events on Home Assistant, but I can't because of this... :(
Thank you very much in advance.

mezz64, 2020-04-07:
This looks to be the same as reported in #48.  I had planned to put something in place more robust to fix it, but I haven't been able to find the time.  I'll implement a more simple patch for this particular case soon.

mezz64, 2020-04-08:
Please give this branch a try: https://github.com/mezz64/pyHik/tree/namespace_fix

Let me know if that fixes the issue.

JoeArny, 2020-04-11:
Sorry for my delay to answer. I not always can touch this during week days...

This version allows it to detect the DVR properly, but still do not see the alerts. (my fix above were not showing alerts either...)

The curious thing is that the AlertStream uses the same namespace as the one originally in the script. In order to fix that I added the following in the line 470 of the branch you pointed:
            self.namespace = 'http://www.hikvision.com/ver20/XMLSchema'

Now it works perfectly recognizing the device and showing alerts.

mezz64, 2020-04-13:
I made some changes to only use the modified namespace for the trigger call.  Please give that branch a try again and let me know if it works now.

JoeArny, 2020-04-13:
Hi mezz64, it's PERFECT! Working like a charm. :)

Thank you very much!

mezz64, 2020-04-14:
Glad it works.  I've merged into master and sent a new release to pypi as well as submitted a PR to update home assistant.  Closing this issue.

closed 2020-04-14 by mezz64