mezz64/pyHik #58: Listen for detection event and take photo.

nathancn, 2020-04-07:
How can I listen to the "motion detection" event, take a photo and capture it using?
In case I need to get a photo every time the "motion detection" event is activated.

thank you since now

mezz64, 2020-04-07:
You'll need to use another application that implements this library to achieve that.  Home assistant is the most popular use case and it can do what you're looking for.

nathancn, 2020-04-07:
Understand. Would I be able to listen to the motion detection event and perform an action every time motion detection is activated? Because if I can, I thought about doing this and use the camera's api to take the photo through an http and every time there is a motion detection.

mezz64, 2020-04-08:
This library listens to the camera/nvr event stream and pushes them to subscribers. If you'd like to roll your own program the basic usage example shows you how to subscribe to the events, from there it would just be a matter of a requests call to the camera image url to save the image.

nathancn, 2020-04-08:
Got it, I'll create the script and return here with the result. Thanks a lot for the help!

ozett, 2020-06-16:
would like to see some fragment-code for this here
(as i try the same as a newbie...) thx

closed 2020-10-24 by mezz64