mezz64/pyHik #60: Accept self signed certs

pdobrien3, 2020-04-20:
With the recent addition of alexa camera streaming needing https, when I set my Hikvision up to use ssl, my binary_sensors go to unknown. I have been told that the integration needs the ability to accept self signed certs. I have no knowledge other than what I have been told. Just respectfully asking if this is an option.

mezz64, 2020-04-20:
There are two options for this, first is to not verify certs at all, which is not something i'd like to add for the obvious security implications.  Second is to allow the specification of the path to your local certificate file to allow it to be verified.

The 2nd is feasible, but requires changes not only in the library, but to the home assistant component to add the additional configuration parameter.  If you're only doing this for alexa compatibility i'd recommend you leave the cameras alone and proxy them through something else to give you your https layer as it's generally not a great idea to give them direct access to the internet.

pdobrien3, 2020-04-20:
Thank you for your quick response. When you say proxy, do you mean something like nqinx?  I just setup a nginx proxy but was under the impression that was giving things direct access to the internet.

mezz64, 2020-04-22:
Sure, nginx would work with something like the rtmp module added (https://github.com/arut/nginx-rtmp-module).  You can let it proxy internally without opening anything up your firewall.

closed 2020-10-24 by mezz64

Harryjholmes, 2020-11-22:
Hi, I'm trying to use the native hikvision `binary_sensor` component in homeassistant and use a self-signed certificate. Surely by having a `verify_ssl` option it could default be `true` but changed to `false` for someone like me who can't install the certificate in homeassistant to verify it?

mezz64, 2020-12-01:
The problem is adding that option to flow into homeassistant requires a significant number of changes at this point.  The edits to pyhik are not extensive, but config changes to homeassistant will require this component to adopt the new config flow and that is not something I have time to complete at the moment.

I'll always accept PR's if it's something you'd like to tackle on your own.