mezz64/pyHik #64: There was a problem: 'NoneType' object has no attribute 'text'

smeegoan, 2020-09-03:
Hi,

I updated my Hikvision NVR firmware and now the XML response namespace is different http://www.isapi.org/ver20/XMLSchema instead of http://www.hikvision.com/ver20/XMLSchema

I created a pull request to fix this [here ](https://github.com/mezz64/pyHik/pull/63)

```
2020-09-02 08:37:00 DEBUG (SyncWorker_21) [pyhik.hikvision] pyHik 0.2.7 initializing new hikvision device at: http://192.168.1.64
2020-09-02 08:37:00 DEBUG (SyncWorker_21) [pyhik.hikvision] Basic authentication failed. Using digest.
2020-09-02 08:37:00 DEBUG (SyncWorker_21) [pyhik.hikvision] Using Namespace: http://www.hikvision.com/ver20/XMLSchema
2020-09-02 08:37:00 DEBUG (SyncWorker_21) [pyhik.hikvision] Processed 48433334-3438-3632-3937-5803fb2309a7 as CAM Device.
2020-09-02 08:37:00 DEBUG (SyncWorker_21) [pyhik.hikvision] Found events: {}
2020-09-02 08:37:00 DEBUG (SyncWorker_21) [pyhik.hikvision] No Events available in dictionary.
2020-09-02 08:37:00 ERROR (SyncWorker_21) [pyhik.hikvision] Entire response: <?xml version="1.0" encoding="UTF-8" ?>
<MotionDetection version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<enabled>true</enabled>
<enableHighlight>true</enableHighlight>
<samplingInterval>5</samplingInterval>
<startTriggerTime>1000</startTriggerTime>
<endTriggerTime>1000</endTriggerTime>
<regionType>grid</regionType>
<Grid>
<rowGranularity>18</rowGranularity>
<columnGranularity>22</columnGranularity>
</Grid>
<MotionDetectionLayout version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<sensitivityLevel>60</sensitivityLevel>
<layout>
<gridMap>00000000000000000000000000000000f80000ff0000ff8000ffc000fffc01fffc01fffc01fffc01fffc01fffc01fffc01fffc01fffc</gridMap>
</layout>
<targetType></targetType>
</MotionDetectionLayout>
</MotionDetection>

2020-09-02 08:37:00 ERROR (SyncWorker_21) [pyhik.hikvision] There was a problem: 'NoneType' object has no attribute 'text'
```

smeegoan, 2020-09-03:
I opened the issue just in case you missed the PR

mezz64, 2020-09-04:
Can you post what your response looks like from the `/ISAPI/Event/triggers` endpoint on your NVR?  Just want to make sure we can't fix this another way earlier in the initialization process.

smeegoan, 2020-09-06:
Here it is:

```
<?xml version="1.0" encoding="UTF-8" ?>
<EventTriggerList version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<EventTrigger version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<id>IO-1</id>
<eventType>IO</eventType>
<inputIOPortID>1</inputIOPortID>
<EventTriggerNotificationList version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<id>IO-2</id>
<eventType>IO</eventType>
<inputIOPortID>2</inputIOPortID>
<EventTriggerNotificationList version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<id>IO-3</id>
<eventType>IO</eventType>
<inputIOPortID>3</inputIOPortID>
<EventTriggerNotificationList version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<id>IO-4</id>
<eventType>IO</eventType>
<inputIOPortID>4</inputIOPortID>
<EventTriggerNotificationList version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<id>VMD-1</id>
<eventType>VMD</eventType>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<EventTriggerNotification>
<id>record-1</id>
<notificationMethod>record</notificationMethod>
<dynVideoInputID>1</dynVideoInputID>
</EventTriggerNotification>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<id>VMD-3</id>
<eventType>VMD</eventType>
<dynVideoInputChannelID>3</dynVideoInputChannelID>
<EventTriggerNotificationList version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<EventTriggerNotification>
<id>record-3</id>
<notificationMethod>record</notificationMethod>
<dynVideoInputID>3</dynVideoInputID>
</EventTriggerNotification>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
[... 388 lines cut ...]
<id>illaccess</id>
<eventType>illaccess</eventType>
<EventTriggerNotificationList version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<id>recordingfailure</id>
<eventType>recordingfailure</eventType>
<EventTriggerNotificationList version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<EventTriggerNotification>
<id>beep</id>
<notificationMethod>beep</notificationMethod>
</EventTriggerNotification>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
</EventTriggerList>
```

mezz64, 2020-10-24:
Please test out the latest commit on the newcamfix branch to see if it fixes your issue.  I refactored things to track a namespace for each endpoint. https://github.com/mezz64/pyHik/tree/newcamfix

smeegoan, 2020-10-24:
The issue is fixed with those changes. Many thanks!

mezz64, 2020-10-24:
Closing. Fixed in 0.2.8.

closed 2020-10-24 by mezz64