mezz64/pyHik #67: Track events not shown in triggers XML endpoint

mezz64, 2020-10-24:
Some newer devices don't seem to have the ISAPI triggers endpoint that returns all camera trigger options, instead only responding to the non-ISAPI endpoint which only lists basic events.

An option should be added to allow the user to pass in a manually defined dictionary of events they would like to track and process those items from the alertStream without caring if they exist in the trigger XML.

theBOFH, 2020-10-29:
followed this over from "New gen Hikvision WIFI-Cam not usable with pyHik? #62"
Just upgraded to HomeAssistant 0.117, thank you very much for getting the basic motion binary sensor working :-)

Very much interested in getting the other sensors working (intrusion detection and line cross).
Looking into the possibility of using the human/vehicle triggers to automate outdoor lighting etc 

Let me know if there is anything I can assist with on the testing front. 

Seeing the same as the testing in #62 can see the triggers being posted when listening to the
`curl  http://ip/ISAPI/Event/notification/alertStream --digest -u username:password `

`<EventNotificationAlert version="1.0" xmlns="http://www.hikvision.com/ver10/XMLSchema">
<ipAddress> </ipAddress>
<portNo>80</portNo>
<protocol>HTTP</protocol>
<macAddress> </macAddress>
<channelID>1</channelID>
<dateTime>2020-10-29T13:33:18+00:00</dateTime>
<activePostCount>1</activePostCount>
<channelName>Camera 01</channelName>
<eventType>linedetection</eventType>
<eventState>active</eventState>
<eventDescription>linedetection alarm</eventDescription>
<DetectionRegionList>
<DetectionRegionEntry>
<regionID>1</regionID>
<sensitivityLevel>50</sensitivityLevel>
<RegionCoordinatesList>
<RegionCoordinates>
<positionX>527</positionX>
<positionY>986</positionY>
</RegionCoordinates>
<RegionCoordinates>
<positionX>543</positionX>
<positionY>432</positionY>
</RegionCoordinates>
</RegionCoordinatesList>
<detectionTarget>human</detectionTarget>
</DetectionRegionEntry>
</DetectionRegionList>
</EventNotificationAlert>
--boundary`

keiron1992, 2020-10-30:
Following. Wish there was a way I could help :(

cllp53, 2020-11-07:
I think it would be helpful to add all the sensors available from the capability list of the "smart" alarms even though the trigger might not be set to Surveillance Center. I currently have only basic motion sensor available in HA, and it's frustrating... Thanks for considering :)

`http://ip/ISAPI/Smart/capabilities`

`<?xml version="1.0" encoding="UTF-8"?>
<SmartCap version="2.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
    <isSupportROI>true</isSupportROI>
    <isSupportAudioDetection>false</isSupportAudioDetection>
    <isSupportFaceDetect>false</isSupportFaceDetect>
    <isSupportIntelliTrace>false</isSupportIntelliTrace>
    <isSupportFieldDetection>true</isSupportFieldDetection>
    <isSupportDefocusDetection>false</isSupportDefocusDetection>
    <isSupportSceneChangeDetection>true</isSupportSceneChangeDetection>
    <isSupportLineDetection>true</isSupportLineDetection>
    <isSupportRegionEntrance>true</isSupportRegionEntrance>
    <isSupportRegionExiting>true</isSupportRegionExiting>
    <isSupportLoitering>false</isSupportLoitering>
    <isSupportGroup>false</isSupportGroup>
    <isSupportRapidMove>false</isSupportRapidMove>
    <isSupportParking>false</isSupportParking>
    <isSupportUnattendedBaggage>false</isSupportUnattendedBaggage>
    <isSupportAttendedBaggage>false</isSupportAttendedBaggage>
    <isSupportPeopleDetection>false</isSupportPeopleDetection>
    <isSupportSmartCalibration>true</isSupportSmartCalibration>
    <isSupportStorageDetection>false</isSupportStorageDetection>
    <isSupportChannelResource>false</isSupportChannelResource>
</SmartCap>`

mezz64, 2020-11-08:
The goal will be to let the user specify, so anything that shows up in the alert stream will be valid, this should include smart events.

mbedk, 2020-11-13:
Any idea when the line crossing will be available?

ruffle-b, 2021-03-26:
If you upgrade your camera firmware (5.5.160 worked for me) then the Smart Events work again. Looks like there was a firmware bug.

pergolafabio, 2021-07-29:
> The goal will be to let the user specify, so anything that shows up in the alert stream will be valid, this should include smart events.

Hi, is this still in development?

I have an intercom device , with events about the door status...
https://github.com/mezz64/pyHik/issues/77

closed 2026-09-01 by mezz64