mezz64/pyHik #76: HomeAssistant 'Problem finding attribute: 'duration'' in logging

callifo, 2021-07-21:
I'm finding with my new ColorVu outdoor cameras they fill the HA logs with these errors when they are generating events sent to HA. E.g. it only happens when I or someone/something is moving in front of the camera.

https://www.hikvision.com/en/products/IP-Products/Network-Cameras/colorvu-series/ds-2cd2347g2-l-u-/

`2021-07-21 10:41:08 ERROR (Thread-7) [pyhik.hikvision] Problem finding attribute: 'duration'`

I've got this from curl http://YOUR_CAM_IP/ISAPI/Event/triggers --digest
```
<?xml version="1.0" encoding="UTF-8"?>
<EventTriggerList version="2.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<EventTrigger>
<id>VMD-1</id>
<eventType>VMD</eventType>
<eventDescription>VMD Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>tamper-1</id>
<eventType>tamperdetection</eventType>
<eventDescription>shelteralarm Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>diskfull</id>
<eventType>diskfull</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>diskerror</id>
<eventType>diskerror</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>nicbroken</id>
<eventType>nicbroken</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>ipconflict</id>
<eventType>ipconflict</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>illaccess</id>
<eventType>illaccess</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
[... 515 lines cut ...]
Content-Type: application/xml; charset="UTF-8"
Content-Length: 779

<?xml version="1.0" encoding="UTF-8"?>
<EventNotificationAlert version="2.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<ipAddress>192.168.100.131</ipAddress>
<ipv6Address>::</ipv6Address>
<portNo>80</portNo>
<protocol>HTTP</protocol>
<macAddress>08:a1:89:6d:c1:9b</macAddress>
<channelID>1</channelID>
<dateTime>2021-07-21T15:42:59+09:30</dateTime>
<activePostCount>1</activePostCount>
<eventType>videoloss</eventType>
<eventState>inactive</eventState>
<eventDescription>videoloss alarm</eventDescription>
<channelName>Camera 01</channelName>
<Extensions>
<serialNumber>DS-2CD2347G2-LU20210311AAWRF64923035</serialNumber>
<eventPush>DS-2CD2347G2-LU20210311AAWRF64923035,2021-07-21T15:42:59+09:30,1.0</eventPush>
</Extensions>
</EventNotificationAlert>
```

I tried it twice and it did the same both times. I cut the amount of garbage down  as there would have been like 20 pages of it.

Mincka, 2021-09-07:
This issue was also troubleshot here:
https://github.com/Supergiovane/node-red-contrib-hikvision-ultimate/issues/16

The exception is thrown [here](https://github.com/mezz64/pyHik/blob/master/pyhik/hikvision.py#L582) because the `duration` eventType is unknown from [sensor map](https://github.com/mezz64/pyHik/blob/master/pyhik/constants.py#L27).

This is a special event type to notify the beginning and ending of the target moving inside the zone for Intrusion Detection (aka FieldDetection). Its parsing is more complicated since the linked events are in a list:

```
<DurationList>
  <Duration>
    <relationEvent>fielddetection</relationEvent>
  </Duration>
</DurationList>
```

Maybe it's useful if you catch up the feed in the middle of an event. The receiver knows there are some kind of events still in progress.

From what I see in the linked issue, it should not be useful for HA since there are already `fielddetection` events with the states `active` then `inactive`. In this case, the easy fix would be to add `duration` to the sensor map but ignore its processing (return directly in `process_stream`.

mezz64, 2021-09-11:
I'm short on time these days to implement fixes, but can usually make time to review and merge a PR if someone wants to take a stab at it.