mezz64/pyHik #81: faceSnap sensor support

Yevgenium, 2021-12-21:
Every time I boot Home Assistant, I get this error:
`WARNING (SyncWorker_1) [pyhik.hikvision] Sensor type "faceSnap" is unsupported.`

I think it has to do with the 'Face Capture' function in the 'VCA Resource':
[image]

Is it possible to make a sensor for this function available in the Home Assistant?

mezz64, 2021-12-23:
It can potentially be added as long as the responses it produces are formatted in the same manner as other events.  Can you supply a curl output from your triggers endpoint as well as a snippet from the eventstream containing a "faceSnap" event?

pete-leese, 2022-01-17:
I have this error, but new to HikVision - if you can give some instructions on how to get the information you require I am happy to help :)

mezz64, 2022-01-21:
Sure thing.

Can you run a curl on the expected trigger path and post the output? Change that last flag if you're not using digest auth.
```
curl http://YOUR_CAM_IP/ISAPI/Event/triggers --digest
```

This will let you monitor the alertStream, I'd expect the first one to work, but please try both and let me know what the outputs are.

```
curl --digest -N http://YOUR_CAM_IP/Event/notification/alertStream
or
curl --digest -N http://YOUR_CAM_IP/ISAPI/Event/notification/alertStream
```

EvertJob, 2022-01-21:
`This XML file does not appear to have any style information associated with it. The document tree is shown below.
<EventTriggerList xmlns="http://www.hikvision.com/ver20/XMLSchema" version="2.0">
<EventTrigger>
<id>VMD-1</id>
<eventType>VMD</eventType>
<eventDescription>VMD Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>tamper-1</id>
<eventType>tamperdetection</eventType>
<eventDescription>shelteralarm Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>diskfull</id>
<eventType>diskfull</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>diskerror</id>
<eventType>diskerror</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
<notificationRecurrence>beginning</notificationRecurrence>
</EventTriggerNotification>
</EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>nicbroken</id>
<eventType>nicbroken</eventType>
<eventDescription>exception Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>ipconflict</id>
<eventType>ipconflict</eventType>
<eventDescription>exception Information</eventDescription>
[... 150 lines cut ...]
<EventTrigger>
<id>purple-1</id>
<eventType>purple</eventType>
<eventDescription>purple Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>ICR-1</id>
<eventType>ICR</eventType>
<eventDescription>ICR Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
<EventTrigger>
<id>protectiveFilm-1</id>
<eventType>protectiveFilm</eventType>
<eventDescription>protectiveFilm Event trigger Information</eventDescription>
<videoInputChannelID>1</videoInputChannelID>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList></EventTriggerNotificationList>
</EventTrigger>
</EventTriggerList>`

EvertJob, 2022-01-21:
> Sure thing.
>
> Can you run a curl on the expected trigger path and post the output? Change that last flag if you're not using digest auth.
>
> ```
> curl http://YOUR_CAM_IP/ISAPI/Event/triggers --digest [...]

I've tried the last 2 links but error 401 is coming up. I've set everything to digest but need's to login anyway.

mezz64, 2022-01-21:
The alertStream endpoint should be a continuous stream of data.  If faceSnap is what I think it is you should start logging the endpoint and then walk in front of the camera to see if it throws a faceSnap event.

EvertJob, 2022-01-21:
No, don’t see anything else exact the line’s I’ve posted above.

mezz64, 2022-01-24:
Can you post the curl command you are using?  If you only get that output and then the connection closes when attaching to the alertStream i'm afraid this component isn't going to work for you since we rely on a continuous stream of events from the device to process.

EvertJob, 2022-01-26:
Ok,

got something

`curl -v  'http://admin:password@192.168.0.195/Event/notification/alertStream' --digest -o  outputface.txt`

When I walk past the camera this is coming up:"
[outputface.txt](https://github.com/mezz64/pyHik/files/7940598/outputface.txt)

Symbiot78, 2022-05-24:
@mezz64  - any update on this? I am seeing the following which I believe is related to this issue:

Logger: pyhik.hikvision
Source: /usr/local/lib/python3.9/site-packages/pyhik/hikvision.py:302
First occurred: May 23, 2022, 10:55:54 (4 occurrences)
Last logged: May 23, 2022, 10:55:54

Sensor type "faceContrast" is unsupported.
Sensor type "faceContrastSuccess" is unsupported.

mezz64, 2022-10-27:
Sorry for the slow follow-up on this one, can @EvertJob and @Symbiot78 try out the changes on this branch (https://github.com/mezz64/pyHik/tree/face_sens)?  From what I can tell the event log formatting is the same so we should be able to pick these up without issue.  Just need someone to test.

closed 2025-01-31 by Yevgenium