mezz64/pyHik #85: Delay is receiving Events

beatmag, 2022-08-17:
I've been investigating why there is a delay when receiving events. It appears to be from the requests http library. The way iter_lines work is that it waits until default chunk size is filled before processing of lines can start. Hence what happens is that if your event just happens to cut off at the default chunk size of 512 then you sit there waiting until the next keep alive event to occur before that event can complete since it's transmission has paused. 

Now there seems to be no way to fix this issue except to set iter_content chunk size to 1. However this causes severe performance problems.  

I have a acusense camera and on intrusion event (field detection) it will force push a jpeg. Hence chunk size 1 it takes 3-4 seconds before the jpeg can completely download. I've not been able to disable the jpeg being pushed by the camera. if it was app xml event only chunk size 1 on iter_content would be alright. 

any ideas on how to fix this
I've been looking at aiohttp async lib and it is able to return data in chunks that was sent from camera. so the break points are how the camera actually sent the data. Not how we need to fill the chunk buffer. however aiohttp doesn't have digest auth support so need to manually do the auth handshake.

BlindChickens, 2022-12-06:
I fixed it here: 
https://github.com/mezz64/pyHik/pull/84

It does have the consequence of being slightly slower per line read from the stream. But the delay in receiving events is way worse.
I patched my local version with this code.

mezz64, 2022-12-06:
I do want to move things over to async eventually and have found some work by others to bolt on digest auth support to aiohttp just need to find the time to do it.  

The iter_content chunk size to 1 is a fix, but not something I would make default due to the performance issues.

closed 2026-09-01 by mezz64