mezz64/pyHik #88: Home Assistant 2022.12.0B0

pdobrien3, 2022-12-01:
so, reference the Hikvision breaking change.  I made sure everything was set according to this:
https://github.com/home-assistant/home-assistant.io/pull/24730/commits/303c30ec60ea36266650de12d6ce3fde17d43555
And I am still getting these errors:
Logger: pyhik.hikvision
Source: /usr/local/lib/python3.10/site-packages/pyhik/hikvision.py:606
First occurred: 10:27:37 PM (114 occurrences)
Last logged: 10:30:36 PM

Problem finding attribute: 'videoloss'
pdobrien3 — Today at 10:48 PM
I tried this:
customize:
  video_loss:
    ignored: true
and still no help

pdobrien3, 2022-12-01:
All the sensors seem to be working it is just flooding the logs with errors.  After posting here, it occurred to me I should have posted this on the Home Assistant github.  Would you prefer I close this issue and open it there for tracking purposes?

mezz64, 2022-12-01:
Leaving it here is fine.  Did it work without the error prior to the updates?  Can you post a full log from startup, not just the summary?

Adding the videoloss to the ignore list will only work if you have videoloss events enabled to notify the surveillance center on your camera.

pdobrien3, 2022-12-01:
Ok, I don’t think I even have a videoloss option on my camera. It did work before the latest beta update. I will get you some logs.

pdobrien3, 2022-12-01:
The binary sensors seem to setup fine. No errors. And then this just floods the logs:

2022-12-01 09:15:26.429 ERROR (Thread-5 (alert_stream)) [pyhik.hikvision] Problem finding attribute: 'videoloss'

pdobrien3, 2022-12-02:
found this also:

2022-12-01 18:18:09.955 WARNING (Thread-4 (alert_stream)) [pyhik.hikvision] Front_Yard Connection Failed (count=1). Waiting 10s. Err: Watchdog failed.
2022-12-01 18:18:09.963 WARNING (Thread-5 (alert_stream)) [pyhik.hikvision] Front Yard 2 Connection Failed (count=1). Waiting 10s. Err: Watchdog failed.
2022-12-01 18:18:09.967 WARNING (Thread-3 (alert_stream)) [pyhik.hikvision] Back_Yard Connection Failed (count=1). Waiting 10s. Err: Watchdog failed.

pdobrien3, 2022-12-06:
I am not sure I am comfortable posting my entire startup log. I went through the entire log and there is nothing but what I posted. I will enable debug and see what it produces.

pdobrien3, 2022-12-06:
Does this help?

2022-12-06 17:30:24.124 INFO (MainThread) [homeassistant.setup] Setup of domain binary_sensor took 0.0 seconds
2022-12-06 17:30:24.124 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.hikvision
2022-12-06 17:30:24.125 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.hikvision
2022-12-06 17:30:24.125 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.hikvision
2022-12-06 17:30:24.125 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.workday
2022-12-06 17:30:24.125 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.command_line
2022-12-06 17:30:24.125 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.command_line
2022-12-06 17:30:24.126 DEBUG (SyncWorker_18) [pyhik.hikvision] pyHik 0.3.1 initializing new hikvision device at: http://xxxxxxxxxxxx
2022-12-06 17:30:24.128 DEBUG (SyncWorker_3) [pyhik.hikvision] pyHik 0.3.1 initializing new hikvision device at: http://xxxxxxxxxxxx
2022-12-06 17:30:24.129 DEBUG (SyncWorker_13) [pyhik.hikvision] pyHik 0.3.1 initializing new hikvision device at: http://xxxxxxxxxxxx
2022-12-06 17:30:24.154 DEBUG (SyncWorker_3) [pyhik.hikvision] Device info namespace: http://www.hikvision.com/ver20/XMLSchema
2022-12-06 17:30:24.158 DEBUG (SyncWorker_18) [pyhik.hikvision] Device info namespace: http://www.hikvision.com/ver20/XMLSchema
2022-12-06 17:30:24.161 DEBUG (SyncWorker_13) [pyhik.hikvision] Device info namespace: http://www.hikvision.com/ver20/XMLSchema
2022-12-06 17:30:24.185 DEBUG (SyncWorker_18) [pyhik.hikvision] Device triggers namespace: http://www.hikvision.com/ver20/XMLSchema
2022-12-06 17:30:24.185 DEBUG (SyncWorker_18) [pyhik.hikvision] Processed e4eb0000-4344-11b5-8401-a414377b3dde as CAM Device.
2022-12-06 17:30:24.185 DEBUG (SyncWorker_18) [pyhik.hikvision] Found events: {'VMD': [1], 'tamperdetection': [1], 'diskfull': [1], 'diskerror': [1], 'illaccess': [1], 'linedetection': [1], 'fielddetection': [1]}
2022-12-06 17:30:24.187 DEBUG (SyncWorker_3) [pyhik.hikvision] Device triggers namespace: http://www.hikvision.com/ver20/XMLSchema
2022-12-06 17:30:24.187 DEBUG (SyncWorker_3) [pyhik.hikvision] Processed e4eb0000-4344-11b5-8401-a414377b3d0a as CAM Device.
2022-12-06 17:30:24.187 DEBUG (SyncWorker_3) [pyhik.hikvision] Found events: {'VMD': [1], 'tamperdetection': [1], 'diskfull': [1], 'diskerror': [1], 'illaccess': [1], 'linedetection': [1], 'fielddetection': [1]}
2022-12-06 17:30:24.189 DEBUG (SyncWorker_3) [pyhik.hikvision] Initialized Dictionary: {'Motion': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 189903)]], 'Tamper Detection': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 189906)]], 'Disk Full': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 189908)]], 'Disk Error': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 189909)]], 'Illegal Access': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 189910)]], 'Line Crossing': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 189912)]], 'Field Detection': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 189913)]]}
2022-12-06 17:30:24.191 DEBUG (SyncWorker_13) [pyhik.hikvision] Device triggers namespace: http://www.hikvision.com/ver20/XMLSchema
2022-12-06 17:30:24.191 DEBUG (SyncWorker_13) [pyhik.hikvision] Processed 29c30000-6580-11b3-81a1-1868cb480b1a as CAM Device.
2022-12-06 17:30:24.192 DEBUG (SyncWorker_13) [pyhik.hikvision] Found events: {'VMD': [1], 'tamperdetection': [1], 'diskfull': [1], 'diskerror': [1], 'illaccess': [1], 'linedetection': [1], 'fielddetection': [1]}
2022-12-06 17:30:24.192 DEBUG (SyncWorker_13) [pyhik.hikvision] Initialized Dictionary: {'Motion': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 192156)]], 'Tamper Detection': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 192159)]], 'Disk Full': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 192161)]], 'Disk Error': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 192161)]], 'Illegal Access': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 192162)]], 'Line Crossing': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 192163)]], 'Field Detection': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 192164)]]}
2022-12-06 17:30:24.193 DEBUG (SyncWorker_18) [pyhik.hikvision] Initialized Dictionary: {'Motion': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 193675)]], 'Tamper Detection': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 193678)]], 'Disk Full': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 193679)]], 'Disk Error': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 193680)]], 'Illegal Access': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 193682)]], 'Line Crossing': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 193683)]], 'Field Detection': [[False, 1, 0, datetime.datetime(2022, 12, 6, 17, 30, 24, 193684)]]

2022-12-06 17:30:24.211 DEBUG (SyncWorker_3) [pyhik.hikvision] Device motion namespace: http://www.hikvision.com/ver20/XMLSchema
2022-12-06 17:30:24.211 DEBUG (SyncWorker_13) [pyhik.hikvision] Device motion namespace: http://www.hikvision.com/ver20/XMLSchema
2022-12-06 17:30:24.213 DEBUG (SyncWorker_18) [pyhik.hikvision] Device motion namespace: http://www.hikvision.com/ver20/XMLSchema

[... cut ...]
2022-12-06 17:31:00.012 ERROR (Thread-5 (alert_stream)) [pyhik.hikvision] Problem finding attribute: 'videoloss'
2022-12-06 17:31:00.015 ERROR (Thread-4 (alert_stream)) [pyhik.hikvision] Problem finding attribute: 'videoloss'
2022-12-06 17:31:00.303 ERROR (Thread-3 (alert_stream)) [pyhik.hikvision] Problem finding attribute: 'videoloss'
2022-12-06 17:31:00.313 ERROR (Thread-5 (alert_stream)) [pyhik.hikvision] Problem finding attribute: 'videoloss'
2022-12-06 17:31:00.319 ERROR (Thread-4 (alert_stream)) [pyhik.hikvision] Problem finding attribute: 'videoloss'

mezz64, 2022-12-07:
Thanks, that was helpful.  I think I know the issue and committed a fix to a new branch (https://github.com/mezz64/pyHik/tree/vlossfix).  Can you test to see if the changes correct the problem?  You can download the hikvision.py and constants.py files and overwrite the local package copy you have to test if needed.

pdobrien3, 2022-12-07:
i run hass on docker, i dont mind replacing the files but have no idea how

yousaf465, 2022-12-07:
> Thanks, that was helpful. I think I know the issue and committed a fix to a new branch (https://github.com/mezz64/pyHik/tree/vlossfix). Can you test to see if the changes correct the problem? You [...]

Running HAOS, how to replace those files?

mezz64, 2022-12-08:
I've been able to replicate and test on my end.  I'll get the fixes pushed into a new version and a PR in to HA.

mezz64, 2022-12-08:
PR submitted (https://github.com/home-assistant/core/pull/83517).  Hopefully they can get it into the first point release.

pdobrien3, 2022-12-08:
cool. thank you very much !!

pdobrien3, 2022-12-09:
error gone.  thank you.  closing.

closed 2022-12-09 by pdobrien3