mezz64/pyHik #91: ANPR DS-2CD4A26FWD-IZS

julianbeere, 2023-01-11:
Is there any way to add Vehicle Detection as a supported sensor?
[image]

clementlecorre, 2023-08-17:
@julianbeere Any news ?

turbokongen, 2023-09-13:
I can provide some event logs if neccesary.

mezz64, 2023-09-13:
An event log would be helpful.  Once the event ID is known as long as the format is the same as the others it's an easy add.

julianbeere, 2024-02-18:
> I can provide some event logs if neccesary.

Did you provide event logs and did anyone manage to solve this?

freschism3000, 2024-04-15:
Want to ask the same question.
Getting Vehicle detection as a supported sensor would be fantastic, as I purchased a Hikvision ANPR camera for License plate detection specifically.

mezz64, 2024-04-15:
Unfortunately, I have not been provided an event log yet so I don't know the ID or format it reports in.  Without that information I can't add it.

freschism3000, 2024-04-15:
Let me know how I can help. How do I get an event log you can use from the NVR?

julianbeere, 2024-04-17:
Thanks @mezz64. How do I get an event log?
The ANPR camera has a whitelist that when triggered notifies the NVR which is configured to send an email to me.
The setting is attached (in the NVR)
[image]

mezz64, 2024-04-17:
You can find guidance for pulling an event log in past issues.  There are instructions here: https://github.com/mezz64/pyHik/issues/82#issuecomment-1020308020

freschism3000, 2024-04-19:
A few questions:

1. The NVR Configuration (Directly on the NVR), uses an older software than if I log into the camera directly.
I started with configuring the camera directly on the NVR, adding the camera as using the HIKVISION protocol.
My main issue here is that I cannot import an "Allowlist" for licenseplates. I can export the empty list from the NVR, add my license plates, but every time I try to Import the updated Allowlist, I just get an "Import Failed" message.

2. When logging into the camera directly, the software is much better. I read that if you use the camera with the NVR, the NVR settings will override the settings directly made on the camera, unless you change the NVR protocol from HIKIVISION to ONVIF. So I changed the NVR protocol for the Camera to ONVIF instead, and logged into the camera directly. 
Here I can add a Licenseplate using a wizard in the configuration under VCA > Roadtraffic > Allowlist.

3. I saved the settings on the camera, and added the camera manually though IP, using the Hikvision integration in home assistance. Then I used the command you linked above, to get realtime event data from the camera.
The data I get when driving up to the camera is this:

"<?xml version="1.0" encoding="UTF-8"?>                                                                                                                                      
<EventNotificationAlert version="2.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">                                                                                     
<ipAddress>192.168.1.64</ipAddress>                                                                                                                                         
<ipv6Address>::ffff:192.168.1.64</ipv6Address>                                                                                                                              
<portNo>80</portNo>                                                                                                                                                         
<protocol>HTTP</protocol>                                                                                                                                                   
<macAddress>e8:a0:ed:9e:60:47</macAddress>                                                                                                                                  
<channelID>1</channelID>                                                                                                                                                    
<dateTime>2024-04-19T10:11:58+01:30</dateTime>                                                                                                                              
<activePostCount>1</activePostCount>                                                                                                                                        
<eventType>VMD</eventType>                                                                                                                                                  
<eventState>active</eventState>                                                                                                                                             
<eventDescription>Motion alarm</eventDescription>                                                                                                                           
<channelName>Gate Camera</channelName>                                                                                                                                      
</EventNotificationAlert>         
"

It seems like it's "Just" a Motion Detection, and not actually a license plate detection.
Any ideas for next steps?

closed 2026-09-01 by mezz64