mezz64/pyHik #92: binary sensor not created

NicusorFlorinBaluta, 2023-08-18:
I added the configuration like this: 

```
binary_sensor:
  - platform: hikvision
    host: 192.168.50.246
    username: admin
    password: *******
```

and the system logger says: 

```
Logger: pyhik.hikvision
Source: components/hikvision/binary_sensor.py:157
First occurred: 11:15:00 (1 occurrences)
Last logged: 11:15:00

Unable to fetch events. Device firmware may be old/bad.
```

Am i doing something wrong?

```
Device info:

Device Name Outdoor Camera
Model DS-2CV2041G2-IDW
Firmware Version V5.5.800 build 210629
Encoding Version V7.3 build 200215
Web Version V4.0.1 build 200218
```

Any assistance is much appreciated 🙇

krismarc, 2024-04-18:
I just had the same problem with 4 out of 6 cameras.  Simply fixed by upgrading firmware.

try with new firmwares:
https://www.hikvision.com/en/products/IP-Products/Network-Cameras/Wi-Fi-Series/ds-2cv2041g2-idw/?subName=DS-2CV2041G2-IDW

http://<camera_ip>/ISAPI/Event/triggers should respond with triggers xml payload

NicusorFlorinBaluta, 2024-04-19:
Did that, same error though :(

closed 2026-09-01 by mezz64