mezz64/pyHik #93: etchannel_num = int(etchannel.text) - TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

acuszka, 2024-01-12:
Hi,

I'm using pyHik through Home Assistant. Everything was working fine for the last 2 years, but suddenly I started receiving the error message below. I recently upgraded my NVR firmware to V4.75.200, and I'm not sure if this is what caused the issue.

NVR: Hikvision DS-7604NI-K1/4P(C)
V4.75.200 build 231110

```
Error while setting up hikvision platform for binary_sensor

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/hikvision/binary_sensor.py", line 110, in setup_platform
    data = HikvisionData(hass, url, port, name, username, password)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/hikvision/binary_sensor.py", line 157, in __init__
    self.camdata = HikCamera(self._url, self._port, self._username, self._password)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pyhik/hikvision.py", line 116, in __init__
    self.initialize()
  File "/usr/local/lib/python3.11/site-packages/pyhik/hikvision.py", line 291, in initialize
    events_available = self.get_event_triggers()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pyhik/hikvision.py", line 435, in get_event_triggers
    etchannel_num = int(etchannel.text)
                    ^^^^^^^^^^^^^^^^^^^
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
```
HA config:
binary_sensor:
  - platform: hikvision
    host: 192.168.0.10
    port: 80
    ssl: false
    username: user
    password: pass

mezz64, 2024-01-12:
That error would suggest the updated firmware has changed the form of the XML response in the event stream which either changed the name for the channel field or removed it.

If you can post a snippet from the event stream when some events are occurring I can see if it can be accounted for.

acuszka, 2024-01-12:
> That error would suggest the updated firmware has changed the form of the XML response in the event stream which either changed the name for the channel field or removed it.
> 
> If you can post a snippet from the event stream when some events are occurring I can see if it can be accounted for.

Certainly. What specific steps should I take? Do you want me to export the log from the NVR through the web interface? Sorry a bit confused

mezz64, 2024-01-12:
You need to use curl to monitor the alertStream output.  Something like this:

`curl -u username --digest -N http://YOUR_NVR_IP/ISAPI/Event/notification/alertStream`

If you look through other active issues you can find some more detailed info if that doesn't work for you.

acuszka, 2024-01-12:
> You need to use curl to monitor the alertStream output. Something like this:
> 
> `curl -u username --digest -N http://YOUR_NVR_IP/ISAPI/Event/notification/alertStream`
> 
> If you look through other active issues you can find some more detailed info if that doesn't work for you.

```
<EventNotificationAlert version="1.0" xmlns="http://www.hikvision.com/ver20/XMLSchema">
<ipAddress>192.168.0.10</ipAddress>
<portNo>80</portNo>
<protocol>HTTP</protocol>
<macAddress>24:28:fd:09:e5:1d</macAddress>
<dynChannelID>3</dynChannelID>
<dateTime>2024-01-12T17:32:52</dateTime>
<activePostCount>1</activePostCount>
<eventType>linedetection</eventType>
<eventState>active</eventState>
<eventDescription>linedetection alarm</eventDescription>
<DetectionRegionList>
<DetectionRegionEntry>
<regionID></regionID>
<sensitivityLevel>0</sensitivityLevel>
<RegionCoordinatesList>
</RegionCoordinatesList>
</DetectionRegionEntry>
</DetectionRegionList>
<detectionPictureTransType>url</detectionPictureTransType>
<detectionPicturesNumber>1</detectionPicturesNumber>
<bkgUrl>http://192.168.0.10:80/picture/Streaming/tracks/303/?name=ch00003_00000000294048583526400388747&amp;size=388747</bkgUrl>
<URLCertificationType>digest</URLCertificationType>
</EventNotificationAlert>
--boundary
Content-Type: application/xml; charset="UTF-8"
Content-Length: 969
```

this?

acuszka, 2024-01-12:
full log
[new 13.txt](https://github.com/mezz64/pyHik/files/13921650/new.13.txt)

acuszka, 2024-01-12:
it looks like <channelID>0</channelID> is there for <eventType>videoloss</eventType>
but there's only <dynChannelID>3</dynChannelID> for <eventDescription>linedetection alarm</eventDescription> and <eventDescription>fielddetection alarm</eventDescription> ?

mezz64, 2024-01-12:
I didn't realize initially this is thrown on setup, the alertStream looks OK so something must be different in triggers.  Please post the output of this:

`curl -u username --digest -N http://YOUR_NVR_IP/ISAPI/Event/triggers`

acuszka, 2024-01-12:
[triggers.txt](https://github.com/mezz64/pyHik/files/13921918/triggers.txt)

mezz64, 2024-01-12:
Thanks.  From a quick look I think the problem is in some of the NVR status sensors.  They are set to notify the center so they are being parsed, but they don't report any ID's so it doesn't know what camera to associate them with.  You also have center notify turned on for videoloss events, which can cause problems.

Try turning off notifying the surveillance center on these alert types and see if it will startup ok.
diskfull
diskerror
illaccess
videoloss for each camera

acuszka, 2024-01-12:
I disabled all events and notifications in both the NVR and in each individual camera. However, I am still encountering the same error in Home Assistant.

[image]

Is it correct to assume that when Home Assistant initializes Hikvision/pyHik, the script fails due to not receiving the expected data during the 'get_event_triggers' process?

acuszka, 2024-01-12:
> Thanks. From a quick look I think the problem is in some of the NVR status sensors. They are set to notify the center so they are being parsed, but they don't report any ID's so it doesn't know what [...]

I disabled all events and notifications in both the NVR and in each individual camera. However, I am still encountering the same error in Home Assistant.

[image]

Is it correct to assume that when Home Assistant initializes Hikvision/pyHik, the script fails due to not receiving the expected data during the 'get_event_triggers' process?

mezz64, 2024-01-12:
Can you generate another output from the trigger endpoint once everything is disabled?  I'm wondering if the new firmware isn't replicating certain changes to the api endpoints.

acuszka, 2024-01-12:
> Can you generate another output from the trigger endpoint once everything is disabled? I'm wondering if the new firmware isn't replicating certain changes to the api endpoints.
[triggers2.txt](https://github.com/mezz64/pyHik/files/13924094/triggers2.txt)

mezz64, 2024-01-12:
With everything disabled do you get the same error or a different one?  Things looked to have changed correctly, but with everything off there won't be any sensors to setup so i'm not surprised it doesn't work.  Enable something (motion, linedetection) on one camera and see if it loads.

acuszka, 2024-01-12:
> With everything disabled do you get the same error or a different one? Things looked to have changed correctly, but with everything off there won't be any sensors to setup so i'm not surprised it doesn't work. Enable something (motion, linedetection) on one camera and see if it loads.

The error remains the same even when all settings are disabled except for motion detection.

`Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/hikvision/binary_sensor.py", line 110, in setup_platform
    data = HikvisionData(hass, url, port, name, username, password)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/hikvision/binary_sensor.py", line 157, in __init__
    self.camdata = HikCamera(self._url, self._port, self._username, self._password)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pyhik/hikvision.py", line 116, in __init__
    self.initialize()
  File "/usr/local/lib/python3.11/site-packages/pyhik/hikvision.py", line 291, in initialize
    events_available = self.get_event_triggers()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pyhik/hikvision.py", line 435, in get_event_triggers
    etchannel_num = int(etchannel.text)
                    ^^^^^^^^^^^^^^^^^^^
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'`

I enabled motion detection in NVR:
[image]

Triggers changed, they now include notificationMethod in VMD-1:

`<id>VMD-1</id>
<eventType>VMD</eventType>
<dynVideoInputChannelID>1</dynVideoInputChannelID>
<EventTriggerNotificationList version="2.0" xmlns="http://www.isapi.org/ver20/XMLSchema">
<EventTriggerNotification>
<id>record-1</id>
<notificationMethod>record</notificationMethod>
<dynVideoInputID>1</dynVideoInputID>
</EventTriggerNotification>
<EventTriggerNotification>
<id>center</id>
<notificationMethod>center</notificationMethod>
</EventTriggerNotification>`

mezz64, 2024-01-12:
Ok, i'll run your trigger output through independently and see if I can track it down.  Should be able to find some time this weekend.

mezz64, 2024-01-14:
I think i've found the problem and put together a fix in a new branch (https://github.com/mezz64/pyHik/tree/trig-fixes) for you to test.  Please replace the hikvision.py file in your local install with the one from that branch and give home assistant a restart to test.  If all works well i'll put together the necessary PR's to update the library and get things fixed in the main branch.

acuszka, 2024-01-14:
> I think i've found the problem and put together a fix in a new branch (https://github.com/mezz64/pyHik/tree/trig-fixes) for you to test. Please replace the hikvision.py file in your local install [...]

I seem to lack access to both `/usr/src/homeassistant` and `/usr/local/lib/python3.11/site-packages/pyhik/ ` while using the root account on SSH, as I'm running Home Assistant Operating System. I believe I need to create a hikvision folder in custom_components to prompt HA to load an updated pyhik script. 

Should I place the files from [](https://github.com/home-assistant/core/tree/dev/homeassistant/components/hikvision) inside this folder and modify them somehow to load a test version of the pyhik.py file?

mezz64, 2024-01-15:
I'm not familiar with the HA OS architecture these days, nor if you can get away with forcing something with custom components.

Let me run the changes through my tests to make sure something unintended didn't break.  If all work well I can push the new version to pypi which should make it easier for you to fetch it and test.

acuszka, 2024-01-15:
Thank you. I've asked about this on the HA forums, but it seems a bit complicated...

[](https://community.home-assistant.io/t/force-ha-to-use-custom-pyhik-version/673512/5)

Marsmz, 2024-06-02:
> I think i've found the problem and put together a fix in a new branch (https://github.com/mezz64/pyHik/tree/trig-fixes) for you to test. Please replace the hikvision.py file in your local install [...]

I encountered the same issue in Home Assistant, and the error messages were identical. After testing your modified hikvision.py, the binary sensor entities can now be generated correctly. Additionally, there were no other error messages found in the Home Assistant logs. It seems to be functioning properly now. I tested the line cross functionality, and it triggers correctly in Home Assistant.

mezz64, 2024-06-06:
@Marsmz Thanks for confirming.  I'll try to find some time to merge this in soon.

kennym, 2024-12-25:
seeing the same issue with the following NVR:

[image]

did you already fix this?

kennym, 2024-12-25:
> I think i've found the problem and put together a fix in a new branch (https://github.com/mezz64/pyHik/tree/trig-fixes) for you to test. Please replace the hikvision.py file in your local install [...]

so, I tested this by directly updating the Python in `/usr/` and it worked, I think:

[image]

Not seeing the cameras, though

Smajlinger, 2025-03-17:
I am strugging with same errors trying integrate hikvisions motion sensors. Not exacly sure how to replace hikvision.py file. Can someone help, I'm running Home Assistant Operating System.

akathimi, 2025-05-21:
@mezz64 I tried the fix and it works. It would be great if you could release it since many home assistant users are facing this. Also, the inline fix is hacky, unsupported and needs to be applied on every upgrade. Thank you 👍

pawee, 2025-07-10:
I had the same issue with DS-7108NI-Q1/8P/M running firmware V4.76.105 build 240820 but I run HASS OS on Yellow and have no access to pip. Here is how I integrated the fix by overriding the core component.

1. Clone the `home-assistant/core` and copy just `/core/homeasistant/components/hikvision` to `/config/custom_components/hikvision`
2. Download pyhik@trig-fixes and copy just the module into `/config/custom_components/hikvision/pyhik`
3. Add `"version": "1.0.0"` to manifest.json inside the hikvision component (required by HASS for custom components to load).
4. Add dot in front of the pyhik.hikvision import on line 8 to use local library: `from .pyhik.hikvision import HikCamera`

For anyone struggling too.